prusa3d/Prusa-Firmware-Buddy #3405: [FEATURE REQUEST] Hold down the knob to adjust speed

Liborsaf, 2023-10-25:
**Printer type** - [MINI, MK4, XL]

**Describe the expected functionality**
  Add hold down the knob to adjust speed function found in MK3 printers.

codingcatgirl, 2024-06-03:
*sigh*

congrats for enabling stalebot, this is totally not going to frustrate the heck out of your users.

anyway, this is a feature request, so of course it shouldn't be "stale"

wafflecart, 2024-06-03:
> _sigh_
> 
> congrats for enabling stalebot, this is totally not going to frustrate the heck out of your users.
> 
> anyway, this is a feature request, so of course it shouldn't be "stale"

Yea the stale bot makes following this repo extremely annoying. I followed just to get an idea of bugs coming in, what’s being worked on, release announcements etc but now get 50 spam messages a day from it 😡.

Prusa set up your own internal bot or your own search filter to run to check for stale issues. You don’t need to leave a comment on dozens of issues every day with a bot.

closed 2024-09-02 by github-actions[bot]