prusa3d/Prusa-Firmware-Buddy #3406: [BUG] BGCODE upload via PrusaLink

the0bone, 2023-10-25:
**Printer type** - XL5T

**Printer firmware version** - 5.1.0-alpha2

**Original or Custom firmware** - Original

**USB drive or USB/Octoprint**
  USB Drive via PrusaLink

**Describe the bug**
  Try to upload a bgcode file via the PrusaLink Dashboard. I only searches for *.gcode.
  Sure you can select *.* and it will work. But that extra step should be avoided.

s0uky, 2023-10-26:
I have a similar issue on MINI+ with FW 5.1.0-alpha2 with Wi-Fi adapter ESP-01S from Prusa. 

Uploading GCODE through the PrusaLink dashboard doesn't work. I choose the file, drop it to the website, upload start and after a short period, the WiFi on MINI+ freezes and connectivity to the printer drops. 
Only restart put the printer back online.

zebpalmer, 2023-11-08:
For me on mac & Mk4 I can drag and drop a bgcode file into prusa link and everything works fine. however opening the finder via "click to choose" shows all the bgcode files grayed out.

jalexvig, 2023-12-01:
> For me on mac & Mk4 I can drag and drop a bgcode file into prusa link and everything works fine. however opening the finder via "click to choose" shows all the bgcode files grayed out.

Same - input file types accepted in the javascript need updating

iridris, 2024-01-03:
There was a [PR](https://github.com/prusa3d/Prusa-Link-Web/pull/452) merged back in November that looks like it added support for BGCODE files, but I think one spot may have been missed.

This particular line only lists GCODE, not BGCODE, and as best I can tell this is the line that is telling the browser what filetypes to filter: https://github.com/prusa3d/Prusa-Link-Web/blob/master/webpack.config.js#L31

bkerler, 2024-01-06:
This issue should be closed and reopened [here](https://github.com/prusa3d/Prusa-Link-Web) instead, as it's not relate to the xbuddy fw but to PrusaLink.

danopernis, 2024-06-03:
This is fixed since version 5.2.0 via commit https://github.com/prusa3d/Prusa-Firmware-Buddy/commit/ae12ea284b3e60acbeb2be27322067bc79f0c719

closed 2024-06-03 by danopernis