prusa3d/Prusa-Firmware-Buddy #3407: [BUG] Unable to navigate folder structure that has more than 40 folders in 5.1.0-alpha2

jonmlee, 2023-10-26:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - [MINI]
Mini

**Printer firmware version** - [e.g. 4.0.5, ...]
5.1.0-alpha2 

**Original or Custom firmware** - [Original/Custom]
Original alpha

**Optional upgrades** - [e.g. Filament Runout Sensor]

**USB drive or USB/Octoprint**
  Please let us know if you print via USB flash drive or USB/Octoprint
USB

**Describe the bug**
  A clear and concise description of what the bug is.
Unable to navigate tree when there are lots of folders works fine alpha1 and 4.4

**How to reproduce**
  Please describe steps to reproduce the behavior.
Turn nob to up and down tree. It freezes when scrolling

**Expected behavior**
  A clear and concise description of what you expected to happen.
Be able to navigate tree

**G-code**
  Please attach the G-code you had trouble with. This will make it easier for us to replicate the error.

**Crash dump file**
  Please attach the crash dump file. This will make it easier for us to investigate the bug.

**Video**
  Please attach a video. It usually helps to solve the problem.

dan2304k, 2023-11-01:
Also seeing VERY slow UI response when navigating a thumb drive with 20+ folders with 5.1.0-alpha2.  This was not a problem with 5.1.0-alpha1

mmcglumphy, 2023-11-06:
Same. Consistent across 4 minis.

dan2304k, 2023-11-18:
Same issue with 5.1.0 beta.  This renders the UI virtually unusable.  @Prusa-Support Please prioritize this item.

mmcglumphy, 2023-11-18:
Also seeing this in 5.1.0 beta on both Mini's and MK4's.

Mryck, 2023-11-23:
I do also have the same issue on the final 5.1.0 if think it was working fine on the early alpha's

dan2304k, 2023-11-23:
Bump.  Makes the mini virtually unusable with my standard workflow.

rianweaver, 2023-11-26:
I just went from 5.0.0 to 5.1.0 (release versions, not betas)  on my Mk4 printers, and this is really a problem.  It's painfully slow to use the same USB drive and files as was working fine in 5.0.0

dstulken, 2023-12-10:
> I just went from 5.0.0 to 5.1.0 (release versions, not betas) on my Mk4 printers, and this is really a problem. It's painfully slow to use the same USB drive and files as was working fine in 5.0.0

I can confirm the same behavior on 5.1.0 release version on the Mini+.

dan2304k, 2023-12-17:
Still a problem with 5.1.2!  @Prusa-Support

Mryck, 2024-01-24:
@Prusa-Support any update on this issue ? I'm stuck on fw 4.4.1 because of it, thanks !

jonmlee, 2024-01-24:
They're trying to drive everyone to Prusa Connect. That's my workaround right now. Navigating via a browser works and is fast on 5.1.2.

dstulken, 2024-01-25:
> @Prusa-Support any update on this issue ? I'm stuck on fw 4.4.1 because of it, thanks !

There were quite a few significant regressions on the 5.x.x branch unfortunately...   

I'm still waiting for any kind of response about the fuzzy skin issue: https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/3571
So in the meantime, we have some of our Minis on 4.4.1 (for fuzzy parts), and some on 5.x for the faster print speeds input shaping allows.  It's a bit of a nightmare from the logistics perspective.

Mryck, 2024-02-08:
I have compile v5.2.0 with the Mini preset to see if something has been done while working on the MK3.5 fw, but unfortunately moving in USB folders is still very slow..

Prusa-Support, 2024-02-10:
Thanks for reporting.

This may be a very specific scenario that may have not been rigorously tested so far, as using multiple USB flash drives could generally be a safer option. These USB flash drives work under arguably difficult conditions due to heat, magnetic fields, and not only, so a flash drive may fail without notice - probably not the most reliable option for heavy file storage.

However, I understand that this is not how it used to work FW 5.1.0 and that some of you may have developed a specific workflow on this.

I hope Prusa Link/Connect and Input Shaper will be worthy allies to boost your production while we investigate this issue.

Michele Moramarco
Prusa Research

Mryck, 2024-02-10:
@Prusa-Support thanks for the investigation to give more information I use Sandisk USB an all my mini's suffer the same issue, as reported it was ok prior 5.1.0 alpha2 , Input Shaper does improve the production indeed ! For Connect It would very nice to have it self host

dstulken, 2024-02-13:
> This may be a very specific scenario that may have not been rigorously tested so far, as using multiple USB flash drives could generally be a safer option. These USB flash drives work under arguably [...]

We appreciate that you are trying to help, but this response reads like we are being told we are using the machines wrong? 
 
Being able to browse folders on the USB stick is something that has worked well for years.  And it's something that works equally well - with the exact same USB sticks - when 5.x.x Minis are reverted to 4.x.x firmware.  This is very much a core functionality regression - not a fault of the USB sticks or any environmental condition around them.

rianweaver, 2024-02-14:
While I hope this gets addressed as it's still very slow on MK4s with 5.12 firmware, I have found SOME improvement using aftermarket USB drives.   They seem somewhat faster than the included USB drives using the same gcode/bgcode folders/files.

CZDanol, 2024-02-15:
Internal ticket: BFW-4765

itrmc, 2024-03-29:
Seeing the same behavior on MK4 and XL with latest firmware. Scrolling is very slow. Flashing light on one USB stick indicates it may be rereading the folder each time the screen is refreshed.

CZDanol, 2024-03-29:
@itrmc what do you mean by the latest firmware specifically? I wrote a fix for that has gone into 6.0 RC, which was released just yesterday - however only for Mk4.

The flashing light is okay, it indeed does re-read the structure on each refresh - we cannot get around that, there is not enough memory on the chip to warrant storing the whole file list. The previous implementation however was doing some unnecessary lookups that I optimized out.

dstulken, 2024-04-02:
>  I wrote a fix for that has gone into 6.0 RC, which was released just yesterday - however only for Mk4 so far.

Happen to know when this will make it to the Mini?   It hasn't had a firmware update since mid December of last year...

CZDanol, 2024-04-02:
> Happen to know when this will make it to the Mini? It hasn't had a firmware update since mid December of last year...

Sorry, I don't know the release schedule.

rianweaver, 2024-04-17:
This is so much better in 6.0 for MK4.  Thank you, @CZDanol

closed 2024-04-17 by CZDanol

jonmlee, 2024-04-17:
Can confirm it's fixed on the Mini.

dstulken, 2024-04-26:
I can also confirm.  Massive improvement, thank you!