prusa3d/Prusa-Firmware-Buddy #3410: [BUG} Octoprint doesn't give printer status on LCD

srjacob999, 2023-10-27:
Printer type MK4
Printer firmware version - 5.00
MMU upgrade - No MMU
MMU upgrade firmware version - NO MMU

** USB/Octoprint**

Describe the bug
When using the MK4 with Octoprint, the printer status and job times do not appear on the printer LCD

To Reproduce
Run the printer with Octoprint

Expected behavior
Print status should appear on the LCD screen

G-code
Any print

Video
None

jdfalk, 2023-10-27:
This is probably a duplicate of https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/189

dandrzejewski, 2023-11-12:
Could also be related to #1497.

srjacob999, 2023-11-12:
No.  That relates to the 8 bit firmware on the Mini.

danopernis, 2024-06-03:
Duplicates https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/189

closed 2024-06-03 by danopernis