prusa3d/Prusa-Firmware-Buddy #3411: [BUG] Nozzle cleaning fails

srjacob999, 2023-10-27:
Printer type - MK4
Printer firmware version - 5.0
MMU upgrade - No MMU
MMU upgrade firmware version - No MMU

Does this with both SD card and USB/Octoprint

Describe the bug
Nozzle cleaning fails See Prusa Forums: https://forum.prusa3d.com/forum/english-forum-original-prusa-i3-mk4-hardware-firmware-and-software-help/nozzle-cleaning-failed-error/

To Reproduce
Doing any print, Using PETG, but other filaments do this

Expected behavior
Should clean nozzle and pass nozzle cleaning

G-code
Please attach a G-code. This will make it easier for us to replicate the error.

Video
Please attach a video. It usually helps to solve the problem.

codingcatgirl, 2023-10-29:
Duplicate of #3346 or #3347?

srjacob999, 2023-10-29:
Similar, but please read the thread on the Prusa forum about this.  I have included a link to that forum discussion.  It is a real problem that needs fixing, and probably a major problem on the XL.

stu54, 2023-11-03:
I had this problem and nothing the tech support people had me try fixed it (after 6 hours on chat). I broke the appendix on the board and downloaded my own firmware compile with some of the nozzle clean code commented out and the printer works fine. The code does not do anything with nozzle cleaning. It is doing some calculations that fail but what is being calculated is unclear. If a developer could help me, we could probably figure out why the code is failing and what might be wrong on the printer that causes the failure.

closed 2024-06-26 by github-actions[bot]