prusa3d/Prusa-Firmware-Buddy #3412: Dual-tool XL having issues with layer shifts and bulging layers possibly related to prusaslicer 2.7 alpha as well

Matt-Chu, 2023-10-27:
Here's a link to photos, videos, .3mf files. 

https://drive.google.com/drive/folders/1RyDuHPW_X0B4YwZBfz7d_azFaTFCfQty

**Printer type** - Prusa XL Dual-Tool (factory assembled)

**Printer firmware version** - 5.1.0 Alpha2

**Original or Custom firmware** - [Original]

**Optional upgrades** - none

**USB drive or USB/Octoprint**
  USB

**Describe the bug**
  Repeated layer shifts, bulging, and low quality layers using new alpha firmware and 2.7 prusaslicer

**How to reproduce**
.3mf files included

**Expected behavior**
To not look like hot garbage

**G-code**
Don't have gcode, but I do have .3mf files

**Crash dump file**
Don't have crash dump file

**Video**
it's in the link

sabin-petru, 2023-10-30:
I have x shift too after installing 5.1.0-alpha + 2.7 slicer in my MK4

Mikyner, 2023-10-30:
Confirm, today I had 2x layer shifts on MK4 5.1.0a2 + 2.7a1 slicer

GuyH77, 2023-11-01:
As above. I've had two prints with layer shifts now as well. The one I was watching detected a crash, homed and carried on printing but slightly off hence the shift. There was nothing on the print to crash in to, everything was printing beautifully up to that point. Sliced with 2.6.1 0.6 IS profile. I'm running the alpha FW though.

BrienKing, 2023-11-02:
I just reported a similar issue on the Slicer side:
https://github.com/prusa3d/PrusaSlicer/issues/11570

I'm not sure can be called a "Layer shift" in my case because the bulge runs all the way around the part evenly.  If it was layer shift, I would expect it to bulge on one side and be indented on the other.

closed 2024-06-26 by github-actions[bot]