prusa3d/Prusa-Firmware-Buddy #3413: Where is the park point coming from during M601

dunkmann00, 2023-10-27:
I have a Mk4 and have done some prints where I use the layer color change feature (which as you know, uses M601).

I was curious of the specifics of the M601 command and found what I am pretty sure is the function for it [here](https://github.com/prusa3d/Prusa-Firmware-Buddy/blob/bfb0ffc745ee6546e7efdba618d0e7c0f4c909cd/lib/Marlin/Marlin/src/gcode/feature/pause/M600.cpp#L62).

I was able to follow most of it but couldn't determine how the park position is set. In the `GcodeSuite::M600` function, the `park_point` variable is set here:
https://github.com/prusa3d/Prusa-Firmware-Buddy/blob/bfb0ffc745ee6546e7efdba618d0e7c0f4c909cd/lib/Marlin/Marlin/src/gcode/feature/pause/M600.cpp#L121

But when examining the `NOZZLE_PARK_POINT` macro in the Mk4 [config](https://github.com/prusa3d/Prusa-Firmware-Buddy/blob/bfb0ffc745ee6546e7efdba618d0e7c0f4c909cd/include/marlin/Configuration_MK4.h#L1574), that is not the location I have seen my printer park during a color change. The location it looks like its parking at when I've used it is `NOZZLE_PARK_POINT_M600` (that macro is in the next line in the same config file). Based on the name, it also would indicate this is the location that is meant to be used, but I couldn't figure out how `NOZZLE_PARK_POINT_M600` gets set in `GcodeSuite::M600`.

Again, this is just out of curiosity and to try to better understand my printer.

Thanks for all the work you do with the various repositories to keep all the Prusa printers happily printing!

bdaroz, 2024-01-22:
See also #3317

bdaroz, 2024-06-04:
Not stale

bdaroz, 2024-08-26:
Still important. :(

vojtanvk, 2024-08-26:
Hey, just double checked it and the parking position is set by the same macro [here](https://github.com/prusa3d/Prusa-Firmware-Buddy/blob/bfb0ffc745ee6546e7efdba618d0e7c0f4c909cd/src/common/marlin_server.cpp#L1852).
As for the M600, it truly parks on the `NOZZLE_PARK_POINT_M600` and is set [here](https://github.com/prusa3d/Prusa-Firmware-Buddy/blob/bfb0ffc745ee6546e7efdba618d0e7c0f4c909cd/src/marlin_stubs/pause/M600.cpp#L100). Hope it helps understand you printer better :)

closed 2024-08-26 by vojtanvk