prusa3d/Prusa-Firmware-Buddy #3418: [ENHANCEMENT] Cancel object issues/enhancements

rcmaniac25, 2023-10-29:
Enter what type of printer or upgrade the enhancement applies to.
**Printer type** - XL Single Tool
**Firmware** - 5.1.0 alpha2
**Optional upgrades** - Stock

So this has a couple parts and my option was to make multiple tickets that all referenced each other, or to make 1 ticket that explained the different requests.

With the new firmware alpha, I wanted to give the cancel object a try. I had the perfect print: a bunch of small pegs useful in assembling the [mega lego skeleton](https://www.printables.com/model/78695-mega-lego-like-skeleton-101-scale) model. 50 pins are provided. Given their size and small footprint, I expected that at least one would detach and would be a good time to use cancel object.

All models were sliced using PrusaSlicer 2.7 alpha1 with stock printer profile (aside from infill amount and some paint on supports).

Issue 1) Cancel Object list is limited to 16 items.

I realize this is a challenge based on memory and UI usability, but PrusaSlicer decided to arrange the pin printing order in a maze-like format with multiple zig-zag motions, instead of clean left/right/top/button movements, meaning if I saw a pin that had failed, I couldn't just count 1, 2, 3, 4... and get the pin and mark cancel. Even if I went through the PrusaSlicer UI to find the order of the objects (since we don't have a [picture representation](https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/3397) to go off of), I am only able to see 16 of the 50 pins.

Issue 2) The general work around for not having the picture is `Cancel Current`, which should be able to cancel the currently printing object... unless it's none reactive

So as my print of many small pins started, 1 pin detached. I went to cancel objects menu and waited for the extruder to come around again because (see issue 1). Extruder moved over the spot, I selected "Cancel Current" and it continued printing it. As this was my first use of the feature in the alpha, I thought maybe it skips it next layer as it may be scheduled/buffered already. But next time it came around, it tried to print it again. I pressed "Cancel Current" and it still continued printing. Eventually a 2nd pin detached. Then the spaghetti from one detached pin knocked into another, now I have 3 detached pins. Each time, I tried "Cancel Current" but to no success. Eventually it became like a [well known comedy routine](https://www.youtube.com/watch?v=AnHiAWlrYQc) (I Love Lucy candy scene) where I kept trying to stop one pin from causing problems for another pin. Without success as it kept going without stopping.

I eventually got 5 pins out of the original 50, which maybe 25 or so half finished pins which are good enough as they're guide pins for the model.

What I noticed, from looking at source code and the UI, is that if there are more then 16 objects being printed, objects 17 and up don't appear to be under the control of cancel object. As I kept trying to cancel pin after pin that failed because of the failed prints being dragged around, I reached a pin that was on the 1-16 list... when I did "Cancel Current" for that one, the extruder instantly stopped and moved to the next object. The UI said "canceled" and next layer, it skipped it. Making me think the 16 object limit isn't just UI.

While it was a good theory, I tried an unrelated print (the face parts for the early mentioned lego skeleton) on my MK4 (original, also running 5.1.0 alpha2) and it only had 4 prints. All 4 were listed in the UI. I noticed a small part broke off one of the objects and it started dragging it around. I went into cancel object and again chose "Cancel Current". Once again, nothing happened. The extruder continued to print it. I pressed it again, and it again continued printing it. Having only 4 objects, uniquely named, I scrolled to the one object and clicked it. The name changed from Printing to Canceled and the extruder immediately stopped and moved to the next object. The Cancel Current did not work, even when there was only 4 objects.

Issue 3) Cancel Object is available for single item prints

After I finished 3 of the 4 mouth prints, I went on to print just the failed part. A single print. I noticed that cancel object is still there. If I needed to cancel the only print available, I'm not sure what would happen. I did not try... but it may be an edge case that can be avoided by simply not having Cancel Object UI if there's only 1 object. It also is redundant UI since canceling a single object is really just stopping a print, which has a "are you sure?" dialog, unlike cancel object, which has no confirmation dialog.

rcmaniac25, 2023-10-31:
Minor update on Issue 3: while not a single object, I tried the same pin print (only 14 pins though) on my MK4 (also running 5.1.0 alpha2) and after a bit they started popping off. I started canceling objects and ran into issue 2, as it didn't cancel things as it was printing all the time. Maybe once 1/3 of the time it canceled the next object instead of the current object.

But it got to the point that I had canceled every object.

At this point, the extruder appeared paused... but the z height would go up in incremental steps. So it was obvious that it was still running in the background. I went to cancel the print, but the UI became almost unresponsive. Lagging, even a few screen tears. As it's not a big print, eventually the print finished. It did the ending sequence, and the UI became responsive again.

Perhaps some logic that indicates if there are no more objects to cancel, it asks if you want to stop the print or similar. Or at least not spin-looping code in the background that causes the UI to become unresponsive.

rcmaniac25, 2024-06-04:
I feel I have to bump this. PrusaConnect adds support for a pictorial version of cancel object and offers cancel buttons for more then 16 items.

Yet only 16 are supported.

Not to mention competitors and various other embedded platforms/firmwares support more then 16 cancel objects.

Ideas:

- Obviously, if there's 16 or less items, there is no issue.
- If there's a small amount of 16 objects, say 32-64 or so, then an LRU could be used and populated every 4-8 objects until there's no more cache misses since there's only 16 left. This would probably cover a vast majority of prints where someone puts an entire model's part onto the built plate. They'll probably be different heights and result in eventually having 16 or less models that are cancel-able.
- If there's 100 pins on a print, then a ring buffer may be better to just cycle through. This way it just cycles through everything.

The LRU and ring buffer could have the added UI of marking an item as done. So instead of seeing an option to cancel an object that finished, it only shows the ones that are still printing. Paged loading would then mean the UI doesn't need to have a hard-coded list of 16 items.

The end goal is to not provide the cancel object feature, only to suddenly discover the failed part 17 can't be canceled because of a hard coded limitation that the UI (Both on device and on PrusaConnect) does not show.

rcmaniac25, 2024-08-26:
Bump (sorry, until this is addressed for manually closed, I have the stubbornness to keep the bot from closing it)

jackTfrost, 2024-10-22:
I ran into this same issue. I was printing a full XL bed of small objects, 98 to be exact. When I went to cancel current object nothing happened. I couldn't load this through Prusa Connect because it was 90mb and it was going to take 90 minutes to transfer it, even though my printer is connected via ethernet to my network (that is another issue all by itself). I had to scrap the whole print because I couldn't cancel the 5 or so objects that pulled off the bed.

Tupson444, 2024-10-25:
Cancel Objects is not a complete feature if it doesn't support at least hundreds of objects. It is most likely needed exactly when printing many small objects.

I currently use this workaround: I first merge e.g. 9 objects into one, then fill the bed with those merged ones. Like this, I can print 108 small parts while having only 12 "objects" in the list for cancelling. The downside is of course that if one part detaches, I have to cancel the whole group of 9, but it is much better than ruining the whole print.

CZDanol, 2024-10-29:
Internal tickeet: BFW-6258

oriddlero, 2024-12-30:
Bug - Prusa Connect w/ XL - Cancel Object Fails Error 422 w/ large number of objects
PrusaConnect - Current Version

Printer type - XL - 5 Head

Printer UUID - 8f54f7ba-2550-4658-865d-0e6abbc44f6d

Printer firmware version - 6.1.3+7898

PrusaSlicer version - 2.9.0

Description of the bug, including steps to reproduce (if possible):

Cancel object is a critical function to me, I am a manufacturer, I fill the plate every print, every time with a large number of small objects (up to 500).  When an individual object fails I must have the ability to cancel it so the entire, multi-day print does not fail.

Bug: can’t seem to cancel objects consistently, seems like the larger number objects will not allowed me to cancel and  some of the smaller number objects still allow cancellation.  The error I received through Prusa connect is “HTTP error 422 - command was rejected by the printer” see screenshot below.  In the screenshot I tried to cancel 422 and got the error, however, if I tried to cancel object 200 the cancellation goes through normally.  I’m not sure if it is a number limitation, or if it is a part:plate orientation related issue but I really need this fixed ASAP.  I was able to replicate the issue after cycling power to the printer with the same job.

While you're at it would make a lot of sense if the  object cancellation GUI  was at the top of the screen, above the list of the objects, not randomly three quarters of the way down the list.   obviously this GUI is the only efficient way to identify the part needing to be canceled with a high number of objects on the bed and will be used each and every time a cancellation is needed.

murk-sy, 2025-01-09:
Copying my comment from https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/3073#issuecomment-1552445123

----

Just as an idea, 2 quick on-printer GUI mockups

### A: Pregenerated image with numbers
[image]
[+] Entirely generated in slicer, relatively little work in firmware
[+] Only needs to display file list and a static image
[-] Numbers may be unreadable with a lot of small parts (or frankly in general)
[-] Requires scrolling through a file list and matching numbers

### B: X/dot marks the spot
[image]
[+] Better readability
[+] Does not actually require the file list displayed
[-] Requires reading model centers and more work in the firmware
[-] If a lot of identical parts are listed, finding the correct one may be tedious if they're not numbered in a logical manner

### Side notes:
- Prusa Mini (and I guess MK4 and XL I believe?) display is 240×320. Therefore you can fairly easily fit Mini and MK4 previews at 1px\~1mm and XL preview at 1px\~2mm. The above images are 160x160 px for the mini bed.
- Perhaps a combination of both would work best.
- [image]
Highlighting the entire object might be the fanciest way to do it that's easiest to read, but that would require a lot more image generation and is likely not worth the extra file size/effort.

Ganj44, 2025-02-16:
I do have same issue, the only difference is I am not getting the error in Prusa Connect instead it says "Command send successfully." Regardless the object(192 and220) is not cancelled.

On printer controls it only shows first 16 objects (they are not numbered). Even after trying "cancel current object" object was not cancelled.

We are printing 80-300 parts per plate and this feature is critical for our operation.

Here is how it looks after canceling all available objects on printer.

[image]

[image]

oriddlero, 2025-02-16:
Two months, since my post... no updates...

CZDanol, 2025-02-17:
Sorry, plenty of other work - I have this in mind though

CZDanol, 2025-03-06:
Hello, just merged support for cancelling 1024 objects. If all goes well, it should come out in 6.4, but I don't have any estimates about the release date.

oriddlero, 2025-03-06:
> Hello, just merged support for cancelling 1024 objects. If all goes well, it should come out in 6.4, but I don't have any estimates about the release date.

Currently for the XL we are on Firmware 6.2.2, so it sounds like we are potentially many many versions away?  sorry if I'm incorrect, could you just clarify a bit for us?

murk-sy, 2025-03-06:
6.4 is the next minor (since the second number changes) release since 6.3 was recently released (but only for Core One, other models will likely follow to that version in the next few weeks).

If I had to guess based on previous releases then I'd say 1.5-4 months from now is a fairly realistic timeframe.

oriddlero, 2025-03-06:
Thank you kindly for the update, I am super excited for this,  is this function missing has been a massive headache for us!   it's great to know it Has completed initial development  and is on the immediate roadmap.   I'm available for testing if you need it!

CZDanol, 2025-06-25:
The support for 1024 objects is done and ready to be shipped with the 6.4 release. I cannot comment on the release date however, sorry :(

oriddlero, 2025-08-27:
Any chance this is coming out soon?  I REALLY need it. =[

oriddlero, 2025-08-31:
in the meantime, here's a picture of me suffering with 2-day prints periodically losing adhesion on 1/300 units causing me to come up with crazy acts like this example to keep myself from losing the entire print...  layers of paper with double-sided tape, lots of stress, **not** fun.   I have several XL printers printing 24/7 so this does come up from time to time more often than I would like.

 my old print farm on older printer Prusa models it was so easy to cancel an object with Octoprint when something like this happened =[ 

please tell me this firmware is going to be released soon?

[image]

murk-sy, 2025-09-01:
As a workaround you could merge the objects into one object with 9 or 12 parts (whichever would be within requirements) and then cancel a larger group of them.
It's maybe not optimal but it's better than nothing

3d-gussner, 2025-09-13:
Thanks for testing, reporting back and update all related github issues. Very appreciated 🤗 

There are different opinions about auto retract see https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/4810#issue-3412106187

murk-sy, 2025-09-20:
@oriddlero See this issue for that problem https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/4812
I've reverted to previous stable, I'd recommend doing the same until a new alpha/other is out

3d-gussner, 2025-09-22:
@oriddlero @murk-sy Thanks for your feedback. As this issue is about Cancel object please don't add other issue here.
Sorry that you lost prints because of the other issues.

closed 2025-12-01 by github-actions[bot]