prusa3d/Prusa-Firmware-Buddy #3422: [BUG] XL 5.1.0 Alpha2 - Blue Screen error - Precise_Stepping

UncleJessy, 2023-10-30:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - XL

**Printer firmware version** - 5.1.0 - Alpha2

**Original or Custom firmware** - [Original]

**Optional upgrades** - [N/A]

**USB drive or USB/Octoprint**
  USB & VIA Connect

**Describe the bug**
When running a print on the XL sliced in Prusa Slicer 2.7 Alpha using input shaper profile I get a Blue Screen / Print pauses and the machine is unresponsive. This has occurred 3x now. 

**How to reproduce**
Sliced a file in 2.7 (sliced file linked) - using the XL input shaper - Sent gcode via Connect & also tried via USB and each time produces Blue Screen - Precise_Stepping error 

Appears to be the same issue as seen on the Mini in 
https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/3288

**Expected behavior**
  Print should continue and not lock up the printer

**G-code**
Gcode and crash report - (https://drive.google.com/drive/u/5/folders/1YP-JpEFf-DUx9nHwIUmvrV7EjXcAzTU1)

**Crash dump file**
(https://drive.google.com/drive/u/5/folders/1YP-JpEFf-DUx9nHwIUmvrV7EjXcAzTU1)

**Video**
Video included in Drive folder
https://drive.google.com/drive/u/5/folders/1YP-JpEFf-DUx9nHwIUmvrV7EjXcAzTU1

SamD98, 2023-10-31:
I have encountered the same bug multiple times, here is the gcode, a crash report, from the printer, and a photo of the error.
[Bluescreen.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/13218346/Bluescreen.zip)

GuyH77, 2023-11-11:
I've had the same error.

TheYang, 2023-11-14:
Hey there, I've got the same issue on my 2 Toolhead XL. 
5.1.0-Alpha as well as Prusa Slicer 2.7.0 Beta.
Happened 3 times in a row.  
Attached are the Project file (80% sure that produces the gcode with the issue, but I may have changed something in the configuration), the buddy dump as well as the gcode.  
[precise_stepping.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/13350584/precise_stepping.zip)  

This seems to be the same as #3463 and #3454 and #3337

tomaszdudziak, 2023-11-21:
I got into the same. The Gcode looks OK but it produces this error (checked two times). My negative value is pretty big (-917) so I'm attaching my gcode (sliced in Prusa Slicer 2.7.0-alpha1), because It could help debugging.
[image]

[blue-screen.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/13423702/blue-screen.zip)

meermanr, 2023-11-22:
Happened to me too. PrusaXL 5T, local USB drive (the Prusa branded one that came with the printer), bgcode uploaded via PrusaLink (LAN) from PrusaSlicer 2.7-rc1.

BGCODE, screenshot, and crash dump attached: 
[precise_stepping_BSOD_PrusaXL-5.1. 0-BETA+13358.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/13443030/precise_stepping_BSOD_PrusaXL-5.1.0-BETA%2B13358.zip)

closed 2024-06-27 by github-actions[bot]

spjt, 2025-06-27:
Still seeing this in 6.2.5 on around 10% of longer prints. I notice the BSOD and prior assert have been commented out in master (with internal issue id BFW-5954). I backported this to 6.2.5 about a week ago and have not seen the error since or had any failed prints that were not attributable to other causes. So I'm pretty sure it's still happening, but setting `step_event.time_ticks = 0` doesn't seem to harm anything noticeable, certainly nothing as bad as just failing the print entirely. 

So there seems to be a workaround, at the loss of your warranty.

Unfortunately, I don't have the knowledge of the code to be able to add reporting if/when this condition happens without failing the print, would possibly be helpful.  

In fact some kind of general warning mechanism would be helpful, I'm not aware of any existing other than logs, but they have to be turned on every time and I have not yet once remembered to do that and then check them after the print.