prusa3d/Prusa-Firmware-Buddy #3424: [FEATURE REQUEST] MK4 Loud Beep when color change M600

gigi84, 2023-10-31:
**Printer type** - [MK4]

With the current FW of the MK4, there is not a loud beep when changing color (GCODE M600). I used to have this beep on my previous MK3S and it was essential to hear the printer when was the time to change color.
Please put back the same feature in the MK4, would be great if the user can customize it too

codingcatgirl, 2023-11-01:
Using gcode M300, you should be able to create custom beep sound to your custom filament change gcode in PrusaSlicer. (see https://help.prusa3d.com/article/prusa-specific-g-codes_112173)

gigi84, 2023-11-01:
I played a bit with the M300 but still it is not loud. I set the "Loud" mode but still the sound is not loud like my MK3S. How can I change it?

codingcatgirl, 2023-11-01:
I can't tell you. Depending on the kind of beeper they use (i don't know any details) different frequencies can have different volumes.

closed 2024-06-27 by github-actions[bot]