prusa3d/Prusa-Firmware-Buddy #3429: [FEATURE REQUEST] MK4 LCD display showing remaining time to next color change

gigi84, 2023-10-31:
**Printer type** - [MK4]

Additionally to the total remaining time, I would like to see on the LCD display (and PrusaLink) the remaining time to the next color change

codingcatgirl, 2023-11-01:
This would be great!

schorsch3000, 2023-11-02:
also include pauses in that, please :-)

Claghorn, 2023-11-03:
Yep. Time to anything that requires human interaction would be very handy.

Pdamga, 2023-11-05:
👍

gdbarron, 2023-12-16:
Duplicate of #3107?

gegy1, 2024-06-12:
Yes, this would be very useful for the XL when adding a pause in the slicer. The printer (and also PrusaConnect) should show the time when the pause will be reached. (Remaining time and clock time)

danopernis, 2024-06-12:
@gegy1 this should be already fixed in the firmware since at least v6.0.0, can you please confirm that?

gegy1, 2024-06-12:
Yes your are right, i can see the remaining time for next filament change. What i mean is, that i do not want to see the remaining time, i want to so the time like "Next filament change (pause) at 7:32pm" :)

[image]

nomike, 2024-08-16:
While the remaining time until the next filament change is shown in PrusaConnect, it would be awesome if this were also displayed in PrusaLink.

schorsch3000, 2025-11-05:
This issue issn't stale because interst is dropped, but because there are no open questions and it is not worked on (which is fine, but closing because its not worked on is not)