prusa3d/Prusa-Firmware-Buddy #3430: [BUG][MINI+] Printer crashes after uploading gcode through Prusa Connect

elxris, 2023-11-01:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - MINI+

**Printer firmware version** - 5.1.0-alpha2

**Original or Custom firmware** - Original

**Optional upgrades** - Filament Runout Sensor, ESP Wi-Fi module (3rd Party, Aliexpress)

**USB drive or USB/Octoprint**
  USB Drive - Prusa Connect

**Describe the bug**
  Uploading a file from Prusa Connect will always crash the printer. The printer was idle all times.

**How to reproduce**
 Install the 5.1.0-alpha2 to the Mini+ printer. Setup the wifi. I had already setup Prusa Connect on the previous alpha so I just downloaded the settings file again. Install the latest PrusaSlicer 2.7.0-alpha1, add a physical printer to Prusa Connect. Use the IS profile for the mini. Just some custom object setting overrides. Try to upload the file through PrusaSlicer *Mini crashes*, Try again from the Prusa Connect Platform *Mini crashes again*. Saving crash dump.
 
 NOTE: The crashing comes from the printer downloading the file. It will not even start the print, but apparently the file gets downloaded before crashing

**Expected behavior**
 To not crash while printer trying to download the file from Prusa Connect.

**G-code**
  Please attach the G-code you had trouble with. This will make it easier for us to replicate the error.
It is not specific to any G-code, but I will provide a G-code that causes my printer to crash while getting the file downloaded from Prusa Connect
[Shape-Cylinder_0.25n_0.15mm_PETG_MINIIS_4h31m.bgcode.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/13222800/Shape-Cylinder_0.25n_0.15mm_PETG_MINIIS_4h31m.bgcode.zip)

**Crash dump file**
  Please attach the crash dump file. This will make it easier for us to investigate the bug.
[dump_buddy.bin.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/13222795/dump_buddy.bin.zip)

**Video**
  Please attach a video. It usually helps to solve the problem.

ErGeeBee, 2023-11-01:
Same issue here.

dmyers7, 2023-11-02:
+1  Same thing.  Uploading via PrusaLink directly to the printer or through PrusaConnect, the file makes it to the printer.  When I go to start the print the printer immediately crashes to a Blue Screen of death.  Hitting the reset button the crash is detected and offers to write out the file.  The file was successfully written to the USB drive and is available to start the print after reboot.  I can upload several files via PrusaLink without any seeming issue.  It's only when I go to start the a print after an upload via the wifi connection (not using the ethernet at this point).  Printing several prints in a row works fine.  Only after an upload does it crash.  Alpha 2 firmware, Prusa mini.

elxris, 2023-11-02:
@dmyers7 please attach your dump file just in case is useful for the Prusa Devs, thanks!

MankeyJohn, 2023-11-02:
Same thing happened to me. Uploaded print and it crash with the .bin file below and .bGcode. Once the printer was turned off and back to normal I printed the file and it printed without any errors and correctly. 

[Crash Files.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/13243145/Crash.Files.zip)

Ronin0000, 2023-11-12:
same thing here

elxris, 2023-11-18:
Now working fine with `5.1.0-beta`. I'll test a few more days.