prusa3d/Prusa-Firmware-Buddy #3432: [BUG] Filament unload issue with new unload procedure on firmware 5.1.0

gegy1, 2023-11-01:
**MINI+**

**5.1.0-alpha2** 

**Original** 

**Filament Runout Sensor ✅** 

**Describe the bug**
Since the new method for unloading the filament in firmware 5.1.0, I have already had a problem unloading a few times. 
The unloading process starts and then stops shortly before it continues. At this moment there is a plug (blob) formation on the hotend side. When the filament is then discharged further, it gets stuck in the bowden tube. Then you have to loosen the tube on both sides, cut the filament and pull it out from the bowden tube with force

**How to reproduce**
 Just unload or change the filament. I uses Prusament PETG

**Expected behavior**
 Filament unload

gegy1, 2023-11-04:
I have this issue on every filament change. I habe attached a photo from just unloaded PLA. You can see the little hook, where the filament is minimal thicker. I gues this is because of moving the filament forward for pressing out the filament from the hotend and then begin pulling it back for unloading.
I did not had this issue on the mini using the firmware before version 5. 

[image]

fergymcferguson, 2023-11-08:
Same issue here - not every unload requires disconnect of the bowden, but 30-ish% of the time I do.

gegy1, 2023-11-20:
Issue still exist on Beta Firmware. Please, please bring back old unload procedure.
[image]

gegy1, 2023-11-22:
Changed today my noozle from a 0.25 back to a 0.4 and did not had the unload issue. 
I also changed the PTFE tube in the hotend. 
So could be an issue from the PTFE tube or the 0.25 nozzle. The nozzle has been bought from prusa e-shop.

gegy1, 2023-11-24:
I close this now, as i do not have the issue anymore.

closed 2023-11-24 by gegy1

twshield, 2024-02-11:
[image]

I just had this problem with 5.1.2 on a all stock Mini+. 

As you can see from the picture (Overture PETG clear) there is a blob on the end of the filament.  I had to remove the PTFE input tube to the extruder and take the filament out by cutting it before the runout sensor input and pulling through the tubing from the blob end.  I'm coming to 5.1.2 from 4.3 and that firmware on unload did a continuous movement ejecting the filament and I always made sure to immediately pull on the filament after it was no longer gripped by the extruder to get it the rest of the way out before the end cooled and would jam in the  tubing.  The method of unloading used by 5.1.2 has a pause after a short retraction and then a long retraction, this pause allows the end of the filament to cool off and be too hard to pull out through the PTFE extruder exit tubing.  

So, I prefer the old unloading method without the pause.