prusa3d/Prusa-Firmware-Buddy #3439: Delete gcode files from LCD screen

Claghorn, 2023-11-03:
Enter what type of printer or upgrade the enhancement applies to.
MK4 (though makes sense for others as well).

I print many one-of tests, I have no use for the gcode again. It would be very handy to be able to delete the gcode files directly from the LCD screen once I see it printed OK.

The printer usually lands on a screen with "Reprint" and "Home" buttons after a print. It would be most useful if there was a "Delete" button on that screen as well (but NOT selected as the default, "Home" should always be the default :-).

As it currently is, I have to either use PrusaLink or physically move the USB stick to my computer.

Gyrxiur, 2023-11-12:
I really want this too <3

Claghorn, 2024-06-04:
Still seems like a good idea to me.  I hereby un-stale it.

Gyrxiur, 2024-06-04:
Please add it <3

closed 2024-09-02 by github-actions[bot]