prusa3d/Prusa-Firmware-Buddy #3440: [FEATURE REQUEST] Abilty to move axis while print is paused

BaconFase, 2023-11-03:
There have been a few times while printing with the XL that I needed to pause a print to manually remove some filament buildup on the nozzle. However, once I pause the print the bed will lower and the extruder will move to the back left of the bed and there is no available option in the Tune menu to move the extruder for ease of access.

I know for a fact that the ability to move axis while a print is paused existed on the MK3S MMU2S, so I'm surprised it's not already a thing on the XL.

If it makes sense, enter what type of printer or upgrade the feature request applies to.
PrusaXL-5T

**Describe the expected functionality**
What I expect is the same Move Axis menu/ability, that already exists, but in the Tune menu while an ongoing print is paused.

DoubleStrike, 2023-12-14:
Frankly, we should have a full set of control options on pause including all temperatures, movement on all XYZ axes, and movement of extruder axes for all toolheads.  Pick/park tools would also be useful.

Dycus, 2024-02-20:
I would also like this feature, but moreover I would like the current tool to move to the very front and center when paused. Sometimes I get a buildup of filament on the nozzle and want to pause the print to clean it. The current park location in the back-left corner makes this very hard, depending on the current Z height and what is on the print bed.

sp847242, 2024-03-11:
> I would also like this feature, but moreover I would like the current tool to move to the very front and center when paused. Sometimes I get a buildup of filament on the nozzle and want to pause the [...]

Same - especially with PETG. Over the years, I've managed to reduce stringing, but it still happens. The strings gradually accumulate on the nozzle, and eventually the printer just drops a glob wherever it happens to fall, which can lead to either a crash, or it breaks a part off of the bed.

tobiashugo, 2024-04-01:
This is a must have feature, pause is pretty useless without the possibility to clean the nozzle.

wafflecart, 2024-04-08:
+1 for this feature, move it to the front and move bed down a bit for easier access on prints that are at the back of the build plate.

morphias2004, 2024-06-27:
Please don't let this go stale.

draeath, 2025-05-18:
It would be very nice if this was revisited.

adamcrume, 2025-05-25:
I would love to have this.  My printer just developed a little blob from the wipe tower, and I'd like to be able to pause the printer and clean the nozzle so the print can be saved.

BaconFase, 2026-09-28:
Still wanted. Please do