prusa3d/Prusa-Firmware-Buddy #3442: [BUG] PrusaLink x10 slower after upgrading to 5.0.1

michalmiddleton, 2023-11-04:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - [MK4]

**Printer firmware version** - [5.0.1]

**Original or Custom firmware** - [Original]

**Optional upgrades** - [n/a]

**USB drive or USB/Octoprint**
  Original USB drive

**Describe the bug**
  After upgrading to 5.0.1 (and new boot loader), the upload to printer is extremely slow. Speed probably around 1.5MB per minute. This is about 25kB/s. I power cycled the printer and still happening. I worked OK earlier today before the FW upgrade.

**How to reproduce**
  Send gcode from PrusaSlicer 2.6.1

**Expected behavior**
  Significantly faster upload.

**G-code**
[retromaker_ornament_globe3_0.25n_0.15mm_PLA_MK4IS_18h22m.gcode.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/13255452/retromaker_ornament_globe3_0.25n_0.15mm_PLA_MK4IS_18h22m.gcode.zip)

**Crash dump file**
 n/a

**Video**
n/a

lumen-novum, 2023-11-05:
My Prusa Mini doesn't have this issue on 5.1.0 alpha 2 (using ethernet). Was this an issue on 5.0.0?

rasor007, 2023-11-16:
Sane problem on my MK4 after upgrade

jzkmath, 2023-11-21:
Same here with my MK4. Its to the point that PrusaLink is practically useless for file upload. Had to revert to the old fashioned method of plugging the flash drive into my PC for gcode upload.

michalmiddleton, 2023-11-23:
I'm not currently having this problem anymore. I haven't changed anything since I reported this issue (no FW/software/environmental changes). Uploads seem to be back to somewhat normal speed now.

I am certain I restarted and power cycled the printer back before I reported this issue and it hasn't helped. But it seems to have fixed itself sometime along the way.

closed 2023-12-07 by michalmiddleton