prusa3d/Prusa-Firmware-Buddy #3444: [FEATURE REQUEST] Option to automatically unload filament when print is finished

beng27, 2023-11-04:
Please, before you create a new feature request, please make sure you searched in open and closed issues and couldn't find anything that matches. - DONE

**Is your feature request related to an existing problem? Please describe.**
I live in Wales and it is a very humid climate. Filament cannot be left on the printer or it absorbs moisture and becomes stringy. Every time I print, afterwards I unload the filament and put it away in its plastic bag. Normally I print overnight and collect my print in the morning. It is frustrating having to manually tell the printer to unload and then wait while it heats up for unloading.

**Describe the expected functionality**
A tick-box option in the slicer labelled 'unload filament when finished'. If the box is ticked, then at the end of printing the printer will unload the filament and then cool down as normal.

murk-sy, 2023-11-06:
This feature would probably work best as a M600 variable that can be set to unload only - this way the printer also knows that it doesn't actually have filament loaded.

But to be honest, I would probably kind of prefer the printer to unload out of the hotend (but not out of the extruder) by default.
This way,
- Ooze is avoided or at least lowered in terms of its volume (especially if using the newly developed MMU ramming)
- less messy for MK4/XL with nozzle based bed leveling
- Unload is always immediate
- When changing filament, the hotend heats up while you're switching spools, saving time

I don't know if there are any major downsides to heating up the hotend with filament in the cold end on a regular basis - the devs would probably know about that a bit more.

----

For your case, there are at least workarounds - Here's examples for the mini:
https://forum.prusa3d.com/forum/user-mods-octoprint-enclosures-nozzles/unload-filament-after-print/
```
G1 F600 E-30
M0 S5
G1 F4800 E-450
```
Note that this is not 100% accurate unload gcode (no ramming) which may cause some gunk buildup over the long term.

Here's also how it would fit into the Printer Settings>Custom G-code>End G-code using the current alpha version settings
```
G1 E-1 F2100 ; retract
{if max_layer_z < max_print_height}G1 Z{z_offset+min(max_layer_z+2, max_print_height)} F720 ; Move print head up{endif}
G1 X178 Y178 F4200 ; park print head
{if max_layer_z < max_print_height}G1 Z{z_offset+min(max_layer_z+30, max_print_height)} F720 ; Move print head further up{endif}
;Unload nozzle start
G1 F600 E-30
M0 S5
G1 F4800 E-450
;Unload nozzle end
G4 ; wait
M104 S0 ; turn off temperature
M140 S0 ; turn off heatbed
M107 ; turn off fan
M221 S100 ; reset flow
M572 S0 ; reset PA
M569 S1 X Y ; reset to stealthchop for X Y
M84 ; disable motors
; max_layer_z = [max_layer_z]
```

codingcatgirl, 2023-11-10:
According to the Prusa docs, M702 should unload your filament. You can try adding that to your custom end Gcode.

FVerbaas, 2024-02-02:
The question of @beng27 is not solved by 'adding something to the gcode'. His request, which I second, is to have a checkbox on the print initiation screen to specify you want the printer to unload the filament, because this particular job will be the last one with this filament, or for the day, and you want to come back after completion and put the spool back in the dry box. Whether that is done via gcodes or not is immaterial. 
Yes, one can make a gcode file with M702 and schedule that after the current print bu then of course one has to give a 'set ready' for that job, which makes it slightly less work but does not reduce the waiting time.

bmirwin89, 2024-04-20:
I came here to request this as well.  I waste a ton of time waiting for the XL to unload all 5 toolheads.  It would be nice to have the option to have it do it automatically while it's still hot after a print.

u89djt, 2024-04-25:
Adding another voice to this request. Could we eject each filament as soon as it finishes its last extrusion rather than wait till the end of the whole print? I don't know if the filament runout logic would get in the way if filament is pulled out while the print is running.

murk-sy, 2024-06-27:
Bumping to keep open

Niflheim123, 2024-08-13:
I would also like this feature, even better it would be trough prusa connect or the prusa app.

awesomekyle, 2024-08-16:
I want this feature so I can change nozzles easily. Not having to hot tighten the nozzle is great, but having to heat up the nozzle, unload filament, and wait for cooldown just to change the nozzle diminishes some of the advantage of the new system. I think this is even more important now that Prusa has high flow nozzles. I may want to go from a precision 0.4 nozzle to a high flow 0.6 nozzle for rough work or prototyping, and skipping the unload step would be great.

3DPrintMunich, 2024-09-17:
This feature would be great! Changing filament with the AMS is much more comfortable..

CZDanol, 2024-09-17:
Hello, I've created an internal ticket for this: BFW-6078

AllVillainsSmile, 2024-12-28:
As @codingcatgirl suggested, there is M702 command you can use.
This: [https://www.reddit.com/r/prusa3d/comments/18hjp6e/mk4_endofprint_filament_ejection_gcode/](https://www.reddit.com/r/prusa3d/comments/18hjp6e/mk4_endofprint_filament_ejection_gcode/) by Prusa3D Team shows how you can use a trigger word in Notes to set the printer to unload - closest thing to checkbox I could find :)

FVerbaas, 2024-12-28:
Again, this is all juggling GCodes, and whether you want to unload the filament after the print is completed, that's an operational thing, governed by my further schedule for the day. It is not related to the design being printed, and therefore it should not be in the gcode. 
The fact that gcodes exist that can trigger a filament unload does not necessarily mean that should be the only method available. 
I am just a user who bought a pretty expensive printer and who wants to have the option to let the printer unload the filament when the job is done.

murk-sy, 2025-04-02:
Core One now has this feature implemented in [v6.3.1](https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v6.3.1)

From release notes:
> Note: This is a pilot implementation of the procedure. If it proves effective, we plan to extend it to other printers as well.

beng27, 2025-04-02:
Awesome, thank-you. Now I just need to be patient for my CoreOne upgrade kit to arrive :-)

CZDanol, 2025-04-03:
Just to be clear, the retraction in 6.3.1 doesn't unload fully, just out of the nozzle. But I'd argue that's even better, because the printer can automatically reload the filament. And when the user presses "unload", the printer doesn't need to heat up, so it's very fast.

murk-sy, 2025-08-06:
Having it done manually in gcode wouldn't make it function in in the filament load/unload menu, unfortunately. You'd have to use your own load/unload routines and at that point it's probably more inconvenient than just doing a normal unload.

michaelruml, 2026-10-04:
Second what @girak49 said. 
@Prusa-Support The auto-retract feature does not solve this use case. The OP (as am I) is asking for a full unload after the print finishes, while the nozzles are still hot.