prusa3d/Prusa-Firmware-Buddy #3447: [ENHANCEMENT] Remove confirmation'Continue' when loading filament

Belott, 2023-11-05:
Enter what type of printer or upgrade the enhancement applies to.
Original Prusa MK4, Firmware 5.0

**Is your suggested improvement related to an existing problem? Please describe.**
No

**Describe the expected functionality**
To load a filament, the current process is as follows:
Step 1: Menu Filament/Load Filament
Step 2: Select filament type (e.g. PLA)
Step 3: Insert filament (The sensor registers the inserted filament)
Step 4: User confirms that the filament has been inserted by selecting "Continue".
Step 5: The printer extrudes filament.
Step 6: User confirmation of correct filament color.

In this process, step 4 is unnecessary. The filament sensor already registers that the filament has been loaded. Step 4 should be removed. (Step 4 is also not available on the Prusa MK3s. So why should it be necessary on the MK4?)

AJErazzor, 2023-11-09:
On a bowden it is possible to have filament pass the sensor but not be at the gears

Belott, 2023-11-10:
I am of the opinion that the original Prusa MK4 is delivered without a Bowden extruder and the sensors and parameters should therefore be adjusted to the original condition. For this reason, the filament loading process should be tailored to the standard version with a correctly functioning filament sensor.

blue0ceanby, 2023-11-18:
Suggestion regarding my way of working is
that the item could be selected in the menu
a: with charging when detected
b: load with confirmation

If b is selected, a button load interrupt appears on the rotary knob menu

anghelos, 2023-12-20:
To add to this, there are inconsistencies with the other way of loading filament (MK4):

Step 1: Insert filament (The sensor registers the inserted filament)
Step 2: The printer partly loads the filament.
Step 3: The printer asks for filament type.
Step 4: User confirmation of correct filament color.

Note that the "Continue" confirmation is absent in this workflow. It appears however when changing filaments.

closed 2024-06-27 by github-actions[bot]