prusa3d/Prusa-Firmware-Buddy #3450: [BUG] Watchdog reset on MK4 while printer is idle

BRFreshness, 2023-11-06:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - 
Original Prusa MK4

**Printer firmware version** - 
5.0.1

**Original or Custom firmware** - 
Original

**Optional upgrades** - 
none

**USB drive or USB/Octoprint**
Octoprint is connected, however no print job was running.

**Describe the bug**
Since upgrading the firmware to 5.0.0 I've been seeing the printer crash (Watchdog reset) for no apparent reason. It was just sitting idle. I've updated the firmware to 5.0.1 but it crashed again. Not sure if it's precisely the same crash signature - did not save the old dump files.

**How to reproduce**
Turn on printer... wait. Eventually it has a watchdog reset crash. Within 24 hours, I think.

**Expected behavior**
Printer shouldn't crash randomly.

**G-code**
n/a - was not printing anything

**Crash dump file**
[dump_buddy.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/13263286/dump_buddy.zip)

**Video**
[image]

BRFreshness, 2023-11-07:
Disconnected the USB cable for Octoprint and left the printer alone to idle today. As expected, it crashed again. No print job was running. New dump file attached.
[dump_buddy.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/13276781/dump_buddy.zip)

7s3, 2023-11-12:
This happened the me 3-5 times over the course of the past week.
MK4 version 5.0.0 and 5.0.1

bdynamic, 2023-11-26:
Same with me. FW Version 5.1
[image]
[dump_buddy.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/13467972/dump_buddy.zip)

Mackalope, 2024-01-25:
Same here idle BSOD every time

closed 2024-06-27 by github-actions[bot]