prusa3d/Prusa-Firmware-Buddy #3455: [FEATURE REQUEST] Cancel object for mini+

ErGeeBee, 2023-11-07:
**Printer type** - [MINI+]
**Optional upgrades** - [Filament Runout Sensor]

**Describe the expected functionality**
Integration of the  'cancel object function' on the Mini+

Ronin0000, 2023-11-10:
This can't be done because of limited memory or ram on the mini. Only certain upgraded will be able to be implemented in the mini's firmware.

ErGeeBee, 2023-11-10:
> Only certain upgraded will be able to be implemented in the mini's firmware.
What exactly do you mean by this? Thanks.

Ronin0000, 2023-11-12:
It means that because of the limited space only certain features that they will add on the MK4 or XL will be available on the Mini because of the limited space. For example the faster file transfer speeds was added to the mini because of the MK4 and XL, but the  Cancel object feature was not. They will probably only pick the most useful upgrades to add to the mini's firmware.

Tommy544, 2023-11-13:
Let's not jump to conclusions. This is a very useful feature and my to my understanding from the posted blog post, this feature was supposed to be available on all 32 bit printers. I hope it was just erroneously left out of this MINI alpha firmware and will be added in the following alpha/beta builds.

AJErazzor, 2023-11-15:
Could the mini USB be used as a way of "adding memory" - write out whatever can be to a USB drive and free up internal memory

ErGeeBee, 2023-11-24:
It was added om the stable 5.1.0 release

ErGeeBee, 2023-11-24:
Actually I can't find the function in the Mini's menu. So therefore the re-opening.
is it only for XL & MK4 as described in the release notes for the 5.1.0 alpha2?

kozross, 2023-12-15:
[Latest firmware release](https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v5.1.2) should fix this?

jerrygrey, 2023-12-15:
@kozross I'll test now

jerrygrey, 2023-12-15:
> [Latest firmware release](https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v5.1.2) should fix this?

Can confirm this has been fixed in 5.1.2

ErGeeBee, 2023-12-16:
> > [Latest firmware release](https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v5.1.2) should fix this?
> 
> Can confirm this has been fixed in 5.1.2

Same here. I’ll close the issue. Thanks.

closed 2023-12-16 by ErGeeBee