prusa3d/Prusa-Firmware-Buddy #3457: [FEATURE REQUEST] bed leveling wizard

gigi84, 2023-11-08:
**Printer type** - [MK4]

I would like to have the bed leveling wizard (like octoprint) using PrusaLink or PrusaConnect

gigi84, 2023-11-18:
There are 2 ways to implement this feature:
1. (easier) --> just to report the actual nozzle height for the 9 points where the plate screws are. Based on the results, the user will decide autonomously what to do (add/remove washers and/or tight/loose screws)
2. (more difficult) --> Based on the actual nozzle height (see point 1), the screen will suggest the user what to do, screw by screw (9 times)

carlocarlo91, 2023-12-01:
It would be great to have it on MINI+ too.
Something similar to Bed Level Visualizer of Octoprint.

closed 2024-06-27 by github-actions[bot]