prusa3d/Prusa-Firmware-Buddy #3461: [BUG] FW 5.1.0-alpha2 - Unable to print. Print suddenly stops.

antimix, 2023-11-09:
### Description of the bug

**PRINTER:**  MK4
**FIRMWARE:** 5.1.0-alpha2
**SLICER:** PrusaSlicer 2.7.0-alpha1

The print suddenly stop/crashed at layer 1 after few shapes. The printer looked like it was  printing on the LCD, but in reality it was doing nothing, head stopped, no movements, no messages special messages on the LCD, the scrolling printing as usual...

After further investigation the result was:
- the gcode that was sent to the printer via wifi with PrusaSlicer was correct. A binary compare with the one on the PC confirmed it was sent correctly.
- However going in step by step nozzle emulation on PrusaSlicer gcode, showed that at the precise point where it stopped. 
- No idea why the offending line stops the print. May be it was badly uncompressed in the printer memory.
- Offending line: 476 
- I did not have any issue in weeks with other print on the MK3S with this PrusaSlicer version, so the issue may be is affecting the MK4 FW or the decompression.

Attached the BGCODE that hand after one minute, and the PROJECT (is the MMU3 spool buffer).

[BUFFER_PART1_0.4n_0.2mm_PETG_MK4IS_3h23m.zip](https://github.com/prusa3d/PrusaSlicer/files/13307245/BUFFER_PART1_0.4n_0.2mm_PETG_MK4IS_3h23m.zip)
[image]

### Project file & How to reproduce
A Zip with the prusaSlicer Project and the .bgcode

1) Open the project
2) Slice
3) Print

or 
Just send the .bgocde to the printer.

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.7.0-alpha1

### Operating system

Windows 10

### Printer model

MK4

antimix, 2023-11-09:
I reprinted the file after having uncompressed it with the PrusaSlicer Gcode Viewer, and it worked without issues.

blue0ceanby, 2023-11-10:
I can confirm that. I had the same behavior under the exact same constellation.
I currently have the printer FW back to 5.0.0.
Can we test something?

blue0ceanby, 2023-11-19:
This morning I completely reset the print to factory settings without firmware and then used the 5.1.0 beta. Then everything was put back into operation. The printer has currently been running with 10 prints for 12 hours without any problems

closed 2024-06-26 by github-actions[bot]