prusa3d/Prusa-Firmware-Buddy #3463: [BUG] MK4 crashes with precise_stepping.cpp Negative step time error when printing specific gcode file

welnanick, 2023-11-11:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - [MK4]

**Printer firmware version** - [5.1.0-alpha2+13126]

**Original or Custom firmware** - [Original]

**Optional upgrades** - [None]

**USB drive or USB/Octoprint** - [USB drive]

**Describe the bug**
 Printer suddenly crashes and reboots with the error "Negative step time" coming from withing precise_stepping.cpp when printing a specific gcode file (attached). It doesn't always stop in the same place, but I've attempted to print this file a few times and it always crashes somewhere within the first 5 to 10 layers of the print.

**How to reproduce**
Attempt to print attached gcode file

**Expected behavior**
Print completes successfully

**G-code**
[gcode.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/13324028/gcode.zip)

**Crash dump file**
[dump_buddy.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/13324042/dump_buddy.zip)

**Video**
https://photos.app.goo.gl/4u59rfxxSoYwynNCA

GuyH77, 2023-11-11:
I have had the same error on the XL 5T. I reprinted the gcode and it repeated at what looks like the same place. 

[image]
[image]

3mf project 2.6.1 IS beta
[TrashCan.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/13325381/TrashCan.zip)
Dump file
[dump_buddy.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/13325383/dump_buddy.zip)

vectorstofinal, 2023-11-12:
I just got this same crash on my 2-tool XL as well.
[Body1_0.6n_0.32mm_PLA_XL_1h53m.bgcode.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/13328658/Body1_0.6n_0.32mm_PLA_XL_1h53m.bgcode.zip)
[image]

nschimke, 2023-11-18:
Just happened on my 5-tool XL, firmware 5.1.0-beta, PS 2.7.0-rc1 on a Mac.
[image]
[dump_buddy.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/13401264/dump_buddy.zip)

GuyH77, 2023-11-18:
@nschimke Was this a print using an older 3mf project? I have a theory that there's something in the settings of older projects that the Xl FW doesn't like which causes this error. The project may help the developers reproduce the issue.

nschimke, 2023-11-18:
> @nschimke Was this a print using an older 3mf project? I have a theory that there's something in the settings of older projects that the Xl FW doesn't like which causes this error. The project may help the developers reproduce the issue.

It might have been but I don't think it was. This was a part I printed months ago but I'm pretty sure I did it with a new scratch project (that I didn't save).

ishaid, 2024-01-25:
I just had the same error on my MK4.

[image]

[dump_buddy and bgcode.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/14048041/dump_buddy.and.bgcode.zip)

new project new step files.

closed 2024-06-26 by github-actions[bot]