prusa3d/Prusa-Firmware-Buddy #3468: [ENHANCEMENT] Add support for 0.2mm and 1.2mm nozzle sizes

nazar-pc, 2023-11-12:
Please, before you create a new feature request, please make sure you searched in open and closed issues and couldn't find anything that matches.

Enter what type of printer or upgrade the enhancement applies to.
**Printer type** - XL
**Optional upgrades** - None

**Is your suggested improvement related to an existing problem? Please describe.**
Currently nozzle size is limited to 0.25-1.0mm range, while I have 0.2mm and 1.2mm nozzles that I'd like to try. I was able to print successfully with 0.2mm nozzle while configuring 0.25mm on the printer, but it complains that gcode doesn't match installed nozzle size.

**Describe the expected functionality**
I'd like to see nozzle size range increased in future firmware versions, probably to 0.2-1.2mm range.

nazar-pc, 2023-11-27:
Those are independent and I have already opened https://github.com/prusa3d/PrusaSlicer-settings/issues/220

Even if I create a slicer profile with a smaller/bigger nozzle size, printer will still complain about mismatch because I can't select a proper size in the firmware, which this issue is requesting to change.

MichaelJLew, 2023-11-27:
Oh, I see. Sorry.

danopernis, 2024-07-08:
This is scheduled for next minor (not patch) version of the firmware.

danopernis, 2024-08-22:
New version of the firmware 6.1.2 should contain the fix, can you test that?

nazar-pc, 2024-08-22:
I have no idea where the option is anymore. It was under Settings->Tools->Tool 1 in 6.0.4 firmware, but now there is no nozzle size option there and I don't see it in any other menus :man_shrugging:

bkerler, 2024-08-24:
@nazar-pc Should be under Settings->Hardware->Nozzle Diameter. In my test, the firmware 6.1.2 seems to work fine, nozzle size up to 1.2 mm is now supported.

nazar-pc, 2024-08-24:
Probably true on single-tool XL, but I recently upgraded mine to dual and nozzle diameter is in a different menu there (for whatever reason) and it is not there anymore, see https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/4151

I have no reason to believe it wouldn't be there if menu wasn't broken, so I guess the issue can be closed as resolved, but it didn't help me much just yet.

nazar-pc, 2024-09-06:
Confirm as fixed on dual-head XL in 6.1.3, thanks!