prusa3d/Prusa-Firmware-Buddy #3469: [BUG] OutOfMemory Mini+ 5.1.0-alpha2

mtrakal, 2023-11-13:
**Printer type** - MINI+

**Printer firmware version** - 5.1.0-alpha2

**Original or Custom firmware** - Original

**Optional upgrades** - none

**USB drive or USB/Octoprint** USB flash drive upload over PrusaLink/Connect

**Describe the bug**
While upload file from Prusa Slicer 2.7.0 alpha1 over PrusaLink/PrusaConnect to Mini+ (5.1.0-alpha2) it crash on OOM.

**How to reproduce**
  Don't happen second time. Turn off and on printer and print start without any issue.

**Expected behavior**
Don't crash on OOM

**G-code**
[Small_spiral_Christmas_Ball_with_Supports_-_Remesh_0.4n_0.15mm_PLA_MINI_4h10m.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/13332546/Small_spiral_Christmas_Ball_with_Supports_-_Remesh_0.4n_0.15mm_PLA_MINI_4h10m.zip)

**Crash dump file**
[dump_buddy.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/13332526/dump_buddy.zip)

Ronin0000, 2023-11-13:
Same Issue here.

thijstriemstra, 2023-11-13:
Just curious and don't want to download the zip right now, but how big was the gcode file?

closed 2024-06-26 by github-actions[bot]