prusa3d/Prusa-Firmware-Buddy #3472: [BUG] Printer Hangs On Black Screen After Manual MMU Print Started on Firmware 5.1.0 Alpha 2

oakbuffalo13676, 2023-11-13:
**Printer type** - [MK4]

**Printer firmware version** - [5.1.0 Alpha 2]

**Original or Custom firmware** - [Original]

**Optional upgrades** - [None]

**USB drive**

**Describe the bug**
A 3d print was started using the traditional manual MMU settings that worked as of MK4 firmware 5.0.1 and PrusaSlicer 2.6.1, but instead using 5.1.0 Alpha 2 with PrusaSlicer 2.7.0 Beta 1. Downgrading to aforementioned inferior revisions solves the issue, but using the latest firmware and software causes the MK4 screen to hang on a black screen after a print is started. The heatbed light turns on, all of the fans are spinning as appropriate, but nothing happens. Different USB tried, tried reslicing the model, etc. Using stock printer settings (except the manual tool change GCODE of M600 for the color change).

**How to reproduce**
 Add a model with text or something to have it at least two colors. Using stock 0.4n printer settings, modify extruder to >= 2 colors, then custom GCODE --> Tool change GCODE --> Add M600. Print settings --> Multiple extruders --> enable wipe tower.  Slice model with using two colors. Start the print in the printer, and the screen went black and unresponsive 100% of the time.

**Expected behavior**
The print to start probing the bed to start the bed, but never does.

**G-code**
[2h51m; Pet Hair Sign MMU; 0.4n; PLA,PLA,PLA.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/13342751/2h51m.Pet.Hair.Sign.MMU.0.4n.PLA.PLA.PLA.zip)

**Crash dump file**
 Doesn't crash - just hangs until the reset button is pushed.

**Video**
In progress... please stand by.

sgaon42, 2023-11-16:
I have the same bug.

magnushacker, 2023-11-29:
Same issue reported here: https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/3525

You can use prusalink and start the print from there. It's only when you start from the printer menu that you get the black screen.

Rebelpine, 2024-01-21:
Seems to still be happening: MK4 with firmware 5.1.2 and Prusaslicer 2.7.1.

bitcoin3us, 2024-02-17:
I've got the same issue.
How to reproduce...
Add a model with text or something to have it at least two colors. Using stock 0.4n printer settings, modify extruder to >= 2 colors, then custom GCODE --> Tool change GCODE --> Add M600. Print settings --> Multiple extruders. Slice model with using two colors. Start the print in the printer, and the screen went black and unresponsive 100% of the time.

ScrlScrl, 2024-04-07:
I have the same issue. Screen goes blank when starting a print manual color change print from printer (two virtual extruders, M600 on tool change). Starting print is possible when started from PrusaLink. 

Happens on PrusaSlicer 2.7.4 and firmware 5.1.2.

lambdahindiii, 2024-04-07:
Having the same issue today. Also on PrusaSlicer 2.7.4 and firmware 5.1.2 with settings exactly as described by @ScrlScrl