prusa3d/Prusa-Firmware-Buddy #3476: [FEATURE REQUEST] Safety: add a way to detect disconnected loadcell

Tupson444, 2023-11-14:
Printer type: MK4

I wondered what would happen if the loadcell cable gets damaged or disconnected so I decided to test that by disconnecting it.

Normally the loadcell reading on my printer is around 15270. When I disconnected it, it was around -26570. I lifted the print head enough and selected autohome. The printer homed X & Y then started homing Z, so I reset the printer to avoid damage.

Conclusion: if loadcell cable is disconnected by accident (or damaged), the printer will probably damage the buildplate and hotend because it won't sense any change in the loadcell.

There should be some way to detect that something is wrong with the loadcell and prevent Z homing. I don't know how, since the normal "idle" reading can vary (probably different from printer to printer and I think it's not always the same value on mine), therefore I don't see a way the printer would know that the reading is false when disconnected, but there may be a way to solve this potential problem.

tjhawkey, 2023-11-18:
This is a great suggestion! I hope there is a programmatic way to do it. From a mechanical view, I think it's fishy that the top of the nozzle pushes against the filament guide near the gears. Seems to me that the there should only be one load path for consistent behavior

Tupson444, 2024-06-04:
Commenting to prevent closing.

danopernis, 2024-08-06:
Closing in favor of #4106

closed 2024-08-06 by danopernis