prusa3d/Prusa-Firmware-Buddy #3478: [ENHANCEMENT] Allow automatic offset tool calibration per toolhead

sassda, 2023-11-14:
On a Prusa XL 5th
My printer came with a faulty screw threads for the 2nd extruder, and while I work that out with support, it would be great to be able to do the offset tool calibration only for some toolheads. Specially as this is already possible for things like dock position or other tests, that would allow me to more easily use the machine as a 4th in the meantime. Longer term I can see this being useful if you have a breakdown for a particular extruder and need to recalibrate.
Thanks!

r3Fuze, 2024-01-22:
I agree that this is an important feature.

I just swapped the nozzle on a single tool and wanted to re-run the tool offset calibration for that tool, but you can only do all 5 tools at once.

I ended up not doing the calibration because all 5 tools need to be perfectly clean and I couldn't be bothered thoroughly cleaning them all since they keep oozing filament while heated and I don't own a brass brush yet.

edsai, 2024-03-21:
Would be lovely if this enhancement were made making nozzle changes much quicker on the XL.

eltron247, 2024-04-06:
+1 
This would be great. No need to go through a full 5 heads if I'm only swapping 1 nozzle.

Dycus, 2024-08-27:
Please, oh please implement this!
Today I swapped my tool 5 nozzle from 0.4mm to a brand-new 0.6mm. No cleaning or prep was needed at all for this new nozzle because it's already clean. I was hoping I could just go into the menu and hit "calibrate tool 5".

But instead I had to unload filament from all 4 other tools (else it would keep oozing when heated for cleaning), then thoroughly clean each one so they could be unnecessarily re-calibrated, just so I could calibrate the one tool that was changed. Then I had to reload all my filaments. It turned a 10-minute job into a 1-hour job.

I'm definitely going to avoid changing my nozzles as much as possible if that's the workflow every time. :(

benjaminbrumbaugh, 2024-09-16:
This just got me too. Calibrating all the nozzles when you do a swap means heating up and cleaning every toolhead and going through an extended calibration. A task that should take 30 seconds is taking over a half hour.

svake, 2024-10-08:
I also would need this feature, because changing only one nozzle (e.g. brass to steel) mean cleaning all nozzles and then calibrating all new. This takes long and if the cleaning is not proper, then I change good values of nozzles that worked so far.

bkerler, 2025-02-08:
I wrote a PR which is untested right now that should implement this functionality.

edsai, 2025-05-12:
> I wrote a PR which is untested right now that should implement this functionality.

Any news?

bkerler, 2025-05-28:
Actually there were issues with implementing the PR that need to be resolved first, that's why it is in draft mode currently.

edsai, 2025-10-06:
This issue is important. Bananas that there has been no movement on it.

3d-gussner, 2025-10-06:
I will assign this issue to myself to prevent stale-bot.

Gik-0101, 2025-12-12:
+1

That would be a huge relief and a useful feature!

edsai, 2026-09-28:
It's absolutely sad that something this basic continues to be ignored by Prusa.

rmpel, 2026-09-28:
It's now even less probable this will get picked up as a lot of XL users already have or will upgrade to XL+ with a completely hands-off calibration that does not require absolutely clean nozzles.