prusa3d/Prusa-Firmware-Buddy #3480: [BUG] XL Fan Test Inconsistent

JeremiahGillis, 2023-11-16:
**Printer type** - [XL Five-Head]

**Printer firmware version** - [5.1.0-alpha2]

**Original or Custom firmware** - [Original]

**Describe the bug**
The fan test produces inconsistent results. I tried this test 15 times and it would not pass. The fan test would tell me different fans failed each time.

A quick Google search shows a number of users with the same problem even on released firmware. Some users take the fan shroud off to get this test to pass.

**How to reproduce**
Power on printer, navigate to the fan test, and run it. In my specific case, this menu came up at power on because I just unboxed my five-head assembled machine, updated the firmware, then went through the printer setup. I ended up skipping this test, but the printer prompted that I had an unfinish test multiple times during the rest of the calibration.

**Expected behavior**
The fan test passes since all 10 fans were functioning visually. Or the fan test fails correctly when a fan is not functional.

This isn't a great experience when powering your machine up for the first time and you cannot get past the first test due to bugs.

JeremiahGillis, 2023-11-16:
I downgraded to 4.7.5. My fan test failed the first time. I tried a second time and it passed.

brettvitaz, 2023-11-17:
FWIW, I had the same problem on 4.7. Prusa Support said it would be fixed in a future firmware.

jmsether, 2023-11-21:
If you want to get it to pass while they are fixing it in the firmware you can remove all the part cooling fan shrouds and then run the test. I was getting this same error and removing all the fan shrouds allowed me to pass so I could get rid of that annoying message to complete calibration.

daavery, 2023-11-25:
Fan tests are air density sensitive.  At high altitudes the fans spin faster.  I'm at 8200 ft (2500 m) and full speed print fan runs at 7200 rpm which fails the fan test

M106 P0 S255
M123
 E0:0 RPM PRN1:7184 RPM E0@:0 PRN1@:255
 E0:0 RPM PRN1:7184 RPM E0@:0 PRN1@:255

JeremiahGillis, 2023-11-25:
> Fan tests are air density sensitive. At high altitudes the fans spin faster. I'm at 8200 ft (2500 m) and full speed print fan runs at 7200 rpm which fails the fan test
> 
> M106 P0 S255 M123 E0:0 RPM PRN1:7184 RPM E0@:0 PRN1@:255 E0:0 RPM PRN1:7184 RPM E0@:0 PRN1@:255

That is a great point. Prusa needs to consider the density altitude, not just the Mean Sea Level Altitude.

eficker, 2024-03-05:
I can confirm that downgrading to 4.7.5 worked on the first fan test. I was getting checkered results on 5. Then flashed back up and no further issues. Not ideal, but neither is going to the basement to click the dial for every print. FYI, removing the shrouds did not change my results on 5.

deniolan, 2024-06-02:
I have submitted a pull request to address this issue.   Currently the fan maximum on self tests is 6799.  I am one of the users who has shown that the fan speed actually goes higher than that (mine hovers in the mid 6800 range).  I have increased the "max" to 7,000 which should remove this error for the majority of users encountering this issue that I have seen.

deniolan, 2024-06-02:
> [deniolan](/deniolan)

Correction...  I had to put the limit down to 6899 (100 rpm over current).  I know for several of the users (myself included) this will fix the issue.  But there is an assertion in the code that the print fan range cannot overlap the heatbreak range - so for me to increase the print fan max to 7000, I would need to increase the heatbreak minimum from 6800 to 7000 which may have adverse effects.   I will need to test more before submitting another PR.   Hoping this one gets a majority resolved.

daavery, 2024-06-02:
Note the debug data I included. At 8200 ft PA the print fan hits 7200 rpm

daavery, 2024-06-02:
Maybe change the swap detection to relative speeds vs hard ranges

deniolan, 2024-06-02:
> Maybe change the swap detection to relative speeds vs hard ranges

Yup.  This is why I was mentioning that there needs to be a more comprehensive change.  I’ll look into it - this is my first time cracking open the firmware so I need to understand some other pieces prior to making a bigger change.  From data in the forums that I saw - just bumping from 6799 to 6899 will fix for a significant number of users.

deniolan, 2024-06-03:
> Maybe change the swap detection to relative speeds vs hard ranges

I closed my previous pull request and submitted a new one this evening that has the notion of a "minimum overlap".   So in the case of the XL, where the print fan maximum can overlap the heatbreak minimum an overlap can be applied where the switched fan detection basically says "both fans are operating in the same range so I can't detect whether they are switched or not".

This allows me to push the print fan maximum to 7200 RPM whilst leaving the heatbreak fan range alone, applying an overlap range on the switched fan check.

We'll see if they approve it - it should work fine, and honestly - it would be a real world scenario in some cases where the fans are spinning in the same range and you can't reliably say whether they are switched or not in that scenario.

Cheers.

danopernis, 2024-06-03:
@deniolan thanks for your effort in https://github.com/prusa3d/Prusa-Firmware-Buddy/pull/4026 as you already know, the fix is not as easy if we want to retain the "switched print fan" check. Since XL comes with [preassembled extruder](https://help.prusa3d.com/guide/5-extruder-accessories-assembly_422991) maybe this check could be completely disabled for XL and we can just adjust ranges :thinking:

deniolan, 2024-06-03:
> @deniolan thanks for your effort in #4026 as you already know, the fix is not as easy if we want to retain the "switched print fan" check. Since XL comes with [preassembled [...]

See my prior comment.   Agreed.  I actually made comments as such in my Pull request.   With the overlap I have "enabled" in this PR, Mini/MK4/MK3 functionality is not changed but there is a small window (500 rpm in the pull request) whereby switched fan detection simply would not be able to determine that the fan connections are reversed.   I feel like this latest PR is a good compromise that would still allow some switched fan detection - because honestly - if someone disassembles the extruder at some point, their fans are likely to be either broken in and spinning more freely, or broken and not spinning at all...  in either of those cases, the small window of "allowed" overlap will be inconsequential.

deniolan, 2024-10-17:
Submitted updated [PR-4258](https://github.com/prusa3d/Prusa-Firmware-Buddy/pull/4258)

CZDanol, 2024-10-17:
No need for PRs - we've fixed this in our codebase and it will be released in the next 6.2.0 candidate

crabdancing, 2024-11-07:
@CZDanol Where is the code? I'd like to compile it to flash on our printer now, if possible. :)

CZDanol, 2024-11-07:
We publish our code together with the binaries, so it is not publicly available yet.

deniolan, 2024-12-31:
> We publish our code together with the binaries, so it is not publicly available yet, sorry :(

*BUMP* - this is a huge issue in my workflow.  Any time I want to flash firmware and factory reset the printer, this is a giant pain.  It's persisted for over a year now.  Please release the binaries or accept a community solution - simple, really.   This isn't rocket science, you're pushing all kinds of new fixes out that break functionality.   For all intents and purposes you could just disable this fan check on the XL and call it a day since most of the printers have fully assembled print heads and Prusa themselves would be responsible for a reversed fan situation.

Just my .02.

Bob

danopernis, 2024-12-31:
@deniolan trust me, this pains me as a developer as much as it pains you. If you look at the [releases](https://github.com/prusa3d/Prusa-Firmware-Buddy/tags) there should have already been 6.2.0 beta months ago and we should have released final before Christmas.

We developers are unfortunately fully consumed by CORE One, as is the testing department. It might seem like an excuse but there is just nothing I can personally do about it.

Just to keep you updated, the solution to your problem which we as developers went for is actually the simplest one, to just disable the switched-fan-selftest for XL as the heads come fully assembled. The testing department is concerned about the printer safety though so there's unfortunately still a debate as to what the final solution will be. Like I said, I am just as frustrated as you are.

deniolan, 2024-12-31:
> @deniolan trust me, this pains me as a developer as much as it pains you. If you look at the [releases](https://github.com/prusa3d/Prusa-Firmware-Buddy/tags) there should have already been 6.2.0 [...]

Appreciate the reply... but, unfortunately, it is an excuse.

I know you can't do anything about it - however:

1. Core One is a fifth of the cost of a fully decked out XL
2. Community has provided viable solutions - my solution allows the fan check to continue to work with an overlap range where it does not.  There is no other solution.   You either disable it, or you allow overlap.

I totally appreciate your response... but prusa shoving aside the folks who bought $5000 printers for those who bought the latest $1000 whizbang is irresponsible at best, in my opinion.

JeremiahGillis, 2024-12-31:
Thanks for the update. Hopefully the developers do share some of our sentiment back with management. This issue was a big disappointment to me because it was the first test presented to me when powering up a 5H semi-assembled machine. Spending a bunch of time trying to get the first test to pass on a expensive machine that supposedly was assembled and tested at the factory it not a great first experience.

This issue should have been addressed early on, even disabling the test temporarily until a correct fix could be implemented. A quick search on the internet reveals that the occurrence level is pretty high now. We're at a year and nine months since release. All self tests should be consistent at this point.

I still have to run alpha firmware to have a working machine due to the USB drive errors. Management needs to get some resources back on the XL firmware until we have released firmware that in a good state. The firmware is almost there in my opinion.

CZDanol, 2025-01-06:
The fan tests will be adjusted in the next upcoming 6.2 release candidate.

CZDanol, 2025-01-09:
Hello, we've just released https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v6.2.0-RC1 :) 

Closing the issue.

daavery, 2025-01-10:
This is NOT fixed in 6.2.0-RC1.  the print fan limit still needs to be raised to at least 7200 rpm to pass at higher altitudes.  All 5 fans still fail in 6.2.0-RC1.

CZDanol, 2025-01-10:
@daavery okay, created another ticket for that: BFW-6598

And reopening.

daavery, 2025-02-07:
> @daavery okay, created another ticket for that: BFW-6598
> 
> 
> 
> And reopening.

Just tested with 6.2.0.   Print fan speed s255 is now close to 7500 rpm

daavery, 2025-02-07:
And still fails self test at high altitude

Prusa-Support, 2025-04-26:
The fan range may slightly vary for several reasons. However, the RPM value tends to stabilize within the expected range over rather short but continued operation.
_Please clean the fan with canned air and try again_ - **avoid** high-pressure compressor machine!
For reference, you can find the current print fan test conditions and the RPM expected range [here in the source code](https://github.com/prusa3d/Prusa-Firmware-Buddy/blob/818d812f954802903ea0ff39bf44376fb0b35dd2/src/common/selftest/selftest_fans_config.hpp).

If the fan is only slightly above the maximum range (nearly 7000 RPM or only slightly higher), it may actually be a temporary condition due to debris or long inactivity, but you can also check the fan without the shroud to check it in the lower range.
- Please _remove the **print fan shroud**_ and repeat the test.
  - If successful, you may want to carefully inspect that fan shroud for damage or deformation.
    - If there is no visible damage/deformation, you can put the print fan shroud back in place, print a few prints, and consider repeating the fan test after several print hours.
  - If not successful, something is potentially wrong with the fan itself.

If you'd like to collect and share data, please do as follows.
1. Identify which tool head is throwing a print fan error via "Calibrations & Tests > Fan Test".
https://help.prusa3d.com/article/lcd-menu-xl-multi-tool_500099
2. Pick the tool/s and follow "menu > control > temperature" and set print fan speed to 100%.
3. Access 'info > sensor info' menu, scroll down to find the fan RPM value while the fan shroud is still mounted on the print fan, and note down the RPM value.
4. Now you may remove the fan shroud, repeat the points 1-to-3, and take notes of the new RPM value.
5. Print a few print tests and then repeat points 1-to-4 to check if the new values are now set back within the expected range.

Michele Moramarco
Prusa Research

daavery, 2025-04-26:
The problem is purely altitude related. XL5 all 5 heads fail print cooling fan test at 7500 rpm.  The lower air pressure at 8000ft/2500m decreases the air resistance and so the fan spins faster.  At lower altitudes removing the fan duct reduces back pressure and makes the fan work harder and slows it down some.  The my altitude it's still not enough to get within the existing fan speed window. Also at the higher altitudes the cooling is reduced due to the lower air density so the naturally faster fan speed moves more lower density air which partially compensates for the lower thermal capacity of the air stream

daavery, 2025-04-26:
As a reference have the prusa sales team record the fan speeds of the printers they are bringing to RMRRF at the office and the repeat when they are at RMRRF. That change from sea level to 5200ft is still less than 2/3 of what is seen here

JeremiahGillis, 2025-04-29:
Please see my data below.

Test Conditions:
Altitude: 4997 Feet (1523 Meters)
Density Altitude: 6100 Feet (1859 Meters)
Firmware: 6.2.5

| Fan Number | Fan Speed [RPM] | Fan Speed No shroud [RPM] | Filament Usage [m] |
| ------------- | ------------- | ------------- | ------------- |
| 1 | 6958 | 5964 | 2000 |
| 2 | 7028 | 5934 | 138.8 |
| 3 | 7095 | 5998 | 56.9 |
| 4 | 7106 | 5847 | 7.6 |
| 5 | 6911 | 5876 | 7.7 |

After running a fan for 30 minutes at 100% with shroud:

| Fan Number | Fan Speed [RPM] |
| ------------- | ------------- |
| 3 | 7046 |
| 4 | 7030 |

Blocking fan shroud, 100%:

| Fan Number | Fan Speed [RPM]
| ------------- | ------------- |
| 1 | 7886 |
| 2 | 7881 |
| 3 | 7901 |
| 4 | 8004 |
| 5 | 7794 |

Fan Test:

[image]

Fan 3 Pictures:

[image]

[image]

[image]

Fan 4 Pictures:

[image]

[image]

[image]

As you can see, there is no debris, but there are long periods of inactivity. The RPM on fan 3 and 4 decreases 49 to 76 RPM after running at 100% for 30 minutes. The fans still fail the test afterwards. The problem is the density altitude. My data does not even represent the hottest month which is August. The density altitude can reach over 8000 feet (2438 meters).

Looking at this as an end user, what is important is the fan turns on and works at different RPMs. This test should be kept simple and allow a wider range of RPMs. Otherwise you have conditions like this based on various altitudes or different shroud designs. Restrictions should be left to the user under maintenance. We can refer to the computer and laptop industry. Computers do not tell a user when their fan or cooling system is dirty. They only alert a user when the fan drops below a minimum RPM.

n8many, 2025-08-17:
Someone else at 5000ft reporting in. I had a similar issue with my Semi-assembled 5T but with a hotend fan. I had similar issues with print fans, but removing the ducts solve that problem.

Unplugging the hotend thermistor to get the fan to run at 100% resulted in an RPM of 8760ish. Doing the same to a fan that was passing self test resulted in an RPM of 8130RPM. Not sure what the average variance is of these fans, but based on the code the passing range is 6500-8700rpm, and the two I was able to easily test were on the high end of that.

Perhaps we need an altitude setting for these printers (or just a high altitude checkbox) to adjust the fan range for air density?

3d-gussner, 2025-10-16:
> [@3d-gussner](https://github.com/3d-gussner) I see a solved flag has been added. Does that mean this will be fixed in the next firmware release?

@JeremiahGillis Yes this should be fixed in the next firmware release.

ewinholt-hub, 2025-12-03:
Also been having this issue semi-recently. I was on 6.4-alpha-whatever, updated to the release, and the problem continues. Only on the first three XL tools, weirdly, four and five test fine.

To be clear: All of the fans turn on, spin, are free of debris, spin at 40%, then three X's on "Print Fan" for the first three tools.

daavery, 2026-03-06:
Still fails all 5 print fans on XL located at 8200 ft 2500 M altitude.  The air pressure here is nominal 22 inHg 745 hpa . The low pressure causes all fans to run much faster due to the lower air drag.  All the print fans average between 8500 and 8600 rpm at full speed.  You should be able to test in the lab by putting a head in a sealed chamber and  evacuating the chamber to 0.75 bar

daavery, 2026-08-29:
still fails 5 of 5 print fans on XL with 6.10.1

murk-sy, 2026-10-16:
Still fails on 7.0.0

Axing the switched fan check in favour of this would be much better since i don't exactly know how to make it pass for sure and I'd rather not experiment.

Or if switched fan passes, but fan tests don't, re-run the test with the higher range.

daavery, 2026-11-03:
firmware 8.0.1 fixes this !!!!!!

JeremiahGillis, 2026-11-03:
I can confirm that 8.0.1 fixes this issue!

While I am happy that this has been addressed, I still am disappointed that it took three years to fix given the number of occurrences.