prusa3d/Prusa-Firmware-Buddy #3484: [BUG] Prusa XL false modular bed error 17255 due to high nozzle temperature

rrsawant1, 2023-11-17:
**Printer type** - XL 5T

**Printer firmware version** - 5.1.0-alpha2

**Original or Custom firmware** - Original

**Describe the bug**
Summary:
A false positive 17255 modular bed error is triggered due to the printing conditions when all heatbed tiles are working properly.

Cause:
When materials with greater printing temperatures and low bed temperatures are printed such as Polymaker PA612-CF, the nozzle heat leaks into the heatbed triggering error 17255. This occurs during the first layer of a model with large surface area on the bed. The greater surface area allows more time for the nozzle to leak heat into the heatbed tile. Since there is a large temperature delta between the nozzle and heatbed tile, it triggers the error quickly.

**How to reproduce**
Print high melting temp, low bed temp filament with a large bottom surface area fully covering a tile. A thin rectangular prism spanning one entire tile is sufficient.

**Expected behavior**
During the first layer, the printer will crash with error code 17255 for that tile. This phenomenon may also occur with certain filaments that have a high bed temperature (Prusament PA11) if the first layer is printed slowly enough for heat to absorb into the tile.

**Possible solutions**
In firmware, detect when the nozzle temperature is high and it will be moving over a heatbed tile slowly. In this case, increase the tolerance for 17255 to be triggered.

Don't detect the high nozzle temp and simply increase the tolerance for all prints. 

**Slicer settings and GCode**
Version 2.7.0-beta1
[Debree Bucket PA612_0.6n_0.25mm_PA,PLA_XL_6h14m.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/13394497/Debree.Bucket.PA612_0.6n_0.25mm_PA.PLA_XL_6h14m.zip)

hardiebotha, 2024-01-16:
I am having the exact same issue printing the same material - PA6-CF at 290C with the bed at 50C.

I was getting an error on tile 3, and per the knowledge base, I swapped the connector to tile 4, at which point the error moved to tile 4. I suspected the error was triggered by heat injected from the print, and moved the printed part. I am now getting the error on tile 10, where the part is now centered.

I've never had bed heating issues before, and do not believe the issue to be with the modular bed elements or thermistors. I suspect the thermistor is detecting an increase in temperature due to heat from the printed part even when heating is switched off for the module, triggering this error.

A possible avenue to resolution is to allow the bed module to rise in temperature, even above the set temp as long it is being printed on. Once the head moves away for more than 10? seconds, the temperature should no longer rise. If it keeps rising, it may indicate thermal runaway conditions.

czei, 2024-01-27:
I am having the same issue, but with regular ABS and PETG.   Any model printed in the middle of the bed fails with a 17255 error.   I can print ABS if the model is tiny and it's only on Tile 1, but pieces that are larger and use most of the bed fail.

Even if I try to print a benchy in the middle of the bed fails with a 17255 error.  

Oddly enough the Tile number remains the same even if that tile is replaced, or swapped or the cables swapped.

hardiebotha, 2024-01-27:
If you swap cables and the tile number remains the same in the error
message, the problem is most likely the cable. Does the reported tile
number change if you move the model around on the print bed in the slicer?

czei, 2024-01-29:
I don't quite get that logic.   If you completely swap out a cable, and the cable is bad, then the reported Tile number of the error would change.   

For example, if you have a bad cable connected to Tile #1, and you completely swap that cable with Tile #2, you'd then have a bad cable on Tile #2, and the reported error would be on Tile #2.  

In any event the reported Tile number of the error stays the same no matter what model is being printed, and in any location on the bed.    In this case the error is on Tile #15, and that cable was completely replaced with a new one.  But I decided to try swapping cables just to make sure, and now I can't get it to fail.   This probably indicates the cable wasn't plugged in completely, but that cable had been unplugged and plugged many times, as well as swapped.   At least everything's working now though!

dhyananbho, 2024-01-30:
I get the same error in a slightly different but similar situation:

### **XL • Printing TPU (Extrudr Flex SemiSoft) • Modular Bed Error Tile No. 11**

I wanted to print a large lid with TPU (Extrudr Flex SemiSoft clear) that would fill almost the entire bed. During the first layer, when tile 11 was mostly covered, the "Modular Bed Error" for tile 11 occurred, resulting in a first restart of the printer due to the error indication and a second restart of the printer after pressing the button. This process resulted in the hot nozzle staying very close above the print sheet in the same position for a long time with no possibility for user intervention. I tried the same print a second time with exactly the same result. I then swapped tiles 10 and 11, including the cables, and tried a third time, again with exactly the same result. This means that it is not a hardware error, but that the software parameters are too tight / too strict for checking this error.

—> Why two reboots in the event of an error?
—> Why is the bed not moved away fron the hot nozzle in the event of an error?
—> The software parameters for checking this error are too norrow / too strict. Please widen these parameters.

—> This bug is very annoying because it prevents me from printing the desired object and I cannot fix it myself. I need a solution very urgently.
—> Is there a possibility for me as an experienced user to temporarily disable this check?

After further research and I found the following:

I still think that the test values of the "Modular Bed Error" are too strict, because no such error should occur during normal printing, but only when there is a real danger. Apart from that, I have found out why the error occurs with extrudr FLEX SemiSoft (and Hard) but not with PETG. The reason is obvious: In the configurations of all extrudr FLEX, the component cooling is turned off completely. This means that less heat is dissipated and the error occurs. But when using a 1.0 mm high flow nozzle with 0.5 mm layer height and an automatic cooling of 60-80 %, the error still occurs (at 0.3 mm layer height it does not). The test values are definitely too strict.

XL Firmware Version 5.1.2+13478
PrusaSlicer Version 2.7.1+MacOS-arm64

[image]
[image]
[image]
[image]
[Sortbox Hüllendeckel 6x4.3mf.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/14104815/Sortbox.Hullendeckel.6x4.3mf.zip)

hardiebotha, 2024-01-31:
> I don't quite get that logic. If you completely swap out a cable, and the cable is bad, then the reported Tile number of the error would change.
>
> For example, if you have a bad cable connected to [...]

Gotcha - I thought you were just swapping cables on the heated side, did not realize it was a complete replacement. My bad, I didn't quite understand what you were doing.

Skaronator, 2024-02-25:
I also ran into error 17255 as well, but always late (5 hours) into the print. I tried to print basically the same model twice, but I sliced it two times with different settings. I always had Tile 15 which is the top right tile which shouldn't even heat up:

[image]

### First Print
[image]
[image]

### Second Print
[image]

hardiebotha, 2024-02-29:
I managed to get around the issue by printing without heating the bed, which probably means it was not being monitored for thermal runaway any more as no power is supplied for heating. Not all filaments re this forgiving, so it is not an appropriate way to address the issue, but it was a good way for me to confirm some suspicions on this issue.

MooseOnTehLoose, 2024-03-31:
[image]

I am also seeing this issue and I am not sure at which point it was introduced, but have done the following troubleshooting steps:

1. Switch the cables for tile 6 and tile 7 - FAIL

2. Switch the physical location of tile 6 and 7 - FAIL

3. Try Prusaslicer 2.7.2 and 2.7.3 - FAIL

4. Upgrade from fw 5.1.2+13478 to 6.0.0-alpha+14120 - FAIL

5. Move the print directly to the edge of the print bed, over the tile reporting the error - SUCCESS

6. Move the print slightly, now reports a completely different tile - FAIL 
[image]

Some weird stuff happened to the error report after upgrading to 6.0.0 as you can see in the images below, the tile number implies my Prusa XL is more like an XXXXXXXXXL:
[image]

I have a .gcode file for 0.4 nozzle ASA that can reproduce the issue but can't upload it to github.

TouchMonkey, 2024-04-01:
I am also hitting this when printing a roughly tile-sized object using PA6-CF at 290C with a bed temp of 50C. Would love the logic to be updated, or a way to disable this check for a print.

gladrusskamrad, 2024-04-08:
The problem when printing is with the table temperature above 100°C. Errors on different tiles are constantly introduced. Replacing the cable with new ones does nothing. And Prusa apparently can't solve this problem. 
[image]
[image]

czei, 2024-04-08:
> 4\. Upgrade from fw 5.1.2+13478 to 6.0.0-alpha+14120 - FAIL

Thank you for reporting everything you've tried!  I was able to try the alpha firmware and now won't bother.

I am also getting a Tile #15 "unexpected temperature peak" error on every single print under identical conditions to yours.   This is a bummer as I've got product to ship and had sold a couple of MK3's to make room for this printer and now it's just sitting there.   The printer does work with PLA and PETG, but everything I sell is ABS.   

It definitely seems to be a firmware problem on the tile temperature sensor.  I've tried swapping out whole tiles, cables, and the heated controller board, and nothing has worked.  

It really would be nice to get an acknowledgment of the issue from Prusa and some sort of timeline for a fix.  Does anyone know how to escalate this bug?   My PrusaXL has been basically a brick for months, not to mention so many wasted hours trying to troubleshoot the issue.

czei, 2024-05-31:
> > Does anyone know how to escalate this bug? My PrusaXL has been basically a brick for months, not to mention so many wasted hours trying to troubleshoot the issue.
> 
> I believe if you raise it with [info@prusa.com](mailto:info@prusa.com) it will at least go into their issue tracker, and you may get some suggestions.

I was working with Prusa for months with no luck.   Luckily the problem just went away when I went from a single head to 5 head machine.

MooseOnTehLoose, 2024-06-04:
Just an update here - I was able to work around this by just heating the entire bed. 

For everybody reading this I want to also give you a heads up that you probably got here after having failures printing ASA, and if you have enclosed your XL to print ASA.... Prepare for meltdown. You need to start reprinting all the extruder parts in ASA NOW. You also need to reprint the belt holder for the x and y axis thats mounted to the front of the toolchanger. The printer is going to fail slowly and all the PETG held under tension won't stay under tension for long. 

Works great now that I've done that and also switch to 5mm OD 3mm ID ptfe tubing and fittings.

Prusa-Support, 2024-06-04:
Thanks for reporting.
I'm under the impression that there have been no new reports since mid-April when FW 6.0.0 (final) was released.
That would mean the issue is potentially addressed in the most recent firmware release, and I'd like you to try and confirm.

- In the past, a few false modular bed errors would be potentially possible due to heated magnetic field inconsistencies. This was addressed in production with complementary controls, plus, an additional "shield" for the modular bed's board.
This shield is not really necessary but you may want to check if your printer is missing it.
https://help.prusa3d.com/guide/how-to-fix-modular-bed-error-hw-solution_512077
- In other cases, the issue is most likely caused by a damaged board/socket or a pulled wire (no longer firmly crimped).
- Last but not least, there have been a few **unconfirmed** cases where the problem was caused by air pockets between the sheet and module bed due to debris stuck between them.

Please test the most recent firmware and, with a reminder that in most cases the problem is caused by hardware problems, please consider contacting Customer Support for troubleshooting.
https://help.prusa3d.com/article/customer-support_2287

Michele Moramarco
Prusa Research

rob-miller, 2024-06-05:
> Thanks for reporting. I'm under the impression that there have been no new reports since mid-April when FW 6.0.0 (final) was released. That would mean the issue is potentially addressed in the most recent firmware release, and I'd like you to try and confirm.
> 

In fact I only found this problem when starting to print ABS in an enclosure after installing FW 6.0.1.  This was for a small test cube, and I could change the error tile by changing the tile I was printing on.  I initially also tried swapping cables as suggested on the support pages.  I eventually found a spot on a tile that did not trigger the error.

czei, 2024-06-05:
> In fact I only found this problem when starting to print ABS in an enclosure after installing FW 6.0.1. This was for a small test cube, and I could change the error tile by changing the tile I was [...]

FYI, my identical problem was also triggered by ABS in an enclosure.   I worked around the problem by switching to PETG until the issues was resolved, and so haven't tested the 6.0.1 release with ABS yet.   Hopefully it will be resolved, as I've got $500 of ABS to use up.

MooseOnTehLoose, 2024-06-07:
This problem still exists for me on 6.0.1 and is easily reproducible with the right STL file, but I will upgrade to 6.0.2 and confirm its still a problem. I have already verified there's nothing wrong with my bed hardware.

tg73, 2024-07-20:
This bug is still ruining people's days in 6.0.3.

Wow, I hate this bug. Had the error at first layer in an ABS print, found this issue report, so tried again with the enclosure a bit open. It was going well, 8 hours or so in. I had to be in the room briefly so closed the enclosure door, 3 minutes later - off goes this damn error. Urgh. I really loath it when a "smart" system is exceedingly dumb because of its broken smartness. Dear devs, *please* fix this. @Prusa-Support - here's a nice new report for you, please exit denial mode.

[image]
[image]
[Dayspring baffles and holders.zip](https://github.com/user-attachments/files/16321705/Dayspring.baffles.and.holders.zip)

czei, 2024-07-20:
> This bug is still ruining people's days in 6.0.3.

I'm still getting this error regularly.  There doesn't seem to be any rhyme or reason to it.   I've got $500 worth of ABS rolls to print and have had to switch to PETG just to get product out the door.   It's definitely triggered by prints that need high bed temperatures, though.  

I also have swapped in brand new tiles, new cables, and even paid for a brand new bed controller board.  It doesn't seem to be hardware related at all since all of us are getting an error on the same Tile number, which would be exceedingly remote.

tg73, 2024-07-21:
I managed to get the print to complete on the 3rd try by 1) aligning all the parts to the centre of the bed rather than front left; 2) heating the whole bed and 3) leaving one enclosure door slightly open (see enXLosure on printables). Maybe I didn't need to leave the enclosure door open, but I didn't want to jinx it - I just wanted it to work. I'm building my RatRig VC4 at the moment, which will be a petg-parts-free ABS/ASA monster.

TouchMonkey, 2024-07-22:
This issue caused so many failures when printing objects with large bed-contact area in PA6-CF at 290 that I made a custom firmware.

NOTE: Before going down this route, MAKE SURE you don't have a hardware issue. Follow the instructions on the error page.

The temperature check is in the function CheckTemperaturePeak in src\puppy\modularbed\control\StateLogic.cpp. By default, the function looks for a temperature reading >(SetTemp+3) for 60 seconds. Possible fixes would be to change the number of degrees by changing TEMPERATURE_PEAK_THRESHOLD_DEGREES, or the number of seconds by changing TEMPERATURE_PEAK_THRESHOLD_SECONDS.

Anther option would be to scale the threshold based on nozzle temperature, so that higher print temperatures have more wiggle-room to account for the extra heat-soak.

I went the easier-but-less-safe route of setting the threshold to BED_MAXTEMP, which is 120 for the XL. The max-temp value is defined in include\marlin\Configuration_XL.h, which isn't actually included by StateLogic.cpp so I hard-coded it. Personally, I don't care how hot the bed gets as long as it doesn't get hot enough to damage anything. There's also BED_MAXTEMP_SAFETY_MARGIN defined as 5, so a slightly-more-safe option might be 120-5=115.

The screenshot below shows the change I made. I combined the two checks because TEMPERATURE_PEAK_AMBIENT_DEGREES is 50, and since 120 > 50 there's no point in doing both.

Again, I'd recommend something less aggressive unless you understand the risks, especially since using a custom firmware requires voiding your electronics warranty.

[image]

danopernis, 2024-07-23:
Hi folks, thank you for reporting. I have created an internal ticket BFW-5841 and we will be working on fixing this issue. I can't give you an estimate as to when the fix will be available. We are going to relax the conditions leading to red screen being shown, similar to what @TouchMonkey is proposing.

raymondagena, 2024-08-05:
Thank you. I have 1200 print hours on my XL and I just ran into this bug. This was printing TPU and it seems to fail over the exact same spot every time.

Mr3Deee, 2024-09-05:
Over the past months I have seen this problem twice:

1. 17. July - tile #15
2. 2. Sept - tile #9

It appears to show up only when printing PETG (which is the material with the highest temperatures)

Looking at [this article from Prusa](https://help.prusa3d.com/guide/how-to-fix-modular-bed-error-hw-solution_512077), I can see that there is a solution - however - I cannot see the parts which should get printed to resolve this. 

Reading now this thread it sounds like that there will be a firmware update soon?

danopernis, 2024-09-06:
Just to give you an update, we still don't have a solution to this, but we have some pretty good reproducers and this is also manifesting on the iX printers (part of the AFS project) which are also using modular bed system. The original idea to just relax the conditions was rejected, as this is an important safety feature and we need to gather more data to judge how to solve this properly.

NickNothom, 2024-09-17:
@danopernis 

> we still don't have a solution to this
> we need to gather more data to judge how to solve this properly

Great, is someone at Prusa going to 'gather more data' to judge how to solve it properly? Or am I going to be forced to void the warranty to get the XL to perform the same functions as something less than half the price? 

Here's an idea; maybe you could consider giving us, **the owners of the hardware**, the option to control the error threshold?

danopernis, 2024-09-18:
> Here's an idea; maybe you could consider giving us, **the owners of the hardware**, the option to control the error threshold?

@NickNothom you as **an owner of the hardware** have full control over the error threshold. You are free to build your own firmware and upload it to the printer and you are also free to risk setting it on fire because you bypassed security features of the printer. Changes to thresholds may have unintended consequences which we are simply not going to risk without further analysis.

For users who are a bit more cautious, thank you for your patience, thank you for all the useful observations you are writing here in comments and thank you for suggesting workarounds :pray:

FyrbyAdditive, 2024-10-13:
I had this problem repeatedly today with a small print on tiles 4, 13, 5, 12 - i.e. in the middle. The printer is in the official enclosure and was pre-heated for polycarbonate. I had the error on tile 15, which I guess was not being heated and had absorbed enough heat to trigger the error due to the print circumstances.

My workaround was to enable the heat-whole-bed option, and as it expects the bed to be hot it now works.

czei, 2024-10-14:
> My workaround was to enable the heat-whole-bed option, and as it expects the tile to be hot it now works.

Thanks for the tip!   BTW, the error always is on Tile #15 for me every time I try to print in the middle.  I’ve been working around it by only printing in corners, but of course that doesn’t work if you’re trying to print a larger object.

czei, 2024-10-14:
> Looking at [this article from Prusa](https://help.prusa3d.com/guide/how-to-fix-modular-bed-error-hw-solution_512077), I can see that there is a solution - however - I cannot see the parts which should get printed to resolve this.

FYI, I tried that solution, even going to far as to replace tiles and the entire bed controller and it did nothing to fix the problem.  I’m convinced it’s a software issue that needs to be fixed by Prusa.

bruffjr, 2024-10-14:
Same problem for me. Been back for repairs. Seems to work for a month the the tile 15 error. I also been working around it. I'll try the entire bed on. I'm also in an enclosure, I use a lot ABS for my EDF projects. Hope somebody comes up with a permanent  fix.

gladrusskamrad, 2024-10-14:
> Same problem for me. Been back for repairs. Seems to work for a month the the tile 15 error. I also been working around it. I'll try the entire bed on. I'm also in an enclosure, I use a lot ABS for my EDF projects. Hope somebody comes up with a permanent fix.

The problem is in the board to which the heating table is connected. When it heats up, it begins to give various errors.

pieindaface246, 2024-10-15:
I am having issues printing [Polymaker PA12-CF.](https://polymaker.com/wp-content/tech-docs/PolyMide_PA12-CF_PIS_EN_V1.pdf) I believe this is due to the build plate temperature being 50C, while the print head is at 285C. I usually get to about 1/2 way through the 2nd layer of one of the largest bed area parts. 

After going through [this PRUSA troubleshooting guide (17255),](https://help.prusa3d.com/article/modular-bed-error-17255-xl_399993) I have ordered additional heat bed tiles that I will be installing and adding a follow-up. 

The printer (PRUSA XL) was purchased new in July/August and has been trouble free with the exception of this error. I would be surprised that the build plate tiles are faulty, based on the troubleshooting guide flowchart and the lack of issues with the heat-bed to this point.

Prusa-Support, 2024-10-19:
> > Looking at [this article from Prusa](https://help.prusa3d.com/guide/how-to-fix-modular-bed-error-hw-solution_512077), I can see that there is a solution - however - I cannot see the parts which should get printed to resolve this.
> 
> FYI, I tried that solution, even going so far as to replace tiles and the entire bed controller and it did nothing to fix the problem. I’m convinced it’s a software issue that needs to be fixed by Prusa.

Please notice that the linked guide is about a Hardware solution which mostly intended for early adopters and surely not required for printers shipped in the last ∼1y. All XL printers come with the so-called _Modular Bed PCB Shield_ by default since the hardware improvement was discovered (= when the guide was made available).
Plus, please notice this is not a printable part.

_The issue discussed in this thread **instead** is a different issue **at the Firmware level**, probably something to do with thresholds.
Our developers and testers are working on a firmware enhancement in observance of our safety standards._

This should be a sporadical problem.
Workarounds may vary from case to case but the methods that may seem to have significantly mitigated the problem in a few reported cases are:
- placing the sheet in the middle of the preheating process (at ∼50°C) may seem to have mitigated the problem in a few reported cases;
- preheating up all tiles together at once before starting the potentially problematic print job (whereas a reduced print area requires partial heat of a few tiles only);
- preheating the bed to a slightly higher bed target temperature before launching the print.

Again, this should be a sporadical problem that will find a solution via firmware.
If the issue is not sporadical but pretty much always reproducible please consider contacting our Customer Support.

Michele Moramarco
Prusa Research

pieindaface246, 2024-10-21:
> I am having issues printing [Polymaker PA12-CF.](https://polymaker.com/wp-content/tech-docs/PolyMide_PA12-CF_PIS_EN_V1.pdf) I believe this is due to the build plate temperature being 50C, while the [...]

After replacing the tiles, I have found that I am still getting the bed-overtemp error. I have tried moving my high temp part to span across the most amount of heat tiles possible, but I still got the overtemp error on the 2nd layer. 

I will try to preheat the entire bed by moving all of the smaller parts in my print to the surrounding area and printing my large part which is creating the error in the center of the build plate. I can also try printing at around 70C for the first few layers and then reducing bed temps. Will reducing bed temps mid print trigger the overheat warning if the print-bed starts rising in temperature when printing?

I'm not really sure what else to do. I have to get these parts printed for a project as an end use part and I don't want to change materials because of the project requirements.

czei, 2024-10-21:
> Please notice that the linked guide is about a Hardware solution which mostly intended for early adopters and surely not required for printers shipped in the last ∼1y. All XL printers come with the [...]

Just an fyi, but I’m surprised to hear this described as sporadic.  It happens every time I try to print large parts that require bed temps over 100C.   I’ve been unable to print anything except for PETG and PLA;  any serious customer work has to be done on other printers.

Support had me replace a bunch of bed hardware which made no difference.  After months working on this they offered a refund but I was hoping you guys would come up with a software fix.

NickNothom, 2024-10-23:
@Prusa-Support 
See if any of your XL's can manage to print this simple cube with a bed temp of <50C and a nozzle temp of >285C. I think you will find it's not sporadic at all. For me it's reproducible 100% of the time and I expect you will experience the same. 
[Shape-Box.zip](https://github.com/user-attachments/files/17483921/Shape-Box.zip)

czei, 2024-11-26:
To follow up, preheating the entire bed seems to solve the issue for me.  Prints that would give a modular bed error are able to print if I preheat first.  

All-in-all, the tile design on the bed isn't helping any.  Rather than speed up the printing process, between the pre-heat and the aborbing heat phases there's a 20m wait to even start a print!  Its faster to print on my old MK3.5 than the XL, which is disappointing.

nichenke, 2024-12-06:
Heating the entire bed didn't work for me - I still saw errors.

I use this custom printer g-code to turn off the bed heating during printing. This prevents the printer from checking for overheated tiles during printing but still warms up the bed enough to keep it in range.  I do have a chamber, which helps, and I'm using Magigoo PA as well. 
YMMV. 

- Start g-code (at the end)
```
M190 S{bed_temperature[initial_tool]} ; Wait for bed to reach temp
M140 S0 ; turn off heatbed for first layer
```

- After layer change
```
;[layer_num]
; if first 4 layers, set and wait for bed temp, rest 20s
; then disable bed to avoid BFW-5841
{if layer_num <=3} M190 S{bed_temperature[initial_tool]} {endif}
{if layer_num <=3} G4 S10 {endif}
{if layer_num <=3} M140 S0 {endif}

; if layer 4, then set bed temp for rest of print
{ if layer_num == 4} M190 S{bed_temperature[initial_tool]} {endif}
```

pieindaface246, 2024-12-09:
> After replacing the tiles, I have found that I am still getting the bed-overtemp error. I have tried moving my high temp part to span across the most amount of heat tiles possible, but I still got [...]

I was able to get around this issue. I first tried to overheat the material, but that was not working because of my specific material warping due to the higher bed temperatures. 

My solution was to overheat the bed for the first layer (20C over the required temp for this material), then for the subsequent bed layers, I reduced the bed temp to the correct temp for the material or maybe 5C over the material bed temp. 

Modifying the g-code could probably get a better rampdown or other temp modification for the bed material up to the 4th layer similar to nichenke's solution.

Prusa-Support, 2024-12-30:
I'm sorry for the trouble but as explained earlier - https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/3484#issuecomment-2423898682 - if the problem is not sporadical in your case, I must encourage you to follow the direct individual support via official Customer Support channels to discuss potential hardware solutions.
https://help.prusa3d.com/article/customer-support_2287

Michele Moramarco
Prusa Research

rob-miller, 2025-01-01:
>  if the problem is not sporadical in your case

The frustration here is that the problem is only 'sporadical' in the sense that it is difficult to predict which prints will have a problem.  As a result, I have set 'heat entire bed' on since I saw it suggested here as a workaround, and have been waiting since then for a firmware fix.  In other words, a prominently advertised feature of the XL (multi-tile heated bed) is unusable because this problem can cause an unrecoverable error leading to a failed print -- thus the XL is unreliable when heating only selected tiles as advertised (unless as noted above you choose to void the warranty).

It seems that the problem lies in the thermal model expected to describe the bed, but it is not clear to me that all aspects can be known by the firmware when deciding if the bed tiles are heating appropriately.  Some details missing might be:

-  is an enclosure (Prusa or not) in use?
- is there a fan on the system?  Is it controlled by the sandwich board connector or a separate thermostat or a switch?
- how is the rate of cooling affected by the ambient temperature? (see Newton's Law of Cooling -- my garage where the printer is can be 37C)

In addition are the details raised above about transfer of heat from the Nextruder and/or part being printed to the heating tile separate from the tile heater itself.  In other words, **there are valid reasons why the heating tiles may absorb more heat than provided internally, and why they may cool more slowly (or faster) than was seen in the design lab**.  

I understand the safety issue around monitoring for thermal runaway with the heated tiles, however the workaround of heating the entire bed actually dumps more heat into the system rather than less.  

We seem to be at an impasse, because we can't currently come up with thresholds and/or a sufficiently nuanced algorithm to predict the heating tiles thermal behaviour in all of the real world environments where the XL is in use.  

I conclude that the current thresholds are too tight, and I think that it should be possible to loosen them such that the XL can work as intended in a wider range of environments but still not be so loose as to run the risk of bursting into flames.

StephenStage, 2025-01-18:
I worked around this by inserting an G106 S255 on layer one to turn on the part fan, and a G107 on around layer 3 to turn it off. The fan will cool off the bed and prevent the thermal runaway code. So far it has worked every time.

jimmysun2000, 2025-03-17:
I’m having the same issue with Polymaker PA6-CF with nozzle at 285 C and bed at 50 C. Error always comes up at similar times. Which prevented me from printing anything large with this filament. 

I didn’t want to break the appendix on the buddy board. I also didn’t want to overheat the bed above 50 C, which Polymaker specifically said not to. And I prefer not to have the print fan running for maximum layer adhesions. 

So I’m working around this for now by turning off the bed heating completely by setting bed temperature to 0 in filament settings in the slicer. 

But I am using the Prusa XL PA Nylon Sheet which sticks to this filament surprising well even for a room temperature bed (at around 30 C inside an enclosure). No wrapping for larger models and prints pop off straight away when the sheet is bent.

rongith, 2025-04-07:
- Enclosure.
- PETG (Nozzle 255 - Bed 90)
- Firmware 6.2.2+8853
[image]

That doesn't help much. For the moment it is more for Prusa statistics.

CZDanol, 2025-04-07:
Just a small update - the cogs finally started turning and we are having discussions over the problem currently.

rongith, 2025-04-07:
In my case, I suspect the sun lights hitting the enclosure and the bed plate bring some radiating heat. That extra heat may force the temperature to reach a firmware threshold.

Mr3Deee, 2025-07-11:
I have seen this problem in the past sporadic with PETG - never with PLA. Maybe once every 2 months. After it happened I had to cut the power and then it worked. Only reset and reprinting was producing the same problem.

Now I have started to print with ASA and I was able to print something between 0 and 10 layers before the print stops. I am now the 5th experiment in and it stops (replacing this time the object to the front left instead of the middle). Either tile 2 or 15 or 11, or 9 or ... .

Yes, I use an enclosure (EnXlosure with my own filter / fan system), but I do not think that that has anything to do with it - why should it only happen with ASA? Also I do not have any sunlight coming in or other temperature or air flow related specials.

After reading all the suggestions here I see a few things I can try:
- No fan wahtsoever for ASA (contrary to what Polymaker said they need)
- Pre-heat the entire bed
- Use a different printing head
==> I will try that all and see.

However... 
@CZDanol - what is the latest status on this? It is haunting me already since I got the printer and while it was sporadic at best, with ASA it appears to be a continuous problem.
@czei - did the fan help with the problem?

Any suggestion is highly welcome. I cannot believe that I cannot print a single ASA part on my printer only because the firmware thinks that there might have been a heat tolerance which cannot be seen by anyone using a heat sensor (at leas that seemed to be the pattern in all the threads I have seen).

Many thanls for all the help in advance!

CZDanol, 2025-07-11:
Hello, we have decided to bump the tolerances for the error frmo 3 to 5 °C, which should hopefully resolve the issue. Planned for the next major release, I've suggested to pick to a next minor release, too. Cannot comment on the release dates, sorry.

rongith, 2025-07-11:
>     * Pre-heat the entire bed

I too took the path of heating the entire bed.

Mr3Deee, 2025-07-15:
I wanted first to test a bit more before answering. 

By using only the lower half of the bed it appeared to have printed properly (finally). I did not yet try the full bed pre-heating.

With that it is either really a problem with multiple heat tiles (just to be secure I ordered a bunch now) or there is something special about the backside of the printer (opening drafts, etc.) or it was pure coincidental and I got lucky. 

Let's see. I am waiting either way now for the tolerance update before continuing with more experiments. 

Either way - many thanks for your quick answer!

StephenStage, 2025-07-28:
Nearest as I can tell, heating the bed only helps if the bed doesn't cool enough between starting the print and getting a few layers between the nozzle and the bed.  As soon as the heating element starts I'm likely to get the same code.  One thing that did solve this completely is running the print fan at 100% for the first few layers using M codes.

Mr3Deee, 2025-08-02:
Hello!

I was invited to test alpha 6.2.5 8912 for the XL. Does that include the new wider range temperature variations?

Many thanks in advance!
Stefan

Mr3Deee, 2025-08-03:
Ah - I see. Daeng. Okay, I will give it a try, but the more important thing is the tempterature.

Prusa-Support, 2025-09-04:
Thanks for your collaboration and patience.
The [final version of FW 6.2.6](https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v6.2.6) drastically reduces the occurrence of unexpected temperature peak errors.
There may be edge cases in specific hardware and ambient conditions where a false positive event may still happen but is hardly reproducible, and theoretically far from realistic/common print scenarios.
Therefore, it seems safe to close this issue as solved, but we will keep monitoring and be attentive in case of further reports.

Michele Moramarco
Prusa Research

Mr3Deee, 2025-09-17:
Awesome Rob! I am glad that you were able to verify it!

However - I did spend today 3 hours more and filed finally a bug #4842 as I have seen BSODs as well as more head crashes (only with a single operation) and sorry, but I couldn't test it and finally switched back.

czei, 2025-09-18:
> Thanks for your collaboration and patience. The [final version of FW 6.2.6](https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v6.2.6) drastically reduces the occurrence of unexpected [...]

Please don't close this ticket!  Unfortunately the same error occurs for me with the firmware 6.4.0-alpha. Its always reproducible!  Too bad there's not some way to capture a bunch of debug output so you can see why this is happening.

[image]

Note that I've swapped cables, replaced cables, replaced the bed controller, upgraded the mounting on the bed controller, done everything Prusa has asked me to do.

murk-sy, 2025-09-18:
You capture printer logs if you connect to serial with specifically baudrate 57600. Not sure if you'll get anything useful, but that's the way to access it.

For own future reference: https://github.com/prusa3d/Prusa-Firmware-Buddy/blob/77255c9c85a25feb22f112724016aba552a1aa3e/src/buddy/usb_device_cdc.cpp#L18-L24

3d-gussner, 2025-09-18:
I hope to find a pattern with the log files and try to see if we can improve the firmware for these cases like it has been done in https://github.com/prusa3d/Prusa-Firmware-Buddy-Private/pull/8102 .

[10 more replies]

closed 2026-02-19 by github-actions[bot]