prusa3d/Prusa-Firmware-Buddy #3485: Ramping Lift at Tool Change with No Wipe Tower

JeremiahGillis, 2023-11-17:
### Description of the bug

I am not sure if this should be a feature request or a bug. I have ramping lift enabled which I can clearly see working at a layer change. However, the standard z-axis lift occurs at the end of a layer when a tool change is the next step. This can lead to stringing. I am not using a wipe tower. Shouldn't ramping lift also occur here?

[image]

https://github.com/prusa3d/PrusaSlicer/assets/13179985/b0683d42-10b6-4fec-983a-864e16fee2df

[PrusaXL-5-Head-Toochanging-Banding-Test.zip](https://github.com/prusa3d/PrusaSlicer/files/13395140/PrusaXL-5-Head-Toochanging-Banding-Test.zip)

### Project file & How to reproduce

You can open my attached 3mf model and view the tool path with tool changes enabled.

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.7.0-rc1

### Operating system

Windows 11 x64

### Printer model

Prusa XL, 5.1.0-alpha2

MechSimon, 2024-02-14:
@rtyr 
I am seeing the same issue on the wipe tower, where it does not have ramping lift as it changes tool and results in a small "blob" where the tool lifts up. The is a risk when the new tool continues and hits the blob, that the wiper tower can tilt and cause failure. 
I have all default ramping lift settings.
Is there any fix for this?

rtyr, 2024-02-14:
The z-lift you are talking about is performed by the firmware after tool change command.

> Is there any fix for this?
It would have to be fixed on the firmware side.

danopernis, 2024-12-30:
I will have to check, reopening for now.