prusa3d/Prusa-Firmware-Buddy #3489: [BUG] Cannot raise nozzle temperature to hot tighten according to instructions

kozross, 2023-11-19:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - MINI

**Printer firmware version** - 5.1.0-beta

**Original or Custom firmware** - Original

**Optional upgrades** - Filament Runout Sensor

**USB drive or USB/Octoprint**

USB drive (not relevant here)

**Describe the bug**

According to [Prusa's own instructions](https://help.prusa3d.com/article/changing-replacing-the-nozzle-mini_134235#procedure), you need to heat the nozzle to 280C to hot tighten a new nozzle. I have done this before plenty of times without issue on previous firmware versions. 

However, on 5.1.0-beta, I cannot set a temperature any higher than 275 for the nozzle: if I try to go higher, the setting won't change. This is a bizarre state of affairs: either the documentation is no longer correct (which I doubt, since this is a procedure that isn't dependent on software in any way), or there is a bug in the firmware. This is _doubly_ puzzling because the [Mini's own product page](https://www.prusa3d.com/product/original-prusa-mini-semi-assembled-3d-printer-4/) specifies a hotend temperature cap of 280.

**How to reproduce**

1. Select the 'Control' menu from the main screen.
2. Select the 'Temperature' sub-menu.
3. Scroll to the 'Nozzle' temperature and try to raise it.
4. Observe that at 275, the value no longer rises, no matter how much you turn the click wheel.

**Expected behavior**

Either of the following:

* The documentation for the Mini, both technical and product page, should be updated to reflect its new 275C minimum; or
* The firmware should allow setting temperatures as they used to be.

**G-code**

No G-code was used here.

**Crash dump file**

No crash dump file was produced for this.

**Video**

No video is needed to reproduce this problem.

allengregoryiv, 2023-12-03:
This needs to be fixed, we can't load nylon filament properly either as it asks for 285C at the nozzle in the menu.

kozross, 2023-12-03:
> This needs to be fixed, we can't load nylon filament properly either as it asks for 285C at the nozzle in the menu.

This is still a problem even in the release version of the firmware.

allengregoryiv, 2023-12-03:
It just gets to 275 and lets you load but continues to try to reach 285 even though it can't.

[image]

kozross, 2023-12-15:
Bumping this, as we've now had _two_ firmware releases with this issue without as much as an acknowledgement, much less a fix.

yodax, 2023-12-16:
Pinging @Prusa-Support , can you please have a look at this or give us an ETA. I’d like to change my nozzle, but I can’t follow the proper instructions.

bkerler, 2023-12-21:
Prusa answered, see the PR #3613 response from danopernis

bkerler, 2023-12-22:
Good news: Seems #3635 will come forward and will be merged soon. So the nozzle temp issue will be fixed.

bkerler, 2023-12-22:
PR #3635 has been merged and issue can be closed :)

kozross, 2023-12-22:
@bkerler - thank you for getting this over the line! Now it's just a matter of a release getting cut.

closed 2023-12-22 by kozross