prusa3d/Prusa-Firmware-Buddy #3493: [BUG] Huge layer shift with 5.1.0 alpha2 on MK4

Krosta8, 2023-11-20:
**Printer type** - MK4IS

**Printer firmware version** - 5.1.0alpha2

**Original or Custom firmware** - Original

**Optional upgrades** - Prusa Nextruder Silicone sock

**USB drive or USB/Octoprint**
  Printing only via PrusaConnect (LAN), so USB since the file is on USB disk? Using original USB stick from Prusa.

**Describe the bug**
I was priting some custom, very easy print from PETG -> Prusament PETG Anthracite Grey 1kg. 0.4mm Nozzle, Input Shaper. All based on PrusaSlicer 2.7.0-beta1 bgcode format, standard profiles. 
I checked on that after a few hours and found out a huge layer shift (almost 18.9mm) at the height of 43.8mm.

**How to reproduce**
Try to print the included bG-Code via PrusaConnect and at 43.8mm it should do the layer shift.

**Expected behavior**
Without layer shifting.

**G-code**
[ANTRACIT_PETG_Kryt zasuvek u virivky_0.4n_0.2mm_PETG_MK4IS_5h54m.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/13412043/ANTRACIT_PETG_Kryt.zasuvek.u.virivky_0.4n_0.2mm_PETG_MK4IS_5h54m.zip)

**Crash dump file**
  Please attach the crash dump file. This will make it easier for us to investigate the bug.

**Video**
  Please attach a video. It usually helps to solve the problem.

[image]
[image]

MichaelJLew, 2023-11-21:
Does this happen reproducibly? How many times have you tried? Why do you think that it is a firmware bug and not a mechanical problem? A failed print is not usually a bug, and does not, by itself, justify a bug report.

Krosta8, 2023-11-21:
Happened only once to be honest. I did not try this again, don't want to waste more filament, so I did slice it again rotated and it was printed without issue. 
I wanted to report it, if someone, for example you, think, that it is not something related to FW, I will close it. I just thought better report it and close it if it is stupid than does not report it.
I would say, that it is not mechanical issue, because I am printing a lot recently bigger things and without any issue. So I assumed that it could be related to FW, some wrong calculation or something like that. The movement was to the very end of the printing bed.

MichaelJLew, 2023-11-21:
Not a bug. Close it.

If you want to discuss print failures or to vent about printing then I suggest that you use the forums rather than GitHub.

Krosta8, 2023-11-21:
Ok, sorry for this meaningless report then. No need to discuss the print failure anywhere.

closed 2023-11-21 by Krosta8