prusa3d/Prusa-Firmware-Buddy #3495: [BUG] Connect not uploading file to printer when filename is too long

ErGeeBee, 2023-11-20:
**Printer type** - [MINI+]
**Printer firmware version** - [5.1.0-beta]
**Original or Custom firmware** - [Original]
**Optional upgrades** - [Filament Runout Sensor]

**Describe the bug**
When printing form Prusaslicer 2.7.0.rc1 using Prusaconnect, I can see the file in connect.prusa.com being uploaded to the printer but it sticks at around 50%.
When I later tried it another way, uploading the file directly to connect.prusa.com, I got the message that the filename was too long with the option to change the filename. After doing so it printed correctly.

**How to reproduce**
 I attached the 3mf & bgcode file.
1. open 3mf file in prusaslicer & upload & print to connect
2. Alternatively go to connect.prusa.com & drag the bgcode file directly

**Expected behavior**
If the filename is too long I expect the message from PrusaConnect after uploading it from Prusaslicer, with the option to change the filename.

**G-code**
  Please attach the G-code you had trouble with. This will make it easier for us to replicate the error.

**used .bgcode & 3MF file**
[files.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/13412999/files.zip)

ErGeeBee, 2023-11-24:
Somehow solved. 
Prusaslicer 2.7.0 now displays this error when filename is too long.
[image]

ErGeeBee, 2023-12-05:
The issue is back.
Running Prusaslicer 2.7.0 and Mini firmware 5.1.0. When uploading a file with long filename from Prusaslicer to Connect I get no feedback of the filename being too long. The file is just not printing. 

- On connect.prusa.com I see the file being uploaded the printer but the progress stops around 50%. 
- The file (with long name) is available in the 'Connect Files' tab. When starting the print for this file it gives me the message that filename is too long and a chance to fix it.

hillsandales, 2024-02-04:
I am also seeing this on the Prusa XL 2T machine (running f/w 5.1.2+13478). The names for XL files are often longer because the generated output file name includes information from more than one extruder (materials, etc.). The file with a long name says it uploads fine to PrusaConnect, but checking on it, it hangs at 50% on PrusaConnect, and doesn't upload to the printer. I took the same file and shortened the name some, and it uploaded to the PrusaConnect and the Prusa XL printer normally.

I never get an error message from PrusaSlicer, PrusaConnect, or the Prusa XL printer when this happens. I often forget about the file name character limit... A check when uploading the file to the XL from PrusaSlicer would be really helpful.

diegolm1, 2024-03-11:
I'm getting the same problem on the MK4 with the last firware, no error message but stuck at 50% on Connect. Shortening the name does the trick.

wafflecart, 2024-06-05:
Please fix it’s amateur that you can’t have long files or special characters.

danopernis, 2024-06-05:
Can you please verify if the issue is still present in newest versions of the firmware + connect? At the moment this issue looks quite outdated, it would be nice to have fresh info.

diegolm1, 2024-08-05:
Somebody should check if this continues to be an issue.

closed 2024-10-07 by ErGeeBee