prusa3d/Prusa-Firmware-Buddy #3497: [BUG] 5.1.0-beta Prusa XL 5 tool layer shift

a-pushkin, 2023-11-21:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - [XL 5 tool]

**Printer firmware version** - [5.1.0-beta]

**Original or Custom firmware** - [Original]

**Optional upgrades** - [None]

**USB drive or USB/Octoprint**
  USB

**Describe the bug**
  Slicing attached file with Prusa Slicer 2.7.0-rc1 in 0.32mm SPEED input shaper mode and printing would result in layer shift in X and Y directions by around 0.5mm. I've tried printing the same G-code 3 times and each time shift happens exactly once within the second block of the print but at slightly different heights.

  I'm attaching pictures of the wipe towers from these prints where layer shift is quite prominent.

  Slicing the same model without input shaping (0.20mm QUALITY) produces print without layer shifts. Verified by printing the same G-code twice.

  Nozzle diameter for all prints is 0.6mm. Filaments are PETG for all prints, all tools.

  Although, not as thoroughly documents, I recall different 5 tool print with input shaping that is less than 10mm high also having a single layer shift someplace inside the pritn.

**How to reproduce**
  Print the attached XLIS bgcode.

**Expected behavior**
  Model is printed without layer shifts.

**G-code**
  
[banding-test.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/13422928/banding-test.zip) contains bgcode for faulty IS and non-faulty regular print, 3mf file and photos of the wipe towers from bad and good prints.

**Crash dump file**
  No crash dump.

**Video**
  No video but photos are included in the zip file.

TravisWilder, 2023-11-21:
BTW .. do you know this:
https://www.youtube.com/watch?v=eZYngEnAgYs&t=1s&ab_channel=YGK3D

codingcatgirl, 2023-11-21:
8 Minutes of video for a very simple answer… oh god.

Anyway, this is a different issue.

a-pushkin, 2023-11-25:
Installed 5.1.0 release, reprinted the same G-code twice - layer shift is gone.

I'm guessing this is related to from[ 5.1.0 release notes](https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v5.1.0):

> Original Prusa XL - Current Input Shaper limitations

> In this firmware release, the Crash Detection feature is temporarily disabled on the Original Prusa XL because overly tight belts may trigger a false crash detection. We’re currently working on improvements in belt tension tuning and once it’s done, we’ll release an updated firmware that will re-enable crash detection on the XL. It will likely be accompanied by a new belt tuning app profile.

closed 2023-11-25 by a-pushkin