prusa3d/Prusa-Firmware-Buddy #3501: [BUG] No Missing Filament warning eh

mcstoufer, 2023-11-22:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - [MINI]
MK4 full upgrade kit

**Printer firmware version** - [e.g. 4.0.5, ...]
5.0.1+12089

**Original or Custom firmware** - [Original/Custom]
Original 

**Optional upgrades** - [e.g. Filament Runout Sensor]
N/A - Stock kit upgrade.

**USB drive or USB/Octoprint**
 Octoprint (1.9.3) printing model sliced with PS (PrusaSlicer-2.6.1+MacOS-x64-202309060812)
InputShaping enabled.

**Describe the bug**
Printer did not prompt for filament loading prior to actually printing the part.

**How to reproduce**
Purposefully initiated a print from Octoprint w/o any insertion of Filament into Extruder. There was never any error a/o alert to load the Filament. Start up sequence and print started normally.

**Expected behavior**
Somewhere in the startup sequence, the printer should have paused and requested the filament be loaded.

**G-code**
Attached to issue.

**Crash dump file**
No crash observed.

**Video**
Can provide if required.

mcstoufer, 2023-11-22:
[Falcon_support_V1_0.4n_0.2mm_PLA_MK4IS_4h31m.gcode.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/13432088/Falcon_support_V1_0.4n_0.2mm_PLA_MK4IS_4h31m.gcode.zip)

Sargastic, 2024-05-01:
Same problem here, with Octoprint 1.10.0, MK4 from kit, no added third party stuff, firmware 5.1.3.

Sargastic, 2024-07-01:
> This issue has been flagged as stale because it has been open for 60 days with no activity. The issue will be closed in 7 days unless someone removes the "stale" label or adds a comment.

Issue is still present in latest firmware.

jneilliii, 2024-07-17:
I'd like to point out that the proper approach for implementing this to be compatible with OctoPrint is to send the proper action commands on the serial connection when filament runout occurs. Specifically:

```
//action:prompt_begin Filament runout detected. Please choose how to proceed:
//action:prompt_choice Swap filament
//action:prompt_choice Home X/Y and pause print
//action:prompt_choice Abort print
//action:prompt_show
```

As it stands now the filament runout basically just blocks serial communication until the swap/load is completed and then starts accepting commands again. I learned this by working with someone on the OctoPrint Community forum and developed the plugin [Prusa Filament Runout](https://github.com/jneilliii/OctoPrint-PrusaFilamentRunoutMonitor) to at least get pause notifications to work with other plugins, ie OctoText by firing the "pause" event internally when a parked position is reported.