prusa3d/Prusa-Firmware-Buddy #3502: [BUG] PrusaLink Macintosh file selector should also include `bgcode` along with `gcode` as supported file type

JohnOCFII, 2023-11-22:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - MK4

**Printer firmware version** - 5.1.0-Beta

**Original or Custom firmware** - Original

**Optional upgrades** -

**USB drive or USB/Octoprint**
PrusaLink

**Describe the bug**
Using FireFox 119.01 on MacBook Pro with M1/arm. While drag and drop works, using the option to open the file selector to choose a file shows `bgcode` files as grayed out.  You CAN work around this by using the `Show Options` feature and switching to ALL, but I believe PrusaLink can/should be include `bgcode` as well as `gcode` as "expected" file types.

**How to reproduce**
Attempt to use PrusaLink to upload a file NOT via drag and drop, but choosing the option to use the local file selector to choose a file.

**Expected behavior**
Both `gcode` and `bgcode` file types should be selectable.

**G-code**
N/A

**Crash dump file**

N/A

**Video**
N/A - screen shot attached. [image]

bbrundert, 2023-11-26:
Same issue here on my MINI (running MINI_en_de_firmware_5.1.0.bbf), uploading from MacOS. 

I also tried a potential workaround. I renamed my PrusaSlicer 2.7 generated .bgcode file to .gcode. But the printer detects it is actually a .bgcode file with wrong extension. Unfortunately, the MINI does not offer to proceed anyways or rename the file to .bgcode. So the only workaround is to transfer the file via USB Stick.

drake127, 2023-11-26:
The same issue is present in final 5.1.0. A workaround on Windows is to select "All Files (*.*)", but it's very unfriendly.

eugali, 2023-11-30:
Hey guys, the workaround is to press "inspect" in the file uploader and change the accept prop to .bgcode
[image]
[image]

JAngerCAN, 2023-12-18:
I see the same problem with Safari browser on Mac. You can't select a .bg file to upload when clicking "click to choose a GCODE file". A workaround is to drag the .bg file from the finder into the upload window - then it uploads. The window says you can drag files in but perhaps it's not obvious you can drag a .bg file in this way.

danopernis, 2024-06-05:
Duplicates https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/3406

closed 2024-06-05 by danopernis