prusa3d/Prusa-Firmware-Buddy #3503: [BUG] Filament sensor in MINI+ with FW 5.1.0-BETA+13358 does not inform about lack of filament during printing.

mixer3d, 2023-11-22:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - [MINI]
MINI+

**Printer firmware version** - [e.g. 4.0.5, ...]
5.1.0-BETA+13358

**Original or Custom firmware** - [Original/Custom]
Original

**Optional upgrades** - [e.g. Filament Runout Sensor]
Filament Runout Sensor

**USB drive or USB/Octoprint**
 USB flash drive

**Describe the bug**
 During printing when the filament runs out, the filament sensor does not recognize the end of the spool and does not prompt to load the filament, and unload the last piece of the previous one. 
Important is that when you load the filament for the first time it recognized, that it's loaded and you can see the prompt with filament type selector, only during printing filament sensor is not checking if the filament is out.

**How to reproduce**
Print something and cut the filament just before the sensor. I tried it two times with two different filaments to confirm this behaviour.

**Expected behavior**
 When the filament will end, the printer should print the prompt for loading the new spool, and reload the leftover of the old one.

**G-code**
every print is affected.

**Crash dump file**
  Please attach the crash dump file. This will make it easier for us to investigate the bug.

**Video**
  Please attach a video. It usually helps to solve the problem.

thijstriemstra, 2023-11-22:
Strange, it's working fine here (although I only use Octoprint, not USB).

mixer3d, 2023-11-25:
Hi' i wanted to update that with the latest stable version of the FW `5.1.0`everything works back as expected. Thank you.

closed 2023-11-25 by mixer3d