prusa3d/Prusa-Firmware-Buddy #3508: [BUG] 5.1.0 beta makes Octoprint/OctoPi unresponsive (print continues)

dirkvranckaert, 2023-11-22:
**Printer type** - MK4

**Printer firmware version** - 5.1.0 Beta

**Original or Custom firmware** - Original

**Optional upgrades** - None

**USB drive or USB/Octoprint** Octoprint

**Describe the bug**
 When a print started with OctoPrint is running all of the sudden my OctoPi becomes unresponsive. The good thing however is that my print continues, so OctoPrint is not dead!
But I'm running several pi's with OctoPrint, they have the exact same setup, plugins,... Never had this issue. Also this specific PI did not suffer from this issue until upgrading to the BETA 5.1.0!

Fact list:
- It does not happen on every print, also did not yet try the same GCode on which it happened agsin a second time.
-  The builds complete normall
- However afterwards only way to get my OctoPi alive again is by force rebooting it (power cycle)
- As started before it only started happening since installation of the 5.1.0 Beta
- I have not installed any plugin updates, new plugins or OctoPrint updates since last Friday, only using this beta since last Monday (3 days ago), so OctoPrint should be stable

**How to reproduce**
  Start a print from OctoPrint

**Expected behavior**
  OctoPrint should stay alive as well as the printer so they end the print gracefully and that progress can be tracked from OctoPrint

**G-code**
  It happened again just now, the print is still running for several hours so only after it finished I can extract the latest exact GCode file from OctoPrint!

The current print it happened too, when that one finishes I will try to

- Get the logs from OctoPrint from when it stoped working for me
- Get the GCode on which it failed
- Need the current print a second time so will restart it to observe if it can be easily reproduced

**Crash dump file**
  N/A

**Video**
  N/A

dirkvranckaert, 2023-11-23:
Update:

- The print job once again ended with success, OctoPrint captured the data of my print (start and end) correctly as well
- I have the OctoPrint logs available from the start to just after the end of my print (not that you start seeing errors from around 2023-11-22 21:48:46): 
[octoprint.log](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/13446495/octoprint.log)
- The GCode on which it failed for this print: [Hinge_0.4n_0.2mm_ASA_MK4IS_2h22m.gcode.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/13446500/Hinge_0.4n_0.2mm_ASA_MK4IS_2h22m.gcode.zip)
- Restarted OctoPrint (power cycle) and it's back to life. Second print of that same instance starting right now

SebastianOsinski, 2023-11-23:
Just had the same thing happen with my Mini on production 5.1.0

dirkvranckaert, 2023-11-24:
Update: I was not able to reproduce it with the same GCode on a second print..

dirkvranckaert, 2023-11-24:
Suffered the issue again during 16hours print. Very annoying that I can't track the printer anymore and even need to reboot OctoPrint to get it alive again... The entire point of having OctoPrint integration goes away...
Did not yet upgrade to production 5.1.0 since the printer was already running when the release got pushed out. Will update this afternoon but since @SebastianOsinski had the same issue already I guess it won't be fixed :-(

closed 2024-07-01 by github-actions[bot]