prusa3d/Prusa-Firmware-Buddy #3510: Printer no longer shows a preview of the part being printed prior to starting a print.

BrienKing, 2023-11-23:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - XL 2TH

**Printer firmware version** -5.1.0 Beta

**Original or Custom firmware** - Original

**Optional upgrades** - None

**USB drive or USB/Octoprint**
USB

**Describe the bug**
Prior to the new (and nice) feature that allows you to select filaments, joining, etc.. it would show you a preview of what you are about to print.   It no longer shows that.

This is an Image I found on the internet that shows the preview I'm referring to.
[image]

Now starts on this screen:
[image]

**How to reproduce**
Print something....

**Expected behavior**
I want it to show the Part that is about to print and allow me to continue or cancel.  If I continue then it should go to the new filament selection menu.

tkircher, 2023-11-24:
You have to upgrade PrusaSlicer to generate the correct image file format, now that PNG isn't decoded. It's in the release notes.

BrienKing, 2023-11-24:
I'm on the latest Slicer and almost latest firmware (don't want to upgrade to the latest firmware because it disables Crash Detection).

tkircher, 2023-11-24:
> I'm on the latest Slicer and almost latest firmware (don't want to upgrade to the latest firmware because it disables Crash Detection).

Go to Printer Settings > General > Firmware > G-code thumbnails and let me know what formats are listed there

BrienKing, 2023-11-24:
Well, I never changed it, so not sure why it didn't get updated...

Incorrect settings:
[image]

Reset to "Default"
[image]

Will have to test with that.

carlocarlo91, 2023-11-24:
Hi guys, same thing for me with Prusa mini with firmware 5.1.0 and Prusa slicer updated to 2.7.0

[image]

I don't see the new format settings

VinceKezel, 2023-11-24:
I just went to firmware 5.1.0 for my Mini+ and PrusaSlicer 2.7.0 ... and I was confused too, but I see NEWLY sliced and exported files show the thumbnails.

I still think it's a bit of an issue that the display on the Mini+ does not show thumbnails for previous versions of PrusaSlicer exports. I wish it did.

sjbrouillard, 2023-11-25:
I would agree that it's an issue that the old PNG format is no longer supported. Is there a technical reason that both formats are not supported?

I upgraded to firmware 5.1 for Octoprint support, but am still having some issues there. I kept the output to PNG so that Octoprint file previews would remain in place, but now that I still have to print via USB, the previews have disappeared if I don't re-slice using the new format.

Is there a workaround? Can I output both types of thumbnails?

tkircher, 2023-11-25:
> I would agree that it's an issue that the old PNG format is no longer supported. Is there a technical reason that both formats are not supported?

There's an explanation here:

https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v5.1.0

sjbrouillard, 2023-11-25:
@tkircher - Ugghhh! Clearly my reading comprehension has taken an extreme hit. As they say, everything is a tradeoff. Thanks for pointing out the obvious.

epetit91, 2023-11-26:
Hi,
with the settings below, the thumbnail is not displayed on the printer screen.
What setting is missing for the thumbnail to display?
Thank you

Printer type - MK4
Printer firmware version -5.1.0
Original firmware 
USB drive
PrusaSlicer 2.7.0
MacOS 10.12.6
[image]
[image]

Yopik23, 2023-11-27:
Hi, I have solved missing preview on the printer with saving default profile (Original Prusa MINI & MINI+ Input Shaper (Alpha) as a copy with these setting:
[image]
 the default one was:
[image]

So you have to rewrite G-Code thumbnails to:
16x16/QOI, 220x124/QOI, 200x240/QOI, 640x480/PNG

Printer type - MINI/MINI+
Printer firmware version -5.1.0
Original firmware 
USB drive
PrusaSlicer 2.7.0

BrienKing, 2023-11-27:
This is a configuration issue with the printer profile and is not a bug.  Closing.