prusa3d/Prusa-Firmware-Buddy #3511: [QUESTION] MINI firmware file names are confusing

reedy, 2023-11-23:
https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v5.1.0

[image]

What does `en_cs` mean? It has en and cs?

They are not named liked that in https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v4.4.1 or https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v5.1.0

The release notes make no mention of why they are named differently (with language codes?) and necessarily, how someone should "pick" which one they should use...

JzPii, 2023-11-24:
+1 confusing

stereoprologic, 2023-11-24:
I installed en_de and got English and German. So I would assume en_cs is English and Czech, with Prusa being a Czech company. ;-)
I guess it's probably an internal memory limitation of the buddy board.

reedy, 2023-11-24:
Almost certainly, yeah, but it's not even documented in the announcement or anywhere else.

It's also not like that in the previous beta... Unless the betas don't include l10n/i18n files (which is possible, but I don't know, I don't use them), it seems odd there is this change in behaviour.

It's no better on https://help.prusa3d.com/downloads which when you grab https://www.prusa3d.com/downloads/firmware/MINI_firmware_5.1.0.zip gives you all of them, and then nothing to document which you should use.

While many 3D printers may be technical, and understand it from the filenames, many may not be. It's also less clear which you "should" use if you just want en.

It's easily fixed by Prusa, by just documenting it, it's the fact that they haven't.

rcmaniac25, 2023-11-25:
Can agree with this issue... I went to do a quick update of firmware and instead went looking to see if there was alternative language codes that I wasn't familiar with. "Maybe en_cs is a typo of en_us or something? Is one of these Czech?" installed and all good, but just a moment I didn't expect I needed to try guess and check for firmware installs.

robloh, 2023-11-25:
When you download it from https://help.prusa3d.com/downloads it just gives you a zip with all the MINI firmwares in it. Super confusing indeed.

reedy, 2023-12-14:
[image]

Looks like they've silently changed them, without any comment here...

rcmaniac25, 2023-12-14:
Good that they changed it. They have their own ticket system, so it's possible that were planning on fixing it and never saw this ticket.

OJFord, 2024-03-15:
For what it's worth I never saw it in the short form, and found this issue because what's confusing to me is that I only want/need English, and there's nothing (that I can see) to confirm that it's essentially an arbitrary choice, they all have English, just plus a second (variable) language.

I assume it's a space constraint that they're separate, but then nothing stopping a `MINI_english_firmware_x.y.z.bbf` version? To be clear I'm not 'anti those languages' or whatever(!) - I just think that would clarify it a lot, I wouldn't be here now, I would've just downloaded the English one knowing what it was.

dstulken, 2024-03-17:
> what's confusing to me is that I only want/need English, and there's nothing (that I can see) to confirm that it's essentially an arbitrary choice, they all have English, just plus a second (variable) language.

Prusa being located in the Czech Republic, I just presume that the "english_czech" version has had the most testing and the most eyes on it. 🤣

danopernis, 2024-06-03:
Hi folks,

MINI builds are split according to their language due to limited FLASH size on the Buddy board, otherwise there will be just one firmware file as is the case for other printers. English is always present at the moment due to some technical reasons, but it may be eventually separated too, in order to save even more FLASH for firmware features.

Since v5.1.0 the `.bbf` filename contains full language name instead of ISO language code to prevent confusion.

If I understand it correctly, the complaint here is that the documentation is lacking. Where would you expect documentation to be located in order for you to find it? I can forward your request to the relevant department. As for this particular repository, I don't see any need for such detailed documentation, as github is mostly used by people with at least some technical background.

reedy, 2024-06-03:
Noting that things have changed since the task was originally filed. The lack of input by Prusa, but then making changes (maybe due to the task) is a little odd.

And while, yes, GitHub will potentially be a more technical audience... In the previous version (ie language codes), you can't guarantee that means anything to anyone. The current situation is slightly improved, but not by much.

If we look at the non technical users.. They may end up on a page such as https://help.prusa3d.com/downloads/mini/firmware, which, for example links to https://www.prusa3d.com/downloads/firmware/MINI_firmware_6.0.1.zip

In that zip is 6 firmware files. Most people only need one of them. The help page doesn't tell anyone which to use (especially if they only want English). There's no mention of it in the zip (no README or similar). I could see people placing all files onto a USB stick.

There's also an argument about download bloat; less of an issue generally in todays internet (I'm sure there's people out there where those MB are slow and/or expensive)... But when people only need 1 file (in most cases); 2MB compresses to 1.1MB, rather than 6MB...

[image]

https://help.prusa3d.com/tag/mini links to https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v6.0.1 for "More info"; so it's very easy for a non technical user to end up here... Though, from the tag link, they can download the individual language one...

reedy, 2024-06-03:
> English is always present at the moment due to some technical reasons, but it may be eventually separated too, in order to save even more FLASH for firmware features.

Maybe get ahead of that curve? :)

danopernis, 2024-06-06:
This was also requested in https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/3511