prusa3d/Prusa-Firmware-Buddy #3514: [BUG] MINI+ 5.1.0 Firmware only available languages English or Polish.

maxwxyz, 2023-11-24:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - MINI*

**Printer firmware version** - 5.1.0

**Original or Custom firmware** - Original

**Optional upgrades** - none

**USB drive or USB/Octoprint** - USB flash drive
 

**Describe the bug**
  After updating to the current firmware 5.1.0, only two languages are available in the settings (English and German). Previously the Printer was set to German and all Language files were present during upgrading the firmware.

**How to reproduce**
  see above

**Expected behavior**
  All Languages should be available again.

**G-code**
  none

**Crash dump file**
  no crash

**Video**
  none

CuriousTimo, 2023-11-24:
I do not think this qualifies as a bug.

If you look at [the release](https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v5.1.0) Prusa has clearly chosen to change to multiple dual language firmware bbf files for the MINI.

My guess is that this is needed to make that all the new features still fit in the MINI's memory.

maxwxyz, 2023-11-24:
Yes might be, the instructions said to copy all files to the flash drive. Did that and inserted and flashed. My printer was in German before and now can only be toggled between English and Polish. There is a German file provided but the printer seems to ignore it.

CuriousTimo, 2023-11-24:
So even if you install [this firmware](https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/download/v5.1.0/MINI_en_de_firmware_5.1.0.bbf) you are still missing German?

Or do yo mean that it will not install?
Then try this:

```
To force a firmware update on the Mini:
* Push the reset-button, immediately followed by double-pressing the selector-knob.
* On the new screen, confirm the update by selecting the "FLASH" button and pressing the knob.
```

maxwxyz, 2023-11-24:
Worked. The instructions for flashing the firmware should be changed, as it states to copy **all** files to the flash drive. The printer will load the first one I guess (polish in my case).

closed 2023-11-24 by maxwxyz