prusa3d/Prusa-Firmware-Buddy #3515: [BUG] MINI Print FAN Error - while ljumping from 4.4.0 to 5.1.0

Moyrn, 2023-11-24:
**Printer type** 
MINI

**Printer firmware version** - 
5.1.0

**Original or Custom firmware**
Original

**Optional upgrades** 
Revo & Bondtech E

**USB drive or USB/Octoprint**
USB

**Describe the bug**
I was on 4.4.0 all fine since months.
I wanted to test input shaper
So i did the 4.4.1 upgrade as recommended, instantly followed by 5.1.0
All went fine
(My settings for Bondtech E were still there fine Z190 E412 Reverse)

First print failed after layer 1
>> impossible to print : print fan error <<

Self test is KO on the printer fan, all is ok
the printer fan is running fine ; and i printed for years all fine until now
i tried a factory reset... no change

if i remove the  fan check : i can print fine

**How to reproduce**
tough to say... from 4.4.0 to 5.1.0

**Expected behavior**
no error

**G-code**
no need - fails even at self test

**Crash dump file**
N/A

**Video**
no need

-----------
Note : related to https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/3260
except that self test is not OK

reedy, 2023-11-24:
FWIW, https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v5.1.0 does say

>In the case of the MINI/MINI+ printer make sure you are running either the previous 5.1.0-alpha2 or at least firmware 4.4.1. Once the 4.4.1 update is installed, proceed to flashing the 5.1.0 firmware. The upgrade might take longer than usual.

Moyrn, 2023-11-24:
yes, that is exactly what i did
mentionned there :"So i did the 4.4.1 upgrade as recommended, instantly followed by 5.1.0"

reedy, 2023-11-24:
"while ljumping from 4.4.0 to 5.1.0" makes it sound like you went straight to 5.1.0

Moyrn, 2023-11-24:
oh in the title - you are right, i will fix that

Moyrn, 2023-11-24:
During test, max RPM=7928 for the printfan... then it fails
I do not know if this value is sufficient, or not. or if this is the problem

Moyrn, 2023-11-24:
I removed the print fan and the self-test was still KO - though RPM still reporting
I then noticed the yellow wire was quasi loose. touching it and it was broken
i then resoldered it, and made another self test in free air : passed OK - reporting something like 7800RPM
i mounted it back fully : passed OK
I renabled "check fan"
I launched a print : passed OK

Moyrn, 2023-11-24:
So possibly a pinched or loose yellow cable in my case. Firmware is just performing tests which were not there before.

closed 2023-11-24 by Moyrn