prusa3d/Prusa-Firmware-Buddy #3518: [BUG] Firmware 5.1.0 is not available through Prusa Connect

DominikPalo, 2023-11-25:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - MK4

**Printer firmware version** - 5.1.0-BETA+13358

**Original or Custom firmware** - Original

**Describe the bug**
  The[ final version of firmware 5.1.0]([Verze 5.1.0 je nyní dostupná](https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v5.1.0)) was released two days ago, but it is still not visible in the _Prusa Connect -> Available firmware versions_ dropdown
  
  
[image]

**How to reproduce**

1. Visit https://connect.prusa3d.com/
2. Select a supported printer (e.g. MK4)
3. Got to "Settings"
4. Click on the ⚡ (Select firmware to flash) icon
5. The dropdown doesn't contain the firmware 5.1.0

**Expected behavior**
Firmware 5.1.0 is displayed in the dropdown

darraghbr, 2023-11-25:
I think it's looking for versions of the firmware that is currently on the USB stick, certainly when I look at mine it shows this:

[image]

After I removed the .bbf file from my USB drive.

DominikPalo, 2023-11-25:
@darraghbr In my case, I don't have any firmware files on my USB stick (I always delete the .bbf file after flashing it), I even tried to disconnect the USB stick and the list was the same.

I inspected the network requests and the list of available firmware versions is obtained by calling the `https://connect.prusa3d.com/app/firmwares?printer_uuid={printer_id}` endpoint and the URLs in the response point directly to GitHub:

```
{
    "files": [
        {
            "type": "FIRMWARE",
            "name": "XL_firmware_4.6.1.bbf",
            "display_name": "XL_firmware_4.6.1.bbf",
            "size": 3500820,
            "hash": "KBMRrpZhV4V6SsInjuHTGIPz_F8.",
            "team_id": 0,
            "printer_type": "3.1.0",
            "uploaded": 1686651560,
            "meta": {
                "version": "4.6.1",
                "release": "stable",
                "state": "outdated"
            },
            "release_url": "https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v4.6.1"
        },
        {
            "type": "FIRMWARE",
            "name": "MK4_MK3.9_firmware_5.0.0.bbf",
            "display_name": "MK4_MK3.9_firmware_5.0.0.bbf",
            "size": 3911588,
            "hash": "MeP6b478klBywgyhihs-x8zlRt0.",
            "team_id": 0,
            "upload_id": 93236,
            "printer_type": "1.4.0",
            "uploaded": 1695064654,
            "meta": {
                "version": "5.0.0",
                "release": "stable",
                "state": "outdated"
            },
            "sync": {
                "synced_by": {
                    "connect": "0.26.2"
                },
                "synced": 1695064654,
                "source": "CONNECT"
            },
            "release_url": "https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v5.0.0"
        },
        {
            "type": "FIRMWARE",
            "name": "XL_firmware_4.7.0.bbf",
            "display_name": "XL_firmware_4.7.0.bbf",
            "size": 3572676,
            "hash": "OQA2jGbXpQo7O-YB_CfYrccS9UE.",
            "team_id": 0,
            "printer_type": "3.1.0",
            "uploaded": 1687950731,
            "meta": {
                "version": "4.7.0",
                "release": "stable",
                "state": "outdated"
            },
            "release_url": "https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v4.7.0"
        },
        {
            "type": "FIRMWARE",
            "name": "MK4_firmware_4.7.2.bbf",
            "display_name": "MK4_firmware_4.7.2.bbf",
            "size": 3501808,
            "hash": "OZaG0BNAcmUdmP8QXmQvfX_a0RY.",
            "team_id": 0,
            "upload_id": 31400,
            "printer_type": "1.4.0",
            "uploaded": 1691993773,
            "meta": {
                "version": "4.7.2",
                "release": "stable",
                "state": "outdated"
            },
            "release_url": "https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v4.7.2"
        },
        {
            "type": "FIRMWARE",
            "name": "MK4_firmware_4.6.4.bbf",
            "display_name": "MK4_firmware_4.6.4.bbf",
            "size": 3445528,
            "hash": "plU_8bts_ce38EMOebcYrfo9zAI.",
            "team_id": 0,
            "printer_type": "1.4.0",
            "uploaded": 1686651564,
            "meta": {
                "version": "4.6.4",
                "release": "stable",
                "state": "outdated"
            },
            "release_url": "https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v4.6.4"
        },
        {
            "type": "FIRMWARE",
            "name": "XL_firmware_4.7.2.bbf",
            "display_name": "XL_firmware_4.7.2.bbf",
            "size": 3574428,
            "hash": "tXSO1Vr-AKA2iQaGkUzmnVsvgjA.",
            "team_id": 0,
            "upload_id": 31399,
            "printer_type": "3.1.0",
            "uploaded": 1691993766,
            "meta": {
                "version": "4.7.2",
                "release": "stable",
                "state": "supported"
            },
            "release_url": "https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v4.7.2"
        },
        {
            "type": "FIRMWARE",
            "name": "MK4_MK3.9_firmware_5.0.1.bbf",
            "display_name": "MK4_MK3.9_firmware_5.0.1.bbf",
            "size": 3915644,
            "hash": "Vgfes78JoOIC7KlmDZdKYruTho0.",
            "team_id": 0,
            "upload_id": 243409,
            "printer_type": "1.4.0",
            "uploaded": 1699618929,
            "meta": {
                "sem_ver": "5.0.1",
                "build_no": 12089,
                "bbf_version": 2,
                "version": "5.0.1",
                "release": "stable",
                "state": "supported"
            },
            "sync": {
                "synced_by": {
                    "connect": "0.31.4"
                },
                "synced": 1699618929,
                "source": "CONNECT"
            },
            "release_url": "https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v5.0.1"
        }
    ]
}
```

darraghbr, 2023-11-25:
Ah, may be a difference between the MINI and the MK4 in that case,

JeremiahGillis, 2023-11-25:
Same issue with my XL.

[image]

maverickk89, 2023-12-15:
me too with latest 5.1.2 released few days ago, that isn't available on my MK4 and MINI.
[image]

seanmalloy, 2023-12-15:
For my MINI+ running firmware version 5.1.0 I do not see any option to update to firmware 5.1.2 yet.

Browsing to `https://connect.prusa3d.com/app/firmwares?printer_uuid={printer_id}` here is what I see:
```
{
  "files": [
    {
      "type": "FIRMWARE",
      "name": "XL_firmware_5.1.0.bbf",
      "display_name": "XL_firmware_5.1.0.bbf",
      "size": 4024056,
      "hash": "0BJiIyZfokPuGe1ZBAphe01q964.",
      "team_id": 0,
      "upload_id": 355382,
      "printer_type": "3.1.0",
      "uploaded": 1701777688,
      "meta": {
        "sem_ver": "5.1.0",
        "build_no": 13455,
        "bbf_version": 2,
        "version": "5.1.0",
        "release": "stable",
        "state": "supported"
      },
      "sync": {
        "synced_by": {
          "connect": "0.33.1"
        },
        "synced": 1701777688,
        "source": "CONNECT"
      },
      "release_url": "https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v5.1.0"
    },
    {
      "type": "FIRMWARE",
      "name": "MK4_MK3.9_firmware_5.1.0.bbf",
      "display_name": "MK4_MK3.9_firmware_5.1.0.bbf",
      "size": 3957812,
      "hash": "28v1xXgrP0CXoNVQb6Js25XkwKo.",
      "team_id": 0,
      "upload_id": 355383,
      "printer_type": "1.4.0",
      "uploaded": 1701777693,
      "meta": {
        "sem_ver": "5.1.0",
        "build_no": 13455,
        "bbf_version": 2,
        "version": "5.1.0",
        "release": "stable",
        "state": "supported"
      },
      "sync": {
        "synced_by": {
          "connect": "0.33.1"
        },
        "synced": 1701777693,
        "source": "CONNECT"
      },
      "release_url": "https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v5.1.0"
    },
    {
      "type": "FIRMWARE",
      "name": "XL_firmware_4.6.1.bbf",
      "display_name": "XL_firmware_4.6.1.bbf",
      "size": 3500820,
      "hash": "KBMRrpZhV4V6SsInjuHTGIPz_F8.",
      "team_id": 0,
      "printer_type": "3.1.0",
      "uploaded": 1686651560,
      "meta": {
[... 155 lines cut ...]
      "hash": "Vgfes78JoOIC7KlmDZdKYruTho0.",
      "team_id": 0,
      "upload_id": 243409,
      "printer_type": "1.4.0",
      "uploaded": 1699618929,
      "meta": {
        "sem_ver": "5.0.1",
        "build_no": 12089,
        "bbf_version": 2,
        "version": "5.0.1",
        "release": "stable",
        "state": "outdated"
      },
      "sync": {
        "synced_by": {
          "connect": "0.31.4"
        },
        "synced": 1699618929,
        "source": "CONNECT"
      },
      "release_url": "https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v5.0.1"
    }
  ]
}
```

seanmalloy, 2024-04-22:
Following up on this I was able to update my MINI+ printer to firmware 6.0.0 a few days after the GA firmware release using Prusa Connect. So it seems that for the MINI+ this issue is fixed.

danopernis, 2024-04-23:
Thanks for the report, I am closing this issue.

maverickk89, 2024-04-23:
I opened it because after several days it was not shown up in prusa connect, I would expect that within few hours it should be visible, I'll check with the next XL release and in case I will open another ticket.