prusa3d/Prusa-Firmware-Buddy #3519: [BUG] No USA Language version availble for download in the the latest release for the MINI

shawnchr, 2023-11-25:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - [MINI]

**Printer firmware version** - 5.1.0

**Original or Custom firmware** - [Original/Custom]

**Optional upgrades** - [e.g. Filament Runout Sensor]

**USB drive or USB/Octoprint**
  Please let us know if you print via USB flash drive or USB/Octoprint

**Describe the bug**
  A clear and concise description of what the bug is.

**How to reproduce**
  Please describe steps to reproduce the behavior.

**Expected behavior**
  A clear and concise description of what you expected to happen.

**G-code**
  Please attach the G-code you had trouble with. This will make it easier for us to replicate the error.

**Crash dump file**
  Please attach the crash dump file. This will make it easier for us to investigate the bug.

**Video**
  Please attach a video. It usually helps to solve the problem.

brettvitaz, 2023-11-25:
They all are English with another language included.

https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/download/v5.1.0/MINI_en_cs_firmware_5.1.0.bbf will work perfectly for you.

cjshaker, 2023-11-25:
Shawn:

As brettvitaz said, all of the firmware images include English. For example, "MINI_en_es_firmware_5.1.0.bbf", includes English "_en_" and Spanish "_es_".

Chris Shaker

closed 2024-07-01 by github-actions[bot]