prusa3d/Prusa-Firmware-Buddy #3520: [BUG] MK4 reboots into a blue screen when changing filaments

Zappes, 2023-11-26:
**Printer type** - MK4

**Printer firmware version** - 5.1.0 "stable" 🤡

**Original or Custom firmware** - Original

**Optional upgrades** - none

**USB drive or USB/Octoprint**
Not relevant here, this happened before printing

**Describe the bug**
When new filament is loaded into the print head, the printer immediately reboots and shows a blue screen after the boot loader is through. This seems to happen to other people as well: https://forum.prusa3d.com/forum/english-forum-original-prusa-i3-mk4-hardware-firmware-and-software-help/filament-changes-causes-printer-to-crash

**How to reproduce**
Turn on the printer, unload filament, push new filament into the extruder.

**Expected behavior**
The printer loads the new filament

**G-code**
n/a

tkircher, 2023-11-26:
I just spent about fifteen minutes swapping filaments a bunch of different ways and couldn't trigger this. Can you be more specific about the steps you take?

Zappes, 2023-11-26:
I did exactly what it said above: Turn on the printer, unload filament, push new filament into the extruder.

tkircher, 2023-11-26:
So the printer starts up, you turn the knob clockwise twice to highlight filament, then press the knob in to select it, turn the knob clockwise twice to highlight 'Unload filament', or do you select 'Change filament'? What, exactly, after that?

Zappes, 2023-11-26:
The first thing I did after startup was navigating to the menu item "filament/unload filament" and I pressed the nob. The filament unloading process then commenced as expected. After removing the filament, I inserted a new string of filament into the extruder. At the point where the display would normally show a message like "filament loading" or something similar, the printer rebooted and the bootloader screen including the progress bar was shown. After completing the boot process, a bluescreen was shown, but I forgot to make a picture. After pressing the reset button, the printer booted up normally, but right after the bootloader, it showed me a screen saying something like "a collision was detected, save to USB?" which I answered with "no". I had pulled out the filament during the reboot, so I pushed it in again. The loading process did now work exactly as expected.

After that I printed a model without issues. When the print was done, I did not immediately remove the sheet as I was on the phone. The printer then suddenly crashed into a bluescreen with some wild beeping and a bluescreen with an error message about the watchdog timer. This additional crash is probably completely unrelated, but for the sake of completeness I wanted to mention it.

Zappes, 2023-11-26:
I am at a point where I think that I need to perform a factory reset on the printer and hope that it helps - but I'll hold off on that just in case you need me to perform any tests before doing so. Please let me know if I should reset the MK4 or wait for further instructions. :)

tkircher, 2023-11-26:
Strange. I followed your instructions exactly and couldn't trigger it. Definitely some kind of glitch. I recommend getting a brand new USB stick, reset your printer and reload the latest firmware, and download the latest prusaslicer again and reinstall that for good measure. Maybe also check to make sure all the connectors on the main board are seated, and on the PSU as well.

Zappes, 2023-11-27:
I had some other issues that broke my will... Reverted to 5.0.0 and the printer works 100% again.

lebalz, 2024-04-16:
I have the same issue with the latest firmware 5.1.3 :(

closed 2024-07-09 by github-actions[bot]