prusa3d/Prusa-Firmware-Buddy #3521: [BUG] Y-axis calibration fails

Paja-git, 2023-11-26:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

MK4

5.1.0

Original

USB drive

The Y-axis calibration fails. The status line shows y=211mm.  I downgraded to 5.0.0, performed the full Self check -> everything passed.

Zappes, 2023-11-26:
Exactly the same here. I had already taken the beast out of the enclosure, checked all the bearings, tensioned the belts - and then it dawned on me that this is not a problem of my printer, but another serious problem of the 5.1 firmware.

wronskidawid, 2023-11-26:
Had the same issue with belt tension setting at 95. When I adjust the belt tension setting to near 98-99 printer pass the y-axis test. Belt need to be very tight.

Zappes, 2023-11-27:
Well, I got desperate and downgraded to 5.0.0. Self-test passed on 5.0.0, no issues with the Y axis. I will not touch 5.1 again...

dr-welt, 2023-11-27:
Same here. I tired also resetting to factory defaults like suggested on reddit. Tried multiple times. Y axis moves to the very front bumps into end. - Calibration failed. Going back to the 5.1.0. alpha. Acutally it behaves the same - leaving the bed after bumping into the limit covering the display - BUT the test passes.

pjcar2000, 2023-11-27:
I have same issue with Y axis moving to very front and failing calibration. Downgrading to 5.0.1.

pjcar2000, 2023-11-27:
@tkircher  so I guess you're saying that Prusa improperly assembled my MK4?

Silvenga, 2023-11-27:
For me, after upgrading to 5.1 GA, calibration randomly succeeds on the Y axis (meaning it randomly fails too). Same symptoms as @pjcar2000 where the plate moves to the front (after successfully calibrating the start) and immediately fails (without the normal retry).

(my printer was factory calibrated, working fine on 5.1.0. Alpha)

dr-welt, 2023-11-28:
I tightened the Y belt all the way to the limit in the tuner app. No change. 
Then I removed the front bracket, holding the guide left rod which is the end stop, when the LMU8 hits it. Also no change. 
From this, I figured out the "movement range" might  be too large on my printer. In the manual they tell you to assemble the LMU8 slight aligned to the inside. Which results in probably 1mm more travel. I moved the "front" LMU8 to the center of the bracket. This did the trick. Manual page here: (https://help.prusa3d.com/guide/7-y-carriage-heatbed-assembly_438310#438972)
Would be super nice to see WHY the self-test fails in the first place. Guess I could have looked in the source then. 

[image]
[image]

Zappes, 2023-11-28:
The interesting bit here is the fact that calibration fails on machines that comply exactly with Prusa's specifications - mine, for example, is factory-assembled, not built from a kit. This definitely can't be categorized as "user errror" or "bad assembly". It's a bug in the calibration routine, plain and simple.

wgentine, 2023-11-29:
same here.... Y-axis calibration fails and then keeps bashing the frame for while.... happening before prints randomly for no reason... never had any issue before 5.1.0.

sphxdr, 2023-12-05:
Another one here! I never had a Y-axis issue on my MK4 kit on previous firmware! Although it prints very good on 5.1.0 it was aware of this self test issue. Downgrading again to 5.0.1 seems to solve this self test issue. I don’t disagree with @tkircher about fine assembly but if Prusa wants to make the firmware more strict or we should do as @dr-welt proposes I would like to be told so from them!

Thank you!

tkircher, 2023-12-05:
> Another one here! I never had a Y-axis issue on my MK4 kit on previous firmware! Although it prints very good on 5.1.0 it was aware of this self test issue. Downgrading again to 5.0.1 seems to solve [...]

Agreed, Prusa needs to mention it at the very least. And they definitely didn't.

bryn51, 2023-12-11:
Well, at the very least I'd Like prusa research to acknowledge the issue and say when they are going to fix it. It seems to me that 2 weeks is long enough time for them to make such a fix if they adopted it as a priority.  Meanwhile they seem content to leave users in limbo, having advertised new features of 5.1 FW, but folks unable to take advantage of it, because they had to downgrade.  Really this is not good enough.

bryn51, 2023-12-11:
> Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.
Yeah sure.  I did that and nothing came up.  The search algorithm on this site sucks OK. Don't blame the users, blame the original developers that made a site for reporting bugs that cant find previous reports.  Instead talk about creating a mechanism that allows similar bug reports to be consolidated into 1.  Or bring it into the current day and age by adding an AI.

zoltan-l, 2023-12-11:
In the past there did at least to add a printer type label to the issue so you could preselect when searching. Nowadays you have to browse everything to see if it is related to your printer type (MINI, MK4,XL).

bildzeitung, 2023-12-14:
> I tightened the Y belt all the way to the limit in the tuner app. No change. Then I removed the front bracket, holding the guide left rod which is the end stop, when the LMU8 hits it. Also no [...]

I hit the same issue and went back and left a  comment on the build instructions for the MK3 -> MK4 upgrade. At least it’ll be a heads up for whoever hits it next.

BID3D, 2023-12-18:
I can verify this bug on my MK4? Please escalate a fix.  Belt tension and y Axis bearings are all within build recommendations. Increases or decreases to belt tension did not resolve the self test failure.

sascape, 2023-12-19:
Ich have the same problem in 5.1.2.

bryn51, 2023-12-20:
> Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.
>
> MK4
>
> 5.1.0
>
> Original
>
> USB drive
>
> The Y-axis calibration [...]

Version 5.1.2 of the Mk4 firmware is recently released.  I have installed the new version and I have verified that this reported issue remains.  I am now contacting Prusa Support to request an explanation as to why this issue has not attained a high priority to be fixed, since it seems likely its experienced by many Mk4 owners.

bryn51, 2023-12-20:
OK, so 5.1.2 version of firmware arrived. I installed and tested it. The reported issue with Y axis failing self test remained.

I got onto Prusa Support (Andres). While calming listening to me harangue about why this issue had not been fixed, he calmly suggested I look at alignment of the Y axis bearings, namely the two on the left. The assembly guide says they need to be slightly offset towards the middle, but do not say how much, except to say the bearing must not overlap the depression. The assembly guide provides no measurements that would help assemblers get it right. You have to judge by eye in reduced light whether the bearing overlaps the edge of a black hole beneath it.
I measured mine using a feeler gauge, and found there was no overlap, but the gap was very small ~0.1 mm.
But after the chat closed, I went about the process of re-aligning those two bearings. I did it to test Andre's assertion that an adjustment to bearing position will cure the problem.
This task not for the faint of heart or those short on time, as this involves at minimum removal of the heated bed shiny rails, the y axis idler, & the y axis motor before you can easily get at the screws holding the bearings. I nudged one of the bearings a tiny bit, so its enough to pass a 0.2 mm feeler gauge. Then reassembled the whole thing, and adjusted belt tensions during the process because no doubt the previous setting was disturbed by all this (it was). I also tuned X axis belt tension at the same time (It was also maladjusted).

And, I found that the 5.1.2 firmware now passes all the geometry self tests, but I cannot for the life of me understand why it checks the Z axis 3 times over. I'll not be convinced that this firmware change (in the 5.1.0 release) was well thought out or executed or tested. And why it has not improved in the 5.1.2 release.

The 5.1.x firmwares clearly have a much tighter tolerance for the Y axis geometry than earlier versions. But One wonders at the sanity of Prusa releasing firmware that requires every Mk4 Kit assembled owner to rejig their Y axis for the sake of a 0.1 mm tweak escapes me. And all without telling anyone. they ought at least own up to this debacle and apologise to all the customers inconvenienced this way. I personally hold the view that the firmware ought to be re-engineered to have a slightly wider tolerance, not requiring owners to pull their printer apart for the sake of some nuance in exactitude that isn't explained or described in release notes.

Zappes, 2023-12-20:
Indeed. If we were talking about tweaking a screw or something, one could accept that easily. Being forced to perform this kind of disassembly on an otherwise flawless, _factory assembled_ unit that is just a few weeks old... No, sorry, that's not OK.

blue0ceanby, 2023-12-20:
After changing the xBuddy board from version 2021 to 2023 (yes, it's strange, it was sent to me by Prusa support) because of an incorrect temperature measurement.
After installing the board, the Y self-test failed. After contacting dme support, I was suggested to downgrade to 5.0.1 and wait until a solution came from the research department.
With the downgrade to 5.0.1 the Y self-test worked.
I was curious that there was a clue with the orientation of the bearings. The first attempt to increase the distance was by pushing the bearings together as far as possible, which didn't fix the error.
Then I separated the bearings to shorten the Y path by about 1mm and it worked reproducibly for me with firmware 5.1.0 and 5.1.2.
I didn't change the belt tension.
could you check which xBuddy board the error occurs with?

Zappes, 2023-12-20:
@blue0ceanby Ich kann Dir die Version nicht wirklich sagen - kann man die rausfinden, ohne das Gerät zu zerlegen? Die Seriennummer ist jedenfalls "10589-3742441631731617".

(I can't tell you the version - can that be determined without disassembling the printer? The serial number is "10589-3742441631731617".)

blue0ceanby, 2023-12-20:
you have to loosen the 4 screws from the xBuddy board cover then it can be read in the center of the board
[image]

Zappes, 2023-12-20:
Yeah, I'd have to pull it from the enclosure for that... Sorry, that would be a bit too much right now.

blue0ceanby, 2023-12-20:
Sorry, that was just a suggestion to narrow down the error.

Zappes, 2023-12-20:
No problem, and I would have loved to help you - but the way Prusa designed their enclosure makes getting the printer out of it quite an adventure. :D

lezetuser, 2023-12-20:
So instead of internally escalate, they have a problem, and ask the programmer to sort out what issue they have, they tend to let the users to adjust even the factory assembled printers to re-adjust. What happens if they fail? Will the warranty remain still valid?
I think they approach will change only when the factory assembled printer users will start mass sending their printers back to factory to re-adjust  on Prusa cost.

lezetuser, 2023-12-20:
> After changing the xBuddy board from version 2021 to 2023 (yes, it's strange, it was sent to me by Prusa support) because of an incorrect temperature measurement. After installing the board, the Y self-test failed. After contacting support, I was suggested to downgrade to 5.0.1 and wait until a solution came from the research department. With the downgrade to 5.0.1 the Y self-test worked. I was curious that there was a clue with the orientation of the bearings. The first attempt to increase the distance was by pushing the bearings together as far as possible, which didn't fix the error. Then I pushed the bearings apart to shorten the Y path by about 1mm and it worked reproducibly for me with firmware 5.1.0 and 5.1.2. I didn't change the belt tension. could you check which xBuddy board the error occurs with?
> 
> SRY I write German with Google Translate

The x-buddy version does not matter. The FW version is the key. As far as I will have a time I will look at the firmware, what they did.

BID3D, 2023-12-20:
My unscientific theory regarding the increase in y axis self failures on my MK4 on newer firmware versions is caused by one or more the following affects….the motors on Mk4 are new and have increased torque values compared to older motors…repeated self test crashing of the y axis is slowly moving (bashing) the left side front y axis front and rear bearing closer together… and when his happens on machines with out of spec belt (+ or -) tensions the cumulative net change during self test is a a slight increase in the overall y axis travel resulting in y axis self test failure.. or maybe the y axis motor encoder step values are off in the firmware for MK4 units…or ??

lezetuser, 2023-12-20:
@Zappes 
you can find your x-buddy version on the Information page of the printer menu

bryn51, 2023-12-20:
> After changing the xBuddy board from version 2021 to 2023 (yes, it's strange, it was sent to me by Prusa support) because of an incorrect temperature measurement.
>
> After installing the board, the Y [...]

Belt tension may not be important, but having pulled apart the bed belt tension has been affected therefore it needs to bd checked upon reassembly. I also found when checked: X belt tension had changed since initial calibration.
This suggests belt tension needs checking every few months, not annually as I previously  thought.

Prusa-Support, 2023-12-29:
Thank you all for sharing.

The axis test and homing have been improved quite often since the release of Prusa MK4/XL and you can still expect rather frequent firmware upgrades for these new products - and for Prusa MINI which now shares most of the new code and features.

- Since [FW 5.0.0](https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v5.0.0) the printer motion system got a major upgrade for the implementation of the new Input Shaper, Pressure Advance and Precise Stepping. Also, the axis test was modified to let the printer automatically identify the MK4 and MK3.9 stepper motors.
- On [FW 5.0.1](https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v5.0.1) the self-test and homing algorithm was slightly improved but mostly to make up for X-axis unexpected failures. Also, the firmware was slightly adjusted to correctly identify invalid measurements of the Y-axis on all xBuddy board revisions.

All of the above can be found on the release notes and I can't exclude minor adjustments of the relevant portions of the code in newer firmware releases - it doesn't seem the case though.

Firmware upgrades could lead to even slight changes but this shouldn't be a reason for our developers _not_ to improve the firmware in all possible ways, and please check with us via email/chat as _changes are not necessarily bugs_.
Firmware upgrades are for the best and we will keep doing so, improving existing features or adding new ones.

However, the problems reported so far are pretty much always hinting at assembly concerns (at least to some extent) - either the cases reported in this thread or the ones in similar already opened issues (#3090, #3215, #3289, #3295...) .

Now please let me share a personal consideration.
I was under the impression that the cases of Y-bearing slippage gradually increased since the introduction of fast printing. I don't have adequate evidence at hand so please notice this is mere speculation.
This said, _in case of Y-axis problems during the test or the homing, I strongly recommend [re-aligning the Y bearings](https://help.prusa3d.com/wp-content/uploads/2023/08/f43e117fc2b419b5e316863c57c0658d_painted-800x600.jpeg) (especially the front one) and setting the bearings clips just a little bit tighter_.

In case of doubts, please mind our [Customer Support](https://help.prusa3d.com/article/customer-support_2287) is always available to assist.

Michele Moramarco
Prusa Research

bryn51, 2023-12-29:
Thanks Michelle for responding to this issue. But sorry it is not good enough.

Some points:
1. The assembly guide showed an image somewhat similar to the one you linked. Its vague and has no measurements that would allow the user to get the placement of the left front bearing correct. I my own case there was a gap of 0.1 mm between the bearing and the depression. To fix the issue I made it 0.2 mm.
This kind of precision in the firmware just does not match the vagueness of the instructions. Prusa should acknowledge and fix those instructions, especially in circumstance that thd firmware y axis test has been made more strict.
2. It simply is not reasonable for Prusa to alter the firmware requiring hundreds of users to pull the bed apart without an explicit advice to that effect, and without giving instructions on what needs to be done to cure a y axis test failure should it occur. As it is users have had to develop their own fix, and there are several, some better than others.
3. It may be that this widespread issue wasn't anticipated by the developers. It's a failure of analysis and testing. But sticking to your guns and refusing to fix this bug just reflects stubbornness and a degree of arrogance.
4. If this was done for IS then it failed. My own test reveal ripples/waves at first layer in a large horizontal printed object near features such as modeled holes. This coincides with places where the extruder quickly changes relative direction. This has been reported on another thread.
5. This issue should have been exposed during testing phases, but evidently it wasn't or was ignored. This points to flaws in the alpha-beta test regime used by prusa. Your testers just aren't good enough. Thats why testers need to be changed often. If they are close to developers and develop a rapport they cannot critically appraise the software. They also need to have a broad focus to see the bigger picture. Such as flawed documentation.
6. Speed. What customers need is faster printing. It's addressed by most printer manufacturers. Yes, quality can be sacrificed, it's a trade off the user can make. But speed is not in play with this bug. It's a canned test procedure not a gcode, and it can  use whatever speed is appropriate.
Yes, it may be the case that frequent use of higher printing speeds somehow triggers this issue, but the pathway isn't clear. 
But certainly past firmwares that caused the bed to bash against end stops with some force multiple times is a more likely candidate, as it could move bearings in their mounts by small amounts. What is  the reason why such force needs to be used ?

Zappes, 2023-12-30:
> However, the problems reported so far are pretty much always hinting at assembly concerns (at least to some extent) - either the cases reported in this thread or the ones in similar already opened issues (#3090, #3215, #3289, #3295...) .

Well, in that case my factory-assembled MK4 has been screwed together by somebody not qualified to do so. If it wasn't such an annoying task to get the damn thing out of the enclosure, I would pack it up and send it in for warranty repair right now. I paid that extra amount for an assembled device because I a) wanted to avoid issues like this and b) didn't want to keep taking apart the printer to fix this kind of issue.

For now, staying on FW 5.0 is a valid option for me even though the constant nagging about a new version being available is rather annoying. I really hope that you revise your stance about this not being an issue that Prusa needs to fix before some PrusaConnect update requires me to upgrade the FW.

Prusa-Support, 2024-01-09:
Our printers are the most rated and demanded [since 2016](https://www.prusa3d.com/page/our-story_875/), and yet the release of Prusa MK4 was our smoothest and most successful one. Last year was intense but the road ahead is long and we are just getting started. :wink:
Please expect more firmware upgrades in the months and years ahead. 
The same goes for assembly guides and quality controls that don't directly concern this GitHub repository - we strongly believe there is always room to improve.
In case of problems, we will be ready to take action - please contact our Customer Support.

I hope you can understand we released a new product and will keep on improving it for years, and this is a regular process.
We thoroughly test all firmware versions before the release and don't expect collateral problems.
We can filter or confirm bugs via the Customer Support service before Github, and possibly solve individual problems along the way.

This repository is mainly intended to help improve firmware.
For open discussions, please refer to the [forum](https://forum.prusa3d.com/).
For those who are not satisfied with the product, our Customer Support is available to evaluate individual cases and offer solutions accordingly. 

- Back to the Buddy Firmware issues now.

Thanks for sharing. We have collected feedback as we do from every single individual chat/email conversation.
_We are not necessarily looking at a bug_ but eventually, all feedback can give us opportunities to improve the firmware too.
This said this problem will not necessarily find a solution at the firmware level so please check the front Y linear bearing and follow the other basic troubleshooting steps with the help of our Customer Support.

If you wish to contribute to improving the product, we would appreciate it if you check the Y bearings (especially the front one), the belt tension, and other possibly related aspects with the help of our Customer Support, and then report back what solved the problem for you, or if the problem persists.

Michele Moramarco
Prusa Research

tkircher, 2024-01-09:
> Our printers are the most rated and demanded [since 2016](https://www.prusa3d.com/page/our-story_875/), and yet the release of Prusa MK4 was our smoothest and most successful one. Last year was intense but the road ahead is long and we are just getting started. 😉 Please expect more firmware upgrades in the months and years ahead. The same goes for assembly guides and quality controls that don't directly concern this GitHub repository - we strongly believe there is always room to improve. In case of problems, we will be ready to take action - please contact our Customer Support.

 This patronizing reply is worse than just saying nothing. I personally haven't had a single issue with my kit MK4 but clearly some people do have issues with their factory-assembled MK4s and it's clearly firmware related. A more constructive response would be to address the issue directly, point out exactly which changes in the code were made to the X and Y axis calibration to cause this behavior, that the intention was to improve something, and acknowledge that maybe the bearings could have shifted during shipping and that it needs to be checked carefully as part of the normal setup.

Silvenga, 2024-01-09:
> This patronizing reply is worse than just saying nothing. 

I would agree with @tkircher, that repose was incredibly condescension - @Prusa-Support if this is how Prusa feels, that's concerning.

Prusa-Support, 2024-01-09:
@tkircher @Silvenga 
That was meant to say that the product and firmware are thoroughly tested but, as usual, you can expect firmware modifications along the way and **this is a regular process**.

Michele Moramarco
Prusa Research

lezetuser, 2024-01-09:
All of us having problem with post 4.7 FW version Y calibration issues are having no problem with Y calibration on 4.7 FW.
So the only thing the programmers have to do is to look at the changes in Y calibration between 4.7 FW and 5.1 FW.
I am a bit concerned why it is labeled as HW issue when  on the 4.7 it works and on 5.1 it does not work. Between two flashes ( max 10 minutes) the HW can not change so dramatically.

I miss here a statement that the issue was tested by  Prusa and and it was unable  to reproduce. If it is so, the evidence is provided above.

tkircher, 2024-01-09:
> @tkircher @Silvenga That was meant to say that the product and firmware are thoroughly tested but, as usual, you can expect firmware modifications along the way and **this is a regular process**.

Maybe this is a language issue, but for a native English speaker in the US, when you start your response to a specific complaint by explaining that your company is popular and successful, what you are actually saying is "you are dumb".

I'll write a better response: It is likely that earlier versions of the firmware just simply did not detect the existing hardware issue, so did not complain. That doesn't mean the issue didn't exist, only that the firmware had not been improved to the point where the hardware issue was detected. Now the checks are more strict.

That doesn't change the fact that the new, more strict standards in the MK4 firmware should still have passed a factory assembled unit. Maybe your delivery person dropped the box, maybe QA failed this one time. It doesn't seem like too many people actually have this problem, so you are probably just the lucky one to get the 0.01% that had a problem.

I imagine if you asked, Prusa would compensate you for your trouble, with some free filament or parts or whatever. After all, if it's super rare for an assembled printer to have an issue, it wouldn't cost them much to make that offer.

Kris5025s, 2024-01-09:
Hello @Prusa-Support, could you give us more details on what was the reason of making Y-axis calibration more strict with the latest version of the FW? I am another victim of "Y-axis calibration fail" and the only way to get it working was downgrade to 4.7.2, calibrate it there (calibration was smooth as butter), and then flash 5.1.0 skipping calibration. In that way, I was able to avoid disassembling (in my humble opinion) fully operational printer which had hundreds of flawless print hours on 4.7.2 FW. Moreover, it seems to also do fine on 5.1.0, apart from calibration issues of course.

teak, 2024-01-09:
Adding my report so maybe this can be looked into properly. I have a kit MK4, it was one of the first kits shipped out. It has been running fine since I built it, however I had to do some maintenance on the X-axis. After completing that I ran the "XY Axis Test" and to my surprise my unmodified Y-axis fails.

I have checked and adjusted my bearings and checked my belt tension, the test still fails. While this was a kit build, I honestly think there is nothing wrong with the hardware or assembly of my printer.

Some feedback:
- More information about why the test failed... Is my Y-axis too long or too short?
- I'd like an option to just tell my printer that everything is fine and ignore the test so I can get on with my printing without it nagging me constantly.
- The "XY Axis Test" should be labelled "XYZ" as it also tests Z-axis, or better yet separate them into three different tests.

jvasileff, 2024-01-16:
> repeated self test crashing of the y axis is slowly moving (bashing) the left side front y axis front and rear bearing closer together

That is exactly what is happening with my MK4. I centered the front bearing which fixed Y axis calibration, but then the calibration routine bumped the bearing back out of place.

The problem is of course that the screws holding the bearing in place need to be tightened, but if they are too tight, bearing performance will suffer. It would be better if the calibration routine didn't so violently bash the bearing into the frame.

Here's a picture of the bearing after being dislodged by calibration:

[image]

Paja-git, 2024-01-16:
In my case the Y calibration measures 211mm instead of 210. The current FW limits are too precise, I suppose. The fix should be, IMHO, quite easy - just to add a tolerance.  I really don't follow the service desk explanation and their wording.

jvasileff, 2024-01-16:
> In my case the Y calibration measures 211mm instead of 210

After successful calibration, my Y axis range is from -4 to 211, so 211 doesn't appear to be a problem.

For anyone trying to tighten the screws holding the bearings, I was able to use a 2.5mm Park Tool Stubby Hex Wrench without disassembling the heat bed.

[image]

tkircher, 2024-01-17:
> The problem is of course that the screws holding the bearing in place need to be tightened, but if they are too tight, bearing performance will suffer. It would be better if the calibration routine didn't so violently bash the bearing into the frame.

In a rigid mounting system it really is easy to distort linear bearings like this to the point where they seize. In this design, the compliant rubber pad on one side and the spring steel on the other makes that nearly impossible.

Something you might consider, if you are convinced you want to keep your bearing mounts loose, is a snap ring (circlip) in the linear bearing groove on the side that experiences the impact. In that case, when the bearing is pushed, the ring will hit the edge of the spring steel mount and it won't be able to go out of alignment as you illustrated in your photo.

There are plenty that have enough clearance and the correct geometry to just snap right in. But if you get one that doesn't quite fit due to interference, a couple minutes with a file should take care of it.

jvasileff, 2024-01-17:
@tkircher thanks for the insight.

When building the MK4 in July, I noticed the bed did not slide as smoothly when the mounts were snug (same on MK3), so I backed off the tension a bit. But it's very possible they further loosened over time. For now, I've tightened the problem mount which I think will fix the problem, at least for another 6 months.

Zappes, 2024-01-19:
Best ask Michele. She surely has a great way of telling you how much better than us she is.

bryn51, 2024-01-19:
> I also have problems with this Y axis and I solved it by limiting the Y travel to 1.7mm So I would say that the race is too long and the test systematically fails Why doesn't @prusa listen to all this feedback and change this tolerance ?? It's still amazing ! There are so many of us who have these problems

Yes.  THere seem to be several possible speculative reasons why they are so slow to come up with a revised version.
1. They live in their own little world and do not hear what Users are saying

2. A certain level of arrogance in insisting that the way it is now is "right" and all the users  are "wrong"
3. The original programmer died and the other programmers do not want to look at his work.

Certainly the current (5.1.2) self test is a real "piece of work" with the z axis test repeated several times, and the x and y axis test having greatly excessive force applied by the steppers to the homing operation.  10 or more heavy bashes against the backstop.  Causes printer wear and tear.
Its really poor workmanship and not a great credit to Prusa Research.  Maybe they are ashamed of it, no programmer wants to own it therefore will not take on the task to fix it.  They certainly ought to be ashamed.

lezetuser, 2024-01-20:
@Zappes 
Just for clarification Michelle is him :) .

Michelle, did you have a chance to talk to the programmers, or do they ignore you?

Paja-git, 2024-01-20:
> Best ask Michele. She surely has a great way of telling you how much better than us she is.

She is he, actually. 😉

Zappes, 2024-01-20:
I hereby apologize for my misinformation. I originally assumed that "he" would be correct, was corrected by somebody who told me I had assumed wrongly and... Whatever, gender is a construct, anyway. :) While Michele's condecending comments made me prickly, their gender was never of any relevance for me or this discussion.

bryn51, 2024-01-20:
Thanks @michelle for your response however I for one am not satisfied by what you said.
- Firstly you have not acknowledged the root of the issue lies in the somewhat vague instructions contained in the assembly guide has led to this issue. Folks assemble the printer according to the instructions and now, months later new firmware comes out with altered selftest that says it was done wrong. Furthermore you say that the said firmware was routinely tested but it would seem that is simply a LIE. Had it been properly tested then the test results would have indicated a certain percentage of mk4 owners would be forced to adjust the Y axis, and the release notes could/would have words to that effect, giving people warning so they could take appropriate action.
   
I DEMAND that Prusa Apologise for the inconvenience this has caused to owners of the mk4 and furthermore DEMAND that the firmware be corrected. Its should not be necessary for mk4 owners to adjust the y axis when the printer was assembled according to the instructions. Either the instructions are flawed or the firmware is flawed you cannot have it both ways. And you owe your customers an apology for failing to properly test the firmware and give them warning about the consequential need for them to make a difficult adjustment to their printer.

Furthermore the x and y axis calibration and selftest REMAIN FLAWED to this day. The force applied (extruder and bed) to test the home position and movement range can if itself lead to exacerbation of any issues. Particularly the Y axis. Hard movement to the rear can cause calibration loss of y belt tension, and Hard movement towards the front can cause movement of the front left bearing in its mount.
And we also observe that the x and y calibration steps appear at random intervals when a print is started. This prolongs print start times, and can result in excess printer wear and the reason for doing these actions is not announced to the user by way of the UI.
The entire Geometric selftest and calibration regime is FLAWED and requires attention, explicit acknowledgement and corrective action.
Its OBVIOUS this is the case and your vague platitudes about ongoing firmware improvement just do not cut the mustard. 

This issue requires a top priority and early resolution to the exclusion of any other firmware improvements that might or might not be on the agenda.

Many mk4 owners are already aware of this issue and the world is watching and waiting for a cogent response. Your response here was a waste of time as you failed to acknowledge the issue and its consequences to users and its root causes. 

Your response displayed an excess of pride when contrition was what is expected.

Jessfo, 2024-01-23:
Hello,
For informational use, I am adding my experience with the Y axis failure after upgrading my MK3S to a MK4. 

When I initially flash firmware 5.1.2 and walk though the setup and self test, the Y axis will pass, but, if I print something and then run the self test (I have been having issues with getting a good first layer) the Y axis then fails. This could be mechanical as I see some people have had success with adjusting the bearings, but I have not touched the bearings. Purely by factory resetting and reinstalling the firmware, I can get my MK4 to pass the initial setup and self test to then later fail if I try to recalibrate. This looks like there may be a quirk in the firmware or fine tuning the axis has left too little tolerance for normal printer use. 

For now, I am using the previous firmware version as I have not had any issues with failing the Y axis calibration step at setup or recalibration.

sphxdr, 2024-02-06:
Hello everyone,

I've readjusted my rods as on this new tutorial video (https://youtu.be/UM3Ms1VA9fE?si=BvQMMfGQnRaLxzgY) and it worked at the first try!

(As stated on my previous comments I have a kit and firmware update 5.1.3)
I hope 🤞 that helps you as well!

lezetuser, 2024-02-06:
sphxdr
Did  you mean rods or bearings?
To move the bearings more far from each other works, there is just a question whether they will bear those strong bumps on the fron side.

sphxdr, 2024-02-07:
Oh, yes!!! I meant the bearings!!

It can be of course that this test is kicking a little bit harder on the front frame...

StayPirate, 2024-02-08:
> I've readjusted my bearings as on this new tutorial video (https://youtu.be/UM3Ms1VA9fE?si=BvQMMfGQnRaLxzgY) and it worked at the first try!

Hi @sphxdr the video is not accessible due to strict permissions (`This video is private`), can you please make it public? Thank you.

StayPirate, 2024-02-08:
I'm facing the same issue as all of you and I didn't manage to fix it. BTW, I have noticed that on the last firmware release [changelog](https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v5.2.0) (v5.2.0), they mentioned that **XYZ calibration test got deprecated!** 

> XYZ calibration
> 
> This feature dates back to the old days of the MK2 platform, where the printer's frame was susceptible to a certain degree of skew and XYZ calibration helped you to align it back.
> 
> Starting with the MK3, the frame became more rigid and perpendicular by design and with further improvements in the 32-bit firmware there is no need to keep this feature anymore.

I know that v5.2.0 is not intended to be flashed on a MK4, but I guess that the explanation they provide would be valid for the MK4 as well.

I'd like to ask @Prusa-Support if the developer team has planned to deprecate the XYZ calibration test on MK4 on its next firmware release. If yes, do you have any estimated release date?

[20 more replies]

closed 2024-12-16 by github-actions[bot]