prusa3d/Prusa-Firmware-Buddy #3524: [BUG] "Heatbed port overcurrent" error when starting to print

EEJoshua, 2023-11-26:
**Printer type** - Prusa MK4 Kit

**Printer firmware version** - 5.1.0 stable from site

**Original or Custom firmware** - Original

**Optional upgrades** - None

**USB drive or USB/Octoprint**
  Sending print from PrusaSlicer over Ethernet, using original USB drive that came with it

**Describe the bug**
Every once in a while I send a print from PrusaSlicer, it goes through, the printer suddenly restarts and I get a "Heatbed overcurrent" error screen and beeps. Restarting the printer makes it boot regularly, but if I try to start the print manually from the USB drive it happens again and again.
The only way to escape this is to go to Control menu and turn up the heatbed temperature manually. No need to wait for it to warm up, just setting it to something high solves the problem and the next print starts as expected.

**How to reproduce**
Not sure, it happens every once in a while on it's own.

**Expected behavior**
The print should start normally, without restart and error.

**G-code**
It happens with different print files, recently with the attached file.
[Venchy_0.4n_0.2mm_PETG_MK4IS_1h14m.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/13466616/Venchy_0.4n_0.2mm_PETG_MK4IS_1h14m.zip)

**Crash dump file**
No crash dump file after restarts

Logicwax, 2024-01-26:
I have this same issue.  Very annoying.  Hopefully there's a firmware fix for this soon.

shledge, 2024-04-09:
Have this issue from time to time, it only goes away when I manually set temps.

zpeters, 2024-07-06:
Did anyone in this thread find a solution?  I recently assembled a mk4 from kit and am experiencing a similar issue.  All self tests (including heating the nozzle and bed) complete.  However the provided prints (haven't tried others) will reboot with an overcurrent error as soon as the print starts.  One time i was able to get a print partially started by pre-heating the bed and nozzle first.

After talking with support they are suspecting the power cabling for the heatbed.  If this ends up being a resolution i will report back.

i believe this to be some sort of hardware issue but i wanted to post here in case anyone found a better solution

danopernis, 2024-07-08:
I was digging into this some more and here is what I have found: There are two "overcurrent" protections: _heater_ and _input_; _heater_ is related just to the nozzle, _input_ is related to the entire printer. _input overcurrent protection_ fires faster than _heater_ so the error message you are seeing is misleading. The team responsible for documentation was already notified and hopefully they will change the documentation soon.

This doesn't solve your issue, but perhaps sheds some more light on what is actually happening. I too believe this is a hardware issue.

zpeters, 2024-07-10:
just to keep up with my saga.  Replacement power cables did not resolve my issue. We are trying a new xbuddy board next

zpeters, 2024-07-15:
For my issue a new Xbuddy board fixed it.  Comparing the good and bad board there are no visible defects or differences.  Hardware rev was : rev37|2023

danopernis, 2024-07-15:
@zpeters can you please share more info? How often did you get the warning before vs. now? Were there any typical actions that would lead to error?

While as a firmware developer I appreciate that in the end it was hardware issue after all, I would still like to hear as many details as possible.

zpeters, 2024-07-15:
Absolutely!  I tried 6.0.2 and 6.0.3 of the firmware with the same issues.  The trigger was any time the nozzle heating and bed heating were initiated at the same time.  Aside from a few "random" instances i was able to get this to repeat consistently.  

Here is how i could trigger it:
- Run the self test for the nozzle/heatbed
- Pre-heat for PLA (i did not try other filament settings)
- Attempt a print

Typically with the selftest, i would get a reset on the heatbed testing stage.  For the PLA preheat it was an instant reset.  For the print it was often (90% of the time) as soon as i clicked the "play" button.  Once i was able to lay down the first layer, but that was maybe 1 time out of dozens of attempts.

If i were to "manually" heat the nozzle _then_ the heatbed i was able to get both up to temperatures higher than a standard PLA setting.  By doing this i was able to start a print (the provided gcode for the keychain) but this only lasted until the 2nd layer.

I hope these are good details.  Please pick my brain for more!

aeonik, 2024-08-05:
This error currently happening to me as well. Though this error began appearing while I was cleaning the **nozzle** during its heating phase. I was using a paper towel with 80% Isopropyl Alcohol. I can no longer get the printer working, this error "Heatbed Overcurrent" error message appears every time I try to increase the Nozzle temperature now. I can trigger it reliably, 100% of the time, and also means my printer currently does not work...

I started probing the circuits with an oscilloscope, and the **heatbed** does not seem to be getting any spikes in current nor voltage. In fact it works just fine when I start heating it by itself.

If I turn the the temperature up on the **nozzle** however, the Mk4 reboots crashes and displays this error message for me.

What's weird is I can manually power the hot end with a power supply after being disconnected from the rest of the printer.

I am currently unable to print, and without any more information or schematics, I am not sure how to collect more data for you all to narrow down the issue.

Regards,
Aeonik

Logicwax, 2024-08-06:
I was able to resolve this by convincing prusa to ship me a replacement xBuddy board.   Once they did that, I've never had the problem again.

CaveRenderPro, 2024-11-22:
My MK4S shows this error message if I print directly. After a filament change or short heating time, printing work without problems. @Prusa Support: Please checkout all comment to this issue https://help.prusa3d.com/article/heatbed-port-overcurrent-26309-mk4s-13309-mk4-21309-mk3-9-23309-mk3-5_405612

tkanarsky, 2025-09-10:
Seeing this on my new-to-me kit built mk4. I've made sure that the heatbed cables are secure and undamaged and checked continuity on all the fuses. I'm also able to sidestep this by preheating a bit before starting a print, but would appreciate an update on whether this can be fixed in firmware or whether I should go beg Prusa for a replacement xbuddy :)