prusa3d/Prusa-Firmware-Buddy #3525: No virtual extruders in firmware 5.1.0

Blubbertje321, 2023-11-26:
### Description of the bug

In previous versions, you could use virtual extruders for multicolor printing. Since the 5.1.0 update for MK4, the printer get stuck when launching a gcode with more than 1 virtual extruder, displaying a black screen.

Make the printer compatible with multiple virtual extruders

### Project file & How to reproduce

[g code.txt](https://github.com/prusa3d/PrusaSlicer/files/13461798/g.code.txt)

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.7.0

### Operating system

Windows 10

### Printer model

Prusa MK4

rianweaver, 2023-11-26:
I just noticed this today as well.  Frustrating to lose functionality with an update.

Pdamga, 2023-11-27:
I have the same problem with FW 5.1.0. But I can`t go back to FW 5.0.1. The reset button don`t seems to work on my MK4.

Blubbertje321, 2023-11-27:
> I have the same problem with FW 5.1.0. But I can`t go back to FW 5.0.1. The reset button don`t seems to work on my MK4.

You have to press the reset button, and then immediately after that, the knob. You must have the 5.0.1 firmware on board of the USB. Have done the same so I'm running on 5.0.1 again. My business is running with multicolor prints so I need this feature.

polarfar, 2023-11-27:
When it boots, as soon as you see the logo, double press the knob. If you have 5.0.1 firmware (or other} on the USB drive, it will ask if you want to flash it.

Pdamga, 2023-11-27:
@Blubbertje321 
That is exactly what I have done. And I have the FW 5.0.1 on my USB drive. But the printer never ask my if I want to flash FW.
@polarfar 
I will try what you say. "double press on the knop"
Thanks both for your comments.

Blubbertje321, 2023-11-27:
> This might help. I had the black screen too when I initiated the print from the controls on the printer. I looked at the printer in PrusaConnect and it was shown as 'waiting for input' or something like. I power cycled the printer then initiated the print from PrusaConnect and it worked as expected. Didn't try from PrusaLink, but that may work as well.

Aha! However, I prefer starting a print from the printer itself. There are files pre sliced and ready on start up :)

dintz, 2023-12-04:
Same problem for me!

dintz, 2023-12-04:
> This might help. I had the black screen too when I initiated the print from the controls on the printer. I looked at the printer in PrusaConnect and it was shown as 'waiting for input' or something [...]

Starting printing via PrusaLink also helped me, thanks. Should still work via the printer!

bkerler, 2023-12-21:
The issue is known and is expected to be fixed in the next major release.

Blubbertje321, 2024-04-18:
Did somebody test this out with the new 6.0 firmware yet?

rianweaver, 2024-04-18:
> Did somebody test this out with the new 6.0 firmware yet?

Starting a test print right now.  It initially gave an error about filament loaded that was wrong (as the file says PETG,PETG), but it seems to be started.  I'll update after the first color switch (only 1 in this particular print)

Blubbertje321, 2024-04-18:
> > Did somebody test this out with the new 6.0 firmware yet?
> 
> Starting a test print right now. It initially gave an error about filament loaded that was wrong (as the file says PETG,PETG), but it seems to be started. I'll update after the first color switch (only 1 in this particular print)

Thanks! Want to test out but heading to my work soon.. Keep me posted!
So no black screen when loading a file? Not using the method that Xargas mentioned?

rianweaver, 2024-04-18:
> Thanks! Want to test out but heading to my work soon.. Keep me posted! So no black screen when loading a file? Not using the method that Xargas mentioned?

I sliced the file using a 2 virtual extruder profile, loaded it to USB, and printed it from the printer. 

Apart from the initial error about a filament type (which I was able to proceed by clicking OK), the print started fine.  

Bed leveled, reheated, did the initial change (which I would always delete from the GCODE, but can't seem to with bcgode). 
It printed extruder 1 just fine, and switched to the 2nd "extruder" successfully.  

I'm willing to call that a success. My print only had a single extruder change in layer 1, but it seems to be going fine, at least for a 2 virtual extruder setup, I haven't tried more yet.

danopernis, 2024-06-05:
It has been quite a while since this was last updated. Did the v6.0.0 fix the issue? Can this be closed?

Blubbertje321, 2024-06-05:
> It has been quite a while since this was last updated. Did the v6.0.0 fix the issue? Can this be closed?

The issue is sort of solved. You can print again with multiple virtual extruders since the 6.0.0 firmware. Only "annoying issue" is now that the printer is telling that the wrong filament is loaded.

closed 2024-08-12 by github-actions[bot]