prusa3d/Prusa-Firmware-Buddy #3532: [BUG] Firmware 5.1: Printer fails Geometry self-test when it does not on 5.0.1

bryn51, 2023-11-28:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - Mk4 (Kit)

**Printer firmware version** - 5.0.1 >> 5.1

**Original or Custom firmware** - Original

**Optional upgrades** - None. Nozzle is ObsXidian 0.4 with sock fitted

**USB drive or USB/Octoprint**
  I print using Prusalink

**Describe the bug**
  When self test is performed after firmware is installed, the XYZ Calibration fails on Y axis. (but not on x & Z axis).  The self test stops at that point and does not carry one like it did with earlier version.

**How to reproduce**
  1. Install 5.1 BBF file onto USB, remove prior bbf files.
  2. after initialisation including language selection, run the calibration.
  3. when it gets to XYZ calibration closely observe behaviour.
  4. On my printer, the Y axis calibration test fails on this version.
  I downgraded to 5.0.1 to check, and on this test, it does not fail.  Q.E.D. the firmware is broken, indicating a problem when there is none, and not giving an indication of what exactly is the problem.  The issue prevent the user from using the printer in normal operations, because the printer thinks that the calibration tests have not been fully completed.

**Expected behavior**
 It is expected that the geometry will be checked by moving the Y axis and testing the precise extents of available movement.
When that is OK, the tests pases, otherwise it fails.
he initial self test should go thru every step of the complete self test, and not stop when an error is encountered.

**G-code**
  <not relevant>

**Crash dump file**
  <printer did not crash>

**Video**
XYZ test with 5.0.1
  https://drive.google.com/file/d/1ouBQcn9RkBXI6oJlCbuWOkydae9pu6RJ/view?usp=sharing
  XYZ test with 5.1
https://drive.google.com/file/d/1LyVo2_OXLfBnARRfgKB1SyQ7ylDB0RUi/view?usp=drive_link

nuv-it, 2023-11-28:
I can confirm the issue. It has been discussed in the forum with a possivble workaround. It should be fixed by new firmware, too.
https://forum.prusa3d.com/forum/english-forum-original-prusa-i3-mk4-hardware-firmware-and-software-help/y-and-z-self-test-failure-after-update-to-5-1-0/#post-686550

Zappes, 2023-11-29:
This is also addressed in this issue: https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/3521

bryn51, 2023-12-20:
OK, so 5.1.2 version of firmware arrived. I installed and tested it.  The reported issue with Y axis failing serftest remained.

I got onto Prusa Support (Andres). While calming listening to be harangue about why this issue had not been fixed, he calmly suggested I look alignment of the Y axis bearings, namely the two on the left. The assembly guide says thhey need to be slightly offset towards the middle, but do not say how much, except to say the bearing must not overlap the depression. 
I measured mine using a feeler guage, and found there was no overlap.
But after the chat closed, I went about the process of re-aligning those two bearings. Its not for the faint of heart, as this involves at minimum removal of the heated bed, the y axis idler, the y axis motor before you can easily get at the screws holding the bearings. I nudged one of them a tiny bit, so its enough to pass a 0.2 mm feeler guage.  Then reassembled the whole thing, and adjusted belt tensions during the process because no doubt the previous setting disturbed by all this.  I also tuned X axis belt at the same time.

And, I found that the 5.1.2 firmware now passes all the geometry self tests, but I cannot for the life of me understand why it checks the Z axis 3 times over.  I'll not be convinced that this firmware change (in the 5.1.0 release) was well thought out or executed.

The 5.1.x firmwares clearly have a much tighter tolerance for the Y axis geometry than earlier versions. But One wonders at the sanity of Prusa releasing firmware that requires every Mk4 Kit assembled owner to rejig their Y axis for the sake of a 0.1 mm tweak escapes me.  And all without telling anyone.  they ought at least own up to this debacle and apologise to all the customers inconvenienced this way.  I personally hold the view that the firmware ought to be re-engineered to have a slightly wider tolerance, not requiring owners to pull their printer apart for the sake of some nuance in exactitude that isn't explained or described in release notes.

Prusa-Support, 2024-01-02:
Thanks for your feedback.
Please allow me to close this issue as a duplicate of #3289 / https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/3295 / https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/3521.

Michele Moramarco
Prusa Research

closed 2024-01-02 by Prusa-Support

lezetuser, 2024-01-03:
Hi Michele,
do you think, that the last users reaction is good enough to close this unresolved issue?
So are you ok with it, that each MK4 owner with factory assembled MK4 having this issue will send the printer to Prusa for repair on the Prusa costs?

bryn51, 2024-01-03:
I'm satisfied (for it to be closed) if this issue is actually being worked on. Prusa Support intimated to me that is the case. I guess they can be trusted. But I have no trust for the developers in Prusa Research. It seems clear they decide for themselves what they work on, and it's an arbitrary non transparent process not involving customers.
Certainly they decide to do things nobody asked for, like the change to self test that was made in 5.1.1
Really this process needs reformation and greater transparency.
Also, the process of firmware release that does not synch with product usage documentation. The release notes do not comprise proper usage documentation.

lezetuser, 2024-01-09:
Early this year I disassembled the heatbed and moved the bearings. Seems not enough as this time the Y calibration sometimes succeeds and sometime not. 
There is also a weird behaviour when the heatbed bumps to the rear frame, it does it several times (I counted maximum even 8 times) like it was not sure, if it was a real rear frame - I would say it has something to do with a stall guard threshold.
Stiil space for the XY calibration improvement.